chipsalliance/Surelog #2890: tcl regression scripts

hs-apotell, 2022-04-27:
Just noticed that CI is using tcl regression scripts for all unix-like platforms. Was this intentional or accidental?

Tcl scripts don't report failure when the tests fail so issues during regression go unnoticed.

@hzeller You updated the workflow script recently. Any reasoning?

hzeller, 2022-04-27:
Ah, I was mostly playing with the new right-aligned numbers there and it also provides the sum of CPU time of all tests ('total user time') which gives a good idea of performance (whereas the Python version just showed the test with the maximum time). Since I needed more accurate process timing information then while working on some things to not introduce performance regressions (until #2802 is using operating system provided process information), it was also useful.

Anyway, ok to change back now (though it would be useful if the regression.py also had right aligned numbers and a sum of user time).

hs-apotell, 2022-05-02:
Fixed by #2897

closed 2022-05-02 by hs-apotell