chipsalliance/Surelog #2898: UHDM coverage point to empty source line

hello-eternity, 2022-05-01:
I got an error like the title. The detailed error information is like this 
[image]

I should delete the empty source line to fix it.
This is the test case [https://github.com/hello-eternity/test_orv64_uhdm](https://github.com/hello-eternity/test_orv64_uhdm )
In the Makefile.in , $(TEST_DIR) need to 
 point to the correct location. Mine is `/root/yosys-uhdm-plugin-integration/UHDM-integration-tests/tests`
This error need replace this file `orv64/subproj/es1y/rtl_gen/station_vp copy.sv` into `orv64/subproj/es1y/rtl_gen/station_vp.sv` to reproduce.

hs-apotell, 2022-05-05:
@alaindargelas I did some digging into this issue and I don't believe this was caused by the recent preprocessor change. For instance, `PpLppdr` test has been reporting a similar error long before this PR was merged in master.

https://github.com/chipsalliance/Surelog/blob/2643c07e32328c5fecc776a312c616ab4c202737/tests/PpLppdr/PpLppdr.log#L118

I believe the issue is related to multiline macros. The error disappears if the macro is written on a single line. I will continue digging further.

In the process, I did find some _questionable_ assumptions in checker logic -

https://github.com/chipsalliance/Surelog/blob/2643c07e32328c5fecc776a312c616ab4c202737/src/DesignCompile/UhdmChecker.cpp#L165-L166

The logic is assuming that the token's startLine and endLine are the same but that's not true for many tokens.

hs-apotell, 2022-05-05:
This seems like an issue in grammar or might need some other way to let embedded macros be evaluated recursively. For the following input -
```
`define MACRO1(arg) call1(``arg``)

`define MACRO2(arg) \
    MACRO1(arg); \
    call2(``arg``)

```
current grammar is treating line 4, 5 and 6 as `text_blob` and end up in the AST which later get reported as error when iterated in UhdmChecker.

hs-apotell, 2022-05-12:
@alaindargelas Need your input on this.

alaindargelas, 2022-05-25:
Complete testcase to reproduce de issue
```
`define MACRO1(arg) call1(``arg``)

`define MACRO2(arg) \
    `MACRO1(arg); \
    call2(``arg``)

module top();

function f();

`MACRO2("here");

endfunction

endmodule

```

alaindargelas, 2022-05-25:
The macro expansion in the text seems correct, since the macro body starts with a \ there is a carriage return offset at the site of the macro:

[image]

hs-apotell, 2022-05-27:
Fixed by #2969

closed 2022-05-27 by hs-apotell