chipsalliance/Surelog #2912: Significant regression slowdown since recent filename trimming in ParseCache

hzeller, 2022-05-06:
Since https://github.com/chipsalliance/Surelog/commit/7058df79d099388e95f9664efdf484c9970aa73d the runtime of the regression is
noticeably slower; on my machine it changed from about 1050 second CPU time to 1720 seconds CPU time.

Wihtout the change in ParseCache (so essentially applying the following patch, reversing that change), the CPU time reported by the regression test (`make regression`) is fast again:

```patch
--- a/src/Cache/ParseCache.cpp
+++ b/src/Cache/ParseCache.cpp
@@ -34,7 +34,6 @@
 #include <Surelog/SourceCompile/ParseFile.h>
 #include <Surelog/SourceCompile/SymbolTable.h>
 #include <Surelog/Utils/FileUtils.h>
-#include <Surelog/Utils/StringUtils.h>
 
 namespace SURELOG {
 namespace fs = std::filesystem;
@@ -56,7 +55,6 @@ fs::path ParseCache::getCacheFileName_(const fs::path& svFileNameIn) {
   if (svFileName.empty()) svFileName = m_parse->getPpFileName();
   fs::path baseFileName = FileUtils::basename(svFileName);
   fs::path cacheFileName;
-  fs::path cacheDirName = m_parse->getSymbol(cacheDirId);
   if (prec->isFilePrecompiled(baseFileName)) {
     fs::path packageRepDir = m_parse->getSymbol(clp->getPrecompiledDir());
     cacheDirId =
@@ -65,15 +63,13 @@ fs::path ParseCache::getCacheFileName_(const fs::path& svFileNameIn) {
     svFileName = baseFileName;
   } else {
     if (clp->noCacheHash()) {
-      std::string svFile = svFileName.string();
-      svFile = StringUtils::ltrim(svFile, '/');
-      svFile = StringUtils::ltrim(svFile, '/');
-      cacheFileName = cacheDirName / (svFile + ".slpa");
+      cacheFileName = svFileName.string() + ".slpa";
     } else {
       svFileName = svFileName.parent_path().filename() / baseFileName;
     }
   }
 
+  fs::path cacheDirName = m_parse->getSymbol(cacheDirId);
   Library* lib = m_parse->getLibrary();
   std::string libName = lib->getName();
   if (cacheFileName.empty()) {
```

Is this expected ? (also I don't understand quite the reason of removing two directory parts, maybe a comment would be good)

alaindargelas, 2022-05-07:
Wow, I noticed the slowdown but could not attribute it to this change. Completely out of expectation.
This change was to fix the path of the slpp and slpa files to be in the same directory with -nohash, before this change, the slpa was saved in the wrong location, outside the cache directory, it was saved alongside the post preprocessor source file.

How can this code change slow the regression beats me. It is for the better in terms of file organization. Can you guess what is slow here?

alaindargelas, 2022-05-07:
Fixed by #2914
That line was the culprit
fs::path cacheDirName = m_parse->getSymbol(cacheDirId);

When called early in the function, it might have received more illegal paths strings, not really sure what is happening in that filesystem::path variable init....

closed 2022-05-07 by alaindargelas