chipsalliance/Surelog #2924: Select Out of Bounds Warning in SystemVerilog Plugin

QuantamHD, 2022-05-10:
This might be unrelated to Surelog/UHDM, but the plugin emits warning on the following snippet of SV.

```systemverilog
module top(a, s ,z);
  input [4:0] a;
  input [1:0] s;
  output [7:0] z;

  shift inst(.a(a),.s(s),.z(z));
endmodule

module shift(a,s,z);
  parameter    width_a = 4;
  parameter    signd_a = 1;
  parameter    width_s = 2;
  parameter    width_z = 8;

  input [width_a-1:0] a;
  input [width_s-1:0] s;
  output [width_z -1:0] z;

  generate if ( signd_a )
  begin: SGNED
    assign z = fshl_s(a,s,a[width_a-1]);
  end
  else
  begin: UNSGNED
    assign z = fshl_s(a,s,1'b0);
  end
  endgenerate

  function [width_z-1:0] fshl_u_1;
     input [width_a  :0] arg1;
     input [width_s-1:0] arg2;
     input sbit;
     parameter olen = width_z;
     parameter ilen = width_a+1;
     parameter len = (ilen >= olen) ? ilen : olen;
     reg [len-1:0] result;
     reg [len-1:0] result_t;
     begin
       result_t = {(len){sbit}};
       result_t[ilen-1:0] = arg1;
       result = result_t <<< arg2;
       fshl_u_1 =  result[olen-1:0];
     end
  endfunction // fshl_u

  function [width_z-1:0] fshl_u;
     input [width_a-1:0] arg1;
     input [width_s-1:0] arg2;
     input sbit;
     fshl_u = fshl_u_1({sbit,arg1} ,arg2, sbit);
  endfunction // fshl_u

  function [width_z-1:0] fshr_u;
     input [width_a-1:0] arg1;
     input [width_s-1:0] arg2;
     input sbit;
     parameter olen = width_z;
     parameter ilen = signd_a ? width_a : width_a+1;
     parameter len = (ilen >= olen) ? ilen : olen;
     reg signed [len-1:0] result;
     reg signed [len-1:0] result_t;
     begin
       result_t = $signed( {(len){sbit}} );
       result_t[width_a-1:0] = arg1; // Line 64
       result = result_t >>> arg2;
       fshr_u =  result[olen-1:0]; // Line 66
     end
  endfunction

  function [width_z-1:0] fshl_s;
     input [width_a-1:0] arg1;
     input [width_s-1:0] arg2;
     input sbit;
     reg [width_a:0] sbit_arg1;
     begin
       if ( arg2[width_s-1] == 1'b0 )
       begin
         sbit_arg1[width_a:0] = {(width_a+1){1'b0}};
         fshl_s = fshl_u(arg1, arg2, sbit);
       end
       else
       begin
         sbit_arg1[width_a] = sbit;
         sbit_arg1[width_a-1:0] = arg1;
         fshl_s = fshr_u(sbit_arg1[width_a:1], ~arg2, sbit);
       end
     end
  endfunction
endmodule
```

## Error

```
read_uhdm file.uhdm
file.sv:64: Warning: Range [3:0] select out of bounds on signal `\fshr_u$func$file.sv:85$5.result_t': Setting 3 MSB bits to undef.
file.sv:64: Warning: Range [3:0] select out of bounds on signal `\fshr_u$func$file.sv:85$5.result_t': Setting 3 MSB bits to undef.
file.sv:64: Warning: Range [3:0] select out of bounds on signal `\fshr_u$func$file.sv:85$5.result_t': Setting 3 MSB bits to undef.
file.sv:66: Warning: Range [7:0] select out of bounds on signal `\fshr_u$func$file.sv:85$5.result': Setting 7 MSB bits to undef.
Generating RTLIL representation for module `\top'.
```

QuantamHD, 2022-05-11:
@alaindargelas I did some debugging and it looks like len is being returned as a VPIRef Object when it appears like it should be a parameter. Is that correct?
[image]

alaindargelas, 2022-05-11:
@QuantamHD, that is OK, as the ref_obj actual is the actual value of the parameter. The plugin knows how to do that.
It took me some time to compile the plugin and debugging, but I got it. I'll fix it tomorrow.
The issue is a Surelog issue, the "reg signed" keyword  combination is the problem, it prevents the parsing of the actual dimension of the reg.
If you temporarily remove "signed" everything works.

alaindargelas, 2022-05-11:
Minimum testcase:
```
module shift();
function int fshr_u;  
   reg signed [4-1:0] result;   
endfunction
endmodule
```
Missing range:

```
|uhdmtopModules:
\_module: work@shift (work@shift), id:13 tests/ScratchPad.sv:1:1: , endln:5:10
  |vpiName:work@shift
  |vpiDefName:work@shift
  |vpiTop:1
  |vpiTaskFunc:
  \_function: (work@shift.fshr_u), id:14, line:2:1, endln:4:12, parent:work@shift, parID:13
    |vpiName:fshr_u
    |vpiFullName:work@shift.fshr_u
    |vpiVariables:
    \_logic_var: (work@shift.fshr_u.result), id:15, line:3:4, endln:3:22, parent:work@shift.fshr_u, parID:14
      |vpiName:result
      |vpiFullName:work@shift.fshr_u.result
      |vpiAutomatic:1
    |vpiVisibility:1
    |vpiReturn:
    \_int_var: , id:1, line:2:10, endln:2:13
    |vpiInstance:
    \_module: work@shift (work@shift), id:13 tests/ScratchPad.sv:1:1: , endln:5:10
  |vpiTopModule:1
```

alaindargelas, 2022-05-12:
@QuantamHD @rkapuscik With #2929, you can now see the range of the var.
But I tried the Yosys plugin and the same warning message appears, so something has to happen in the plugin code.

While digging, I found more sub issues that I'll update later on, they are about the sign-ness and typespec of the functions return values in general.

hzeller, 2022-05-12:
@rkapuscik @tgorochowik can you coordinate what needs to be done in the plugin ?

hzeller, 2022-05-12:
If the parameters are changed, the problem arises again; I think the problem is parameter propagation. Let's modify @QuantamHD example above and replace top()

```systemverilog
module top (a,s,z);
  // Different size parameters than are the default of the shift module.
  parameter width_a = 240;
  parameter width_s = 9;
  parameter width_z = 240;

  input [width_a-1:0] a;
  input [width_s-1:0] s;
  output [width_z-1:0] z;

  shift #(
      .width_a(width_a),
      .width_s(width_s),
      .width_z(width_z)
  ) inst (
      .a(a),
      .s(s),
      .z(z)
  );
endmodule

// rest as above
```

The expectation is that all the functions that derive not only the return value (which now seems to be fine), but also some of its other input/output widths from parameters
```
input [width_a-1:0] arg1;
input [width_s-1:0] arg2;
```

... still have the width coming from the `module shift()` defaults in the parameters, not the values passed when instantiating `shift()`

Passing  this through Surelog -> UHDM -> Yosys and have Yosys emit Verilog we can see from the initialization values widths something interesting: it hints that Yosys is assuming the width of `arg1` and `arg2` to be what used to be the default width 4 and 2 while they should be now 240 and 9 (note that the result width looks like to be correctly calculated):

```
  assign \fshr_u$func$select_out_of_bounds.sv:101$5.$result  = 240'hxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx;
  assign \fshr_u$func$select_out_of_bounds.sv:101$5.arg1  = 4'hx;
  assign \fshr_u$func$select_out_of_bounds.sv:101$5.arg2  = 2'hx;
  assign \fshr_u$func$select_out_of_bounds.sv:101$5.sbit  = 1'hx;
  assign \fshr_u$func$select_out_of_bounds.sv:101$5.result  = 240'hxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx;
  assign \fshr_u$func$select_out_of_bounds.sv:101$5.result_t  = 240'hxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx;
```

Are the changed parameters properly propagated and emitted to UHDM ?

alaindargelas, 2022-05-14:
Fixed by #2936 , typespec where reduced to constants prematurely, so the parameters values (correctly propagated) could not be substituted (as the expressions where reduced). Also fixed the UHDM elab to point the function calls to the cloned functions in scope of the module instance instead of the function declaration.

Yosys seems happy after this fix:
[  FATAL] : 0
[ SYNTAX] : 0
[  ERROR] : 0
[WARNING] : 2
[   NOTE] : 7
Warning: Removing unused module: \shift from the design.
Generating RTLIL representation for module `\top'.
Generating RTLIL representation for module `$paramod\shift\width_a=32'd240\signd_a=32'd1\width_s=32'd9\width_z=32'd240'.

yosys>

closed 2022-05-14 by alaindargelas

hzeller, 2022-05-14:
Thank you, looks like that fixed it!