chipsalliance/Surelog #2931: [Bug] Parser just spins forever

teqdruid, 2022-05-12:
OK... this might be the least useful bug report of all time since I can't share the code which triggers the hang and haven't had time (and likely won't anytime soon) to boil it down to something I can share. It's non-trivial code (it uses structs and enums which are in packages), though nothing particularly heinous (syntactically speaking) jumps out at me. It looks to be an FSM (not my code).

Here's session:

```
~/Surelog/dbuild/bin/hellosureworld -sv -verbose -parse foo.sv
[INF:CM0023] Creating log file ./slpp_all/surelog.log.

[INF:PP0122] Preprocessing source file "export/foo.sv".

[INF:PA0201] Parsing source file "export/foo.sv".
```

`-d 4` doesn't add much I can share.

I also grabbed a stack trace ([surelog_hang_stacktrace.txt](https://github.com/chipsalliance/Surelog/files/8683044/surelog_hang_stacktrace.txt)), which I've attached. I paused it a few times, and the `antlr4::atn::ParserATNSimulator::closure_` function is the greatest denominator.

I'm running Surelog commit 807b8375d5544d6a36cd57505296911189f5881a (May 10th) debug build on Ubuntu 22.04. I've also included the cmake output: [surelog_cmake_log.txt](https://github.com/chipsalliance/Surelog/files/8683093/surelog_cmake_log.txt)

teqdruid, 2022-05-13:
Actually, whittling it down was pretty simple. The below is something which "hangs" and I feel comfortable posting. One thing I noticed: if there's only one 'if' block it parses pretty quickly, but as I add back more and more it gets worse and worse until it _appears_ to hang. So it could just be bad exponential behavior. Last week, I let it run for a few hours and it didn't finish.

```systemverilog

module Foo ( );

    generate
        for( genvar i = 0 ; i <= 17 ; i++ ) begin
            if( i == ADDR_OFFSET_PART_0 ) begin
                always_ff @(posedge clk) begin
                    if( shift_foo_bar ) begin
                        foo_bar_4[i] <= foo_bar_4[i+1];
                    end

                end
            end
            else if( i == ADDR_OFFSET_PART_1 ) begin
                always_ff @(posedge clk) begin
                    if( shift_foo_bar ) begin
                        foo_bar_4[i] <= foo_bar_4[i+1];
                    end

                end
            end
            else if( i == ADDR_OFFSET_PART_2 ) begin
                always_ff @(posedge clk) begin
                    if( shift_foo_bar ) begin
                        foo_bar_4[i] <= foo_bar_4[i+1];
                    end

                end
            end
            else if( i == ADDR_OFFSET_PART_3 ) begin
                always_ff @(posedge clk) begin
                    if( shift_foo_bar ) begin
                        foo_bar_4[i] <= foo_bar_4[i+1];
                    end

                end
            end
            else if( i == ADDR_OFFSET_0 ) begin
                always_ff @(posedge clk) begin
                    if( shift_foo_bar ) begin
                        foo_bar_4[i] <= foo_bar_4[i+1];
                    end

                end
            end
            else if( i == ADDR_OFFSET_1 ) begin
                always_ff @(posedge clk) begin
                    if( shift_foo_bar ) begin
                        foo_bar_4[i] <= foo_bar_4[i+1];
                    end

                end
            end
            else if( i == ADDR_OFFSET_2 ) begin
                always_ff @(posedge clk) begin
                    if( shift_foo_bar ) begin
                        foo_bar_4[i] <= foo_bar_4[i+1];
                    end

                end
            end
            else if( i == ADDR_OFFSET_3 ) begin
                always_ff @(posedge clk) begin
                    if( shift_foo_bar ) begin
                        foo_bar_4[i] <= foo_bar_4[i+1];
                    end

                end
            end
            else if( i == ADDR_OFFSET_PART_0_STRIDE ) begin
                always_ff @(posedge clk) begin
                    if( shift_foo_bar ) begin
                        foo_bar_4[i] <= foo_bar_4[i+1];
                    end

                end
            end
            else if( i == ADDR_OFFSET_PART_1_STRIDE ) begin
                always_ff @(posedge clk) begin
                    if( shift_foo_bar ) begin
                        foo_bar_4[i] <= foo_bar_4[i+1];
                    end

                end
            end

        end
    endgenerate

endmodule
```

alaindargelas, 2022-05-13:
@QuantamHD, your grammar fix was supposed to address these scenarios. Can you check what is the twist in this case?

QuantamHD, 2022-05-13:
Initial thoughts are that this caused by the for loop. I'll upload a profile of the grammar to confirm

alaindargelas, 2022-05-13:
Actually, I didn't read carefully the code.
The generate statement is the loop.
@QuantamHD, your optimization was for if-else-generate.

The if-else here is just a regular if-else, not a generate one, we don't have an optimization for that so the grammar goes exponential.

QuantamHD, 2022-05-13:
@teqdruid I don't know if your real module has any inputs, but if you replace

```
module Foo ( );
```

with

```
module Foo;
```

Things will get a bit faster, especially if this a giant file..

teqdruid, 2022-05-13:
It does have ports.

More to the point, I was testing if Surelog parses one of our large code bases unmodified.

QuantamHD, 2022-05-16:
@alaindargelas So it looks like main pain point in this code is the generate_item. I would expect that the semantic_predicates would selectively filter this out. Any thoughts on confounding variables?

[image]

QuantamHD, 2022-05-16:
@teqdruid How long does the above snippet take to run on your machine?

teqdruid, 2022-05-17:
With the same commit, the clip I reported takes 66 seconds. When I copy-paste two additional `if` blocks, it takes 581 seconds.

alaindargelas, 2022-05-17:
Fixed by #2939, now runs in 0.1s
@QuantamHD, please check if that does not break your internal testcase.
If this changelist works for all, we don't need the predicates anymore.

QuantamHD, 2022-05-17:
@alaindargelas Our internal module is at about 60 seconds of parsing which is well within acceptable.

alaindargelas, 2022-05-17:
I'll keep the predicate in the grammar as code example, but they are not used, until the next issue alike arises.

closed 2022-05-17 by alaindargelas