chipsalliance/Surelog #2932: [Bug?] Casting to a width stored in a struct

teqdruid, 2022-05-13:
I have no idea if this is valid SystemVerilog, but Questasim and Quartus accept it and we use it all over our code.

```systemverilog
typedef struct {
    int WIDTH;
} width_t;

module Foo ();
    localparam width_t w = width_t'{32};
    wire [31:0] t = (w.WIDTH'(0));
endmodule
```

```
[SNT:PA0207] foo_struct_width.sv:7:28: Syntax error: no viable alternative at input 'module Foo ();\n    localparam width_t w = width_t'{32};\n    wire [31:0] t = (w.WIDTH'',
    wire [31:0] t = (w.WIDTH'(0));
                            ^-- ./slpp_all/work/_6142509188972423790/foo_struct_width.sv:7:28:.
```

In order to get Surelog to parse it, I have to surround `w.WIDTH` with parens. Seems like a precedence issue in your grammar.

alaindargelas, 2022-05-13:
It's legal, the grammar had that particular rule commented out for runtime issues, I'll reactive it. UHDM compilation still need to get update to support hier_path object as a cast typespec.

alaindargelas, 2022-05-13:
#2933 fixes the grammar part, UHDM still has to be updated.

alaindargelas, 2022-05-17:
Fixed by #2940

closed 2022-05-17 by alaindargelas

teqdruid, 2022-05-17:
This worked on our large internal codebase. I think our code now parses, but now `hellosureworld` is segfaulting after a `CP0335` message. The `COMP` phase is after the parse phase, yes?

teqdruid, 2022-05-17:
Nope. Nevermind