chipsalliance/Surelog #2942: Segfault in parser

teqdruid, 2022-05-17:
Same large code base I've been pushing on, now encountering this segfault. Again, I can't provide the code which triggers this segfault.

```
__cxa_throw (@__cxa_throw:3)
antlr4::BailErrorStrategy::recoverInline(antlr4::Parser*) (\home\jodemme\Surelog\third_party\antlr4\runtime\Cpp\runtime\src\BailErrorStrategy.cpp:49)
antlr4::Parser::match(unsigned long) (\home\jodemme\Surelog\third_party\antlr4\runtime\Cpp\runtime\src\Parser.cpp:134)
SURELOG::SV3_1aParser::package_or_generate_item_declaration() (\home\jodemme\Surelog\dbuild\generated\src\parser\SV3_1aParser.cpp:17740)
SURELOG::SV3_1aParser::package_item() (\home\jodemme\Surelog\dbuild\generated\src\parser\SV3_1aParser.cpp:17508)
SURELOG::SV3_1aParser::package_declaration() (\home\jodemme\Surelog\dbuild\generated\src\parser\SV3_1aParser.cpp:9430)
SURELOG::SV3_1aParser::description() (\home\jodemme\Surelog\dbuild\generated\src\parser\SV3_1aParser.cpp:5471)
SURELOG::SV3_1aParser::source_text() (\home\jodemme\Surelog\dbuild\generated\src\parser\SV3_1aParser.cpp:5289)
SURELOG::SV3_1aParser::top_level_rule() (\home\jodemme\Surelog\dbuild\generated\src\parser\SV3_1aParser.cpp:4559)
SURELOG::ParseFile::parseOneFile_(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, unsigned int) (\home\jodemme\Surelog\src\SourceCompile\ParseFile.cpp:384)
SURELOG::ParseFile::parse() (\home\jodemme\Surelog\src\SourceCompile\ParseFile.cpp:516)
SURELOG::CompileSourceFile::parse_() (\home\jodemme\Surelog\src\SourceCompile\CompileSourceFile.cpp:196)
SURELOG::CompileSourceFile::compile(SURELOG::CompileSourceFile::Action) (\home\jodemme\Surelog\src\SourceCompile\CompileSourceFile.cpp:116)
SURELOG::Compiler::compileOneFile_(SURELOG::CompileSourceFile*, SURELOG::CompileSourceFile::Action) (\home\jodemme\Surelog\src\SourceCompile\Compiler.cpp:130)
SURELOG::Compiler::compileFileSet_(SURELOG::CompileSourceFile::Action, bool, std::vector<SURELOG::CompileSourceFile*, std::allocator<SURELOG::CompileSourceFile*> >&) (\home\jodemme\Surelog\src\SourceCompile\Compiler.cpp:790)
SURELOG::Compiler::compile() (\home\jodemme\Surelog\src\SourceCompile\Compiler.cpp:968)
SURELOG::start_compiler(SURELOG::CommandLineParser*) (\home\jodemme\Surelog\src\API\Surelog.cpp:28)
main (\home\jodemme\Surelog\src\hellosureworld.cpp:49)
___lldb_unnamed_symbol3139 (@___lldb_unnamed_symbol3139:29)
__libc_start_main (@__libc_start_main:43)
_start (@_start:15)
```

Commit daa52046ab877a1807a1233188ddb5d1db02c1d3.

```
$ /usr/bin/c++ -v
Using built-in specs.
COLLECT_GCC=/usr/bin/c++
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/11/lto-wrapper
OFFLOAD_TARGET_NAMES=nvptx-none:amdgcn-amdhsa
OFFLOAD_TARGET_DEFAULT=1
Target: x86_64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Ubuntu 11.2.0-19ubuntu1' --with-bugurl=file:///usr/share/doc/gcc-11/README.Bugs --enable-languages=c,ada,c++,go,brig,d,fortran,objc,obj-c++,m2 --prefix=/usr --with-gcc-major-version-only --program-suffix=-11 --program-prefix=x86_64-linux-gnu- --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --enable-bootstrap --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-vtable-verify --enable-plugin --enable-default-pie --with-system-zlib --enable-libphobos-checking=release --with-target-system-zlib=auto --enable-objc-gc=auto --enable-multiarch --disable-werror --enable-cet --with-arch-32=i686 --with-abi=m64 --with-multilib-list=m32,m64,mx32 --enable-multilib --with-tune=generic --enable-offload-targets=nvptx-none=/build/gcc-11-gBFGDP/gcc-11-11.2.0/debian/tmp-nvptx/usr,amdgcn-amdhsa=/build/gcc-11-gBFGDP/gcc-11-11.2.0/debian/tmp-gcn/usr --without-cuda-driver --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu --with-build-config=bootstrap-lto-lean --enable-link-serialization=2
Thread model: posix
Supported LTO compression algorithms: zlib zstd
gcc version 11.2.0 (Ubuntu 11.2.0-19ubuntu1)
```

teqdruid, 2022-05-17:
That's the wrong stack trace. Hold on.

teqdruid, 2022-05-17:
OK. We're good now. I had catch on throw on. Oops.

QuantamHD, 2022-05-17:
Good as in you no longer get the segfault?

teqdruid, 2022-05-17:
Let me rephrase: that's the correct stack trace. I still get the segfault.

hs-apotell, 2022-05-17:
Could you try PR #2935. I fixed one such issue in Google test. This was the only test that caused segfault with my change. Unfortunately the change was over a few weekends and I don't remember the exact root cause or the specific change itself.

alaindargelas, 2022-05-18:
That test also produces an infinite stack (simpler), it looks like I need to implement a loop detection mechanism in the evaluation.

module Foo ();
 parameter P1 = P2;
 parameter P2 = P1; 
endmodule

QuantamHD, 2022-05-18:
It feels like the above shouldn't be allowed because it references a variable that isn't defined yet, but I'm sure this is allowed by the spec.

It feels like this could get really complicated. Is the rule that no parameter can expand into itself?

alaindargelas, 2022-05-18:
@teqdruid, please try #2944
You should see an error message like:
https://github.com/chipsalliance/Surelog/blob/beb0e674a0b6915ab49e78f617f24e9d35776b48/tests/LoopParam/LoopParam.log#L59

that should point to the right place to start creating a small design that reproduces the error, or potentially find an error in the design (Self referencing expressions).

alaindargelas, 2022-05-18:
@QuantamHD, my example is definitely not legal, decent tools would report a self referencing expression in parameters.
If it was nets, there are many tools that report asynchronous loops (across multiple modules), that is legal by the language, but usually bad design, unless you have a thing for free-running oscillators...

teqdruid, 2022-05-18:
With #2944 it spins forever and eats memory continuously.

teqdruid, 2022-05-18:
@hs-apotell With #2935, it still segfaults but with a different stack trace:

```
SURELOG::SymbolId::SymbolId(SURELOG::SymbolId const&) (\home\jodemme\Surelog\include\Surelog\Common\SymbolId.h:42)
SURELOG::FileContent::getFileName[abi:cxx11](SURELOG::NodeId) const (\home\jodemme\Surelog\src\Design\FileContent.cpp:70)
SURELOG::CompileHelper::compileExpression(SURELOG::DesignComponent*, SURELOG::FileContent const*, SURELOG::NodeId, SURELOG::CompileDesign*, UHDM::BaseClass*, SURELOG::ValuedComponentI*, bool, bool) (\home\jodemme\Surelog\src\DesignCompile\CompileExpression.cpp:2991)
SURELOG::NetlistElaboration::elab_parameters_(SURELOG::ModuleInstance*, bool) (\home\jodemme\Surelog\src\DesignCompile\NetlistElaboration.cpp:291)
SURELOG::NetlistElaboration::elaborate_(SURELOG::ModuleInstance*, bool) (\home\jodemme\Surelog\src\DesignCompile\NetlistElaboration.cpp:474)
SURELOG::NetlistElaboration::elaborateInstance(SURELOG::ModuleInstance*) (\home\jodemme\Surelog\src\DesignCompile\NetlistElaboration.cpp:113)
SURELOG::DesignElaboration::elaborateInstance_(SURELOG::FileContent const*, SURELOG::NodeId, SURELOG::NodeId, SURELOG::ModuleInstanceFactory*, SURELOG::ModuleInstance*, SURELOG::Config*, std::vector<SURELOG::ModuleInstance*, std::allocator<SURELOG::ModuleInstance*> >&) (\home\jodemme\Surelog\src\DesignCompile\DesignElaboration.cpp:846)
SURELOG::DesignElaboration::elaborateInstance_(SURELOG::FileContent const*, SURELOG::NodeId, SURELOG::NodeId, SURELOG::ModuleInstanceFactory*, SURELOG::ModuleInstance*, SURELOG::Config*, std::vector<SURELOG::ModuleInstance*, std::allocator<SURELOG::ModuleInstance*> >&) (\home\jodemme\Surelog\src\DesignCompile\DesignElaboration.cpp:1510)
SURELOG::DesignElaboration::elaborateInstance_(SURELOG::FileContent const*, SURELOG::NodeId, SURELOG::NodeId, SURELOG::ModuleInstanceFactory*, SURELOG::ModuleInstance*, SURELOG::Config*, std::vector<SURELOG::ModuleInstance*, std::allocator<SURELOG::ModuleInstance*> >&) (\home\jodemme\Surelog\src\DesignCompile\DesignElaboration.cpp:1510)
SURELOG::DesignElaboration::elaborateInstance_(SURELOG::FileContent const*, SURELOG::NodeId, SURELOG::NodeId, SURELOG::ModuleInstanceFactory*, SURELOG::ModuleInstance*, SURELOG::Config*, std::vector<SURELOG::ModuleInstance*, std::allocator<SURELOG::ModuleInstance*> >&) (\home\jodemme\Surelog\src\DesignCompile\DesignElaboration.cpp:1510)
SURELOG::DesignElaboration::elaborateInstance_(SURELOG::FileContent const*, SURELOG::NodeId, SURELOG::NodeId, SURELOG::ModuleInstanceFactory*, SURELOG::ModuleInstance*, SURELOG::Config*, std::vector<SURELOG::ModuleInstance*, std::allocator<SURELOG::ModuleInstance*> >&) (\home\jodemme\Surelog\src\DesignCompile\DesignElaboration.cpp:1510)
SURELOG::DesignElaboration::elaborateInstance_(SURELOG::FileContent const*, SURELOG::NodeId, SURELOG::NodeId, SURELOG::ModuleInstanceFactory*, SURELOG::ModuleInstance*, SURELOG::Config*, std::vector<SURELOG::ModuleInstance*, std::allocator<SURELOG::ModuleInstance*> >&) (\home\jodemme\Surelog\src\DesignCompile\DesignElaboration.cpp:1510)
SURELOG::DesignElaboration::elaborateInstance_(SURELOG::FileContent const*, SURELOG::NodeId, SURELOG::NodeId, SURELOG::ModuleInstanceFactory*, SURELOG::ModuleInstance*, SURELOG::Config*, std::vector<SURELOG::ModuleInstance*, std::allocator<SURELOG::ModuleInstance*> >&) (\home\jodemme\Surelog\src\DesignCompile\DesignElaboration.cpp:1510)
SURELOG::DesignElaboration::elaborateInstance_(SURELOG::FileContent const*, SURELOG::NodeId, SURELOG::NodeId, SURELOG::ModuleInstanceFactory*, SURELOG::ModuleInstance*, SURELOG::Config*, std::vector<SURELOG::ModuleInstance*, std::allocator<SURELOG::ModuleInstance*> >&) (\home\jodemme\Surelog\src\DesignCompile\DesignElaboration.cpp:1510)
SURELOG::DesignElaboration::elaborateInstance_(SURELOG::FileContent const*, SURELOG::NodeId, SURELOG::NodeId, SURELOG::ModuleInstanceFactory*, SURELOG::ModuleInstance*, SURELOG::Config*, std::vector<SURELOG::ModuleInstance*, std::allocator<SURELOG::ModuleInstance*> >&) (\home\jodemme\Surelog\src\DesignCompile\DesignElaboration.cpp:1510)
SURELOG::DesignElaboration::elaborateModule_(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, SURELOG::FileContent const*, bool) (\home\jodemme\Surelog\src\DesignCompile\DesignElaboration.cpp:657)
SURELOG::DesignElaboration::elaborateAllModules_(bool) (\home\jodemme\Surelog\src\DesignCompile\DesignElaboration.cpp:581)
SURELOG::DesignElaboration::elaborate() (\home\jodemme\Surelog\src\DesignCompile\DesignElaboration.cpp:86)
SURELOG::CompileDesign::elaboration_() (\home\jodemme\Surelog\src\DesignCompile\CompileDesign.cpp:371)
SURELOG::CompileDesign::elaborate() (\home\jodemme\Surelog\src\DesignCompile\CompileDesign.cpp:275)
SURELOG::Compiler::compile() (\home\jodemme\Surelog\src\SourceCompile\Compiler.cpp:1032)
SURELOG::start_compiler(SURELOG::CommandLineParser*) (\home\jodemme\Surelog\src\API\Surelog.cpp:28)
main (\home\jodemme\Surelog\src\hellosureworld.cpp:49)
___lldb_unnamed_symbol3139 (@___lldb_unnamed_symbol3139:29)
__libc_start_main (@__libc_start_main:43)
_start (@_start:15)
```

But after a different log message (still CP0355).

alaindargelas, 2022-05-18:
I don't think this draft PR is going to help at all here: https://github.com/chipsalliance/Surelog/pull/2935

@tedruid, Can you set a breakpoint here:
https://github.com/chipsalliance/Surelog/blob/beb0e674a0b6915ab49e78f617f24e9d35776b48/src/DesignCompile/CompileHelper.cpp#L79
If it breaks there, you can 
print fileName
print lineNumber
print instName

and get an idea where this is happening.

If it does not break,
then please set a breakpoint here:
https://github.com/chipsalliance/Surelog/blob/beb0e674a0b6915ab49e78f617f24e9d35776b48/src/DesignCompile/CompileExpression.cpp#L791

To try to understand why we are looping but the stack does not increment, you might have to do quite a few "cont/steps".

alaindargelas, 2022-05-19:
@teqdruid, #2947 should unwind the loop and we should see a location for the loop error(s).

teqdruid, 2022-05-20:
@alaindargelas I pulled down master, and it does indeed not segfault at that particular point. I'm now getting a _ton_ of elaboration errors (mostly a combination of `[ERR:EL0535]`, `[WRN:EL0515]`, `[ERR:CP0317]` and `[WRN:EL0500]`) so I'm not certain which one was causing the issue. I'm definitely missing some modules, but shouldn't be missing any packages. And I'm not sure the SV files are in the correct order. No time to debug them comprehensively.

I'm now getting another segfault sometime after the `[INF:UH0706] Creating UHDM Model...` message.

```
SURELOG::Netlist::nets() (\home\jodemme\Surelog\include\Surelog\Design\Netlist.h:55)
SURELOG::UhdmWriter::writeElabGenScope(UHDM::Serializer&, SURELOG::ModuleInstance*, UHDM::gen_scope*, SURELOG::ExprBuilder&) (\home\jodemme\Surelog\src\DesignCompile\UhdmWriter.cpp:1542)
SURELOG::UhdmWriter::writeInstance(SURELOG::ModuleDefinition*, SURELOG::ModuleInstance*, UHDM::BaseClass*, SURELOG::CompileDesign*, std::map<SURELOG::DesignComponent const*, UHDM::BaseClass*, std::less<SURELOG::DesignComponent const*>, std::allocator<std::pair<SURELOG::DesignComponent const* const, UHDM::BaseClass*> > >&, std::map<SURELOG::ModPort*, UHDM::modport*, std::less<SURELOG::ModPort*>, std::allocator<std::pair<SURELOG::ModPort* const, UHDM::modport*> > >&, std::map<SURELOG::ModuleInstance*, UHDM::BaseClass*, std::less<SURELOG::ModuleInstance*>, std::allocator<std::pair<SURELOG::ModuleInstance* const, UHDM::BaseClass*> > >&, SURELOG::ExprBuilder&) (\home\jodemme\Surelog\src\DesignCompile\UhdmWriter.cpp:2543)
SURELOG::UhdmWriter::writeInstance(SURELOG::ModuleDefinition*, SURELOG::ModuleInstance*, UHDM::BaseClass*, SURELOG::CompileDesign*, std::map<SURELOG::DesignComponent const*, UHDM::BaseClass*, std::less<SURELOG::DesignComponent const*>, std::allocator<std::pair<SURELOG::DesignComponent const* const, UHDM::BaseClass*> > >&, std::map<SURELOG::ModPort*, UHDM::modport*, std::less<SURELOG::ModPort*>, std::allocator<std::pair<SURELOG::ModPort* const, UHDM::modport*> > >&, std::map<SURELOG::ModuleInstance*, UHDM::BaseClass*, std::less<SURELOG::ModuleInstance*>, std::allocator<std::pair<SURELOG::ModuleInstance* const, UHDM::BaseClass*> > >&, SURELOG::ExprBuilder&) (\home\jodemme\Surelog\src\DesignCompile\UhdmWriter.cpp:2631)
SURELOG::UhdmWriter::writeInstance(SURELOG::ModuleDefinition*, SURELOG::ModuleInstance*, UHDM::BaseClass*, SURELOG::CompileDesign*, std::map<SURELOG::DesignComponent const*, UHDM::BaseClass*, std::less<SURELOG::DesignComponent const*>, std::allocator<std::pair<SURELOG::DesignComponent const* const, UHDM::BaseClass*> > >&, std::map<SURELOG::ModPort*, UHDM::modport*, std::less<SURELOG::ModPort*>, std::allocator<std::pair<SURELOG::ModPort* const, UHDM::modport*> > >&, std::map<SURELOG::ModuleInstance*, UHDM::BaseClass*, std::less<SURELOG::ModuleInstance*>, std::allocator<std::pair<SURELOG::ModuleInstance* const, UHDM::BaseClass*> > >&, SURELOG::ExprBuilder&) (\home\jodemme\Surelog\src\DesignCompile\UhdmWriter.cpp:2583)
SURELOG::UhdmWriter::writeInstance(SURELOG::ModuleDefinition*, SURELOG::ModuleInstance*, UHDM::BaseClass*, SURELOG::CompileDesign*, std::map<SURELOG::DesignComponent const*, UHDM::BaseClass*, std::less<SURELOG::DesignComponent const*>, std::allocator<std::pair<SURELOG::DesignComponent const* const, UHDM::BaseClass*> > >&, std::map<SURELOG::ModPort*, UHDM::modport*, std::less<SURELOG::ModPort*>, std::allocator<std::pair<SURELOG::ModPort* const, UHDM::modport*> > >&, std::map<SURELOG::ModuleInstance*, UHDM::BaseClass*, std::less<SURELOG::ModuleInstance*>, std::allocator<std::pair<SURELOG::ModuleInstance* const, UHDM::BaseClass*> > >&, SURELOG::ExprBuilder&) (\home\jodemme\Surelog\src\DesignCompile\UhdmWriter.cpp:2583)
SURELOG::UhdmWriter::writeInstance(SURELOG::ModuleDefinition*, SURELOG::ModuleInstance*, UHDM::BaseClass*, SURELOG::CompileDesign*, std::map<SURELOG::DesignComponent const*, UHDM::BaseClass*, std::less<SURELOG::DesignComponent const*>, std::allocator<std::pair<SURELOG::DesignComponent const* const, UHDM::BaseClass*> > >&, std::map<SURELOG::ModPort*, UHDM::modport*, std::less<SURELOG::ModPort*>, std::allocator<std::pair<SURELOG::ModPort* const, UHDM::modport*> > >&, std::map<SURELOG::ModuleInstance*, UHDM::BaseClass*, std::less<SURELOG::ModuleInstance*>, std::allocator<std::pair<SURELOG::ModuleInstance* const, UHDM::BaseClass*> > >&, SURELOG::ExprBuilder&) (\home\jodemme\Surelog\src\DesignCompile\UhdmWriter.cpp:2583)
SURELOG::UhdmWriter::writeInstance(SURELOG::ModuleDefinition*, SURELOG::ModuleInstance*, UHDM::BaseClass*, SURELOG::CompileDesign*, std::map<SURELOG::DesignComponent const*, UHDM::BaseClass*, std::less<SURELOG::DesignComponent const*>, std::allocator<std::pair<SURELOG::DesignComponent const* const, UHDM::BaseClass*> > >&, std::map<SURELOG::ModPort*, UHDM::modport*, std::less<SURELOG::ModPort*>, std::allocator<std::pair<SURELOG::ModPort* const, UHDM::modport*> > >&, std::map<SURELOG::ModuleInstance*, UHDM::BaseClass*, std::less<SURELOG::ModuleInstance*>, std::allocator<std::pair<SURELOG::ModuleInstance* const, UHDM::BaseClass*> > >&, SURELOG::ExprBuilder&) (\home\jodemme\Surelog\src\DesignCompile\UhdmWriter.cpp:2631)
SURELOG::UhdmWriter::writeInstance(SURELOG::ModuleDefinition*, SURELOG::ModuleInstance*, UHDM::BaseClass*, SURELOG::CompileDesign*, std::map<SURELOG::DesignComponent const*, UHDM::BaseClass*, std::less<SURELOG::DesignComponent const*>, std::allocator<std::pair<SURELOG::DesignComponent const* const, UHDM::BaseC [... cut ...]
SURELOG::CompileDesign::writeUHDM(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) (\home\jodemme\Surelog\src\DesignCompile\CompileDesign.cpp:380)
SURELOG::Compiler::compile() (\home\jodemme\Surelog\src\SourceCompile\Compiler.cpp:1063)
SURELOG::start_compiler(SURELOG::CommandLineParser*) (\home\jodemme\Surelog\src\API\Surelog.cpp:28)
main (\home\jodemme\Surelog\src\hellosureworld.cpp:49)
__libc_start_call_main (@__libc_start_call_main:29)
__libc_start_main_impl (@__libc_start_main@@GLIBC_2.34:43)
_start (@_start:15)
```

`this` is null in the `nets()` method.

alaindargelas, 2022-05-21:
#2951 will fix that latest core dump, but there might be others behind, we have to go through the onion peeling, as the errors indicate missing defparams targets (WRN:EL0515), maybe because you have missing module defs WRN:EL0500. So some datastructures might be missing.
Those error messages should be addressed at some point, they'll make the elaboration more correct.

alaindargelas, 2022-05-31:
@teqdruid, can we close this issue?

teqdruid, 2022-05-31:
Yes, I believe this bug is fixed now.

Sorry I haven't had the chance to peel the onion further.

closed 2022-05-31 by alaindargelas