chipsalliance/Surelog #2954: Duplicate variables in function

hs-apotell, 2022-05-21:
Input:
```
package riscv;
  function string func();
    automatic string str = "str";
    return "";
  endfunction
endpackage
```
Output:
```
\_package: riscv (riscv::), id:23 test.sv:1:1: , endln:6:11, parent:unnamed, parID:22
  |vpiName:riscv
  |vpiFullName:riscv::
  |vpiDefName:riscv
  |vpiTaskFunc:
  \_function: (riscv::func), id:24, line:2:3, endln:5:14, parent:riscv::, parID:23
    |vpiName:func
    |vpiFullName:riscv::func
    |vpiVariables:
    \_string_var: (riscv::func::str), id:25, line:3:22, endln:3:25, parent:riscv::func, parID:24
      |vpiName:str
      |vpiFullName:riscv::func::str
      |vpiAutomatic:1
    |vpiVisibility:1
    |vpiReturn:
    \_string_var: , id:14, line:2:12, endln:2:18
    |vpiStmt:
    \_begin: (riscv::func), id:26, parent:riscv::func, parID:24
      |vpiFullName:riscv::func
      |vpiVariables:
      \_string_var: (riscv::func::str), id:27, line:3:22, endln:3:25, parent:riscv::func, parID:26
        |vpiName:str
        |vpiFullName:riscv::func::str
        |vpiAutomatic:1
      |vpiStmt:
      \_assign_stmt: , id:28, line:3:22, endln:3:33, parent:riscv::func, parID:26
        |vpiRhs:
        \_constant: , id:19, line:3:28, endln:3:33, parID:18
          |vpiDecompile:str
          |vpiSize:3
          |STRING:str
          |vpiConstType:6
        |vpiLhs:
        \_string_var: (riscv::func::str), id:29, line:3:22, endln:3:25, parID:28
          |vpiName:str
          |vpiFullName:riscv::func::str
          |vpiAutomatic:1
      |vpiStmt:
      \_return_stmt: , id:30, line:4:5, endln:4:11, parent:riscv::func, parID:26
        |vpiCondition:
        \_constant: , id:21, line:4:12, endln:4:14, parID:20
          |STRING:
          |vpiConstType:6
    |vpiInstance:
    \_package: riscv (riscv::), id:23 test.sv:1:1: , endln:6:11, parent:unnamed, parID:22
```

Note the three instances of `string_var` variable `str` with ids 25, 27 and 29. Issue is not specific to `string` only.

hs-apotell, 2022-06-26:
@alaindargelas Any pointers on what needs to happen here. I can work on this if you can provide some pointers.

alaindargelas, 2022-06-27:
@hs-apotell, this is not a duplication. They are all attached to different part of the tree, they are not in the same tree branch (That would be a duplication).

1) While create all variables in a given context (ie: a function), there should be a local symbol table that merge to a singleton all the variables that are the same. 

2) This is also a byproduct of the cloning and it is why we discussed the need for a comparison operator, if you need to know these objects represent the same thing.

alaindargelas, 2022-06-27:
BTW, this is not an issue for either Yosys or Verilator UHDM plugins. Both tools / their UHDM-readers have a symbol table that reconcile these.

hs-apotell, 2022-06-27:
> While create all variables in a given context (ie: a function), there should be a local symbol table that merge to a singleton all the variables that are the same.

Is this something that needs to be happen in Surelog or are you proposing we do this on our end?

> BTW, this is not an issue for either Yosys or Verilator UHDM plugins. Both tools / their UHDM-readers have a symbol table that reconcile these.

Shouldn't this reconciliation happen as part of Surelog? If this is already implemented in other Surelog clients, we would just be reinventing the wheel again. Making it part of Surelog has other advantages like less memory usage which can be significant for large designs.