chipsalliance/Surelog #2962: Investigate why regression script keeps stalling on Unix

hs-apotell, 2022-05-24:
Too many CI builds seem to be getting stuck on regression and never finishing.

hzeller, 2022-05-27:
Unclear why they hange, but I think I'd start with using `getrusage()` to determine the resource usage in case the script runs under a Posix system (#2802) (or use the regression tcl script for now on Posix systems).

That way, we eliminate the polling of all processes which could conceivably be rate-limited or even blocked in the github virtual machines running CI.

hs-apotell, 2022-05-27:
The regression script shouldn't be hogging the resource anymore. I introduced a longer sleep time a while back to release the CPU more aggresively and not compete with the running Surelog processes.

Switching to Tcl version is no more an option since it doesn't run RoundTrip process and there are other changes that were introduced to code generation which haven't been ported. The outputs of the two versions aren't the same anymore.

I am introducing a few changes to narrow down the root cause of the stall. For one, there is no indication of which two (or more) tests are still running since there is no output to declare a completed test. My hunch is two (or more) large tests are running simultaneously and that's what is hogging the resources. The currently running tests might help narrow down this case.

Also, to completely eliminate the regression script hogging the resources, I will further increase the sleep time to see if that helps.

hzeller, 2022-05-27:
I think the resource hogging is already greatly reduced, about 5% so about one core per 20 jobs run in parallel, so my only concern could be that the virtualization of the sandbox the CI is running in is limiting the access to poll the process list.

Runing two resource-intensive jobs in parallel also sounds like a potential plausible explanation for the hang.

hs-apotell, 2022-05-28:
Fixed by #2975

alaindargelas, 2022-05-31:
That problem still exist.

alaindargelas, 2022-05-31:
@hs-apotell, I think it is more likely to hang when the diff output is large.

alaindargelas, 2022-06-01:
Limiting the number of jobs to the number of cores with the python script makes it work all the time:
python3 scripts/regression.py run --jobs 8

hs-apotell, 2022-06-01:
The default is to use `multiprocessing.cpu_count` so it shouldn't be running any more jobs than the number of cores on the machine. I am not so sure about explicitly giving it on the command line should make any difference.

I would go with the _large diff output_ being more likely the cause. Question for you - Is the diff output in console window helpful? Generally speaking when the output logs are different they differ hugely and the limited diff in the console doesn't really say the whole story. You always end up comparing the logs. Is it acceptable to turn that spew off, at least for the sake of narrowing down the root cause for this issue? Alternatively, I would make a separate workflow that runs only the Posix jobs compare the results of main workflow with this new one (will have to add an option in script to turn off the spew from command line).

alaindargelas, 2022-06-01:
The partial diff output is useful for most of the changes I make like adding new property or fixing logic bugs.
It is also very handy to write the tkdiff command line at the end of the run for all the files that have diffs, instead to have to reconstruct that line manually.

hs-apotell, 2022-06-01:
Okay! If the problem persists then I might add a parallel workflow with diff disabled to confirm the hunch. I don't see a failure in last few builds on CI so will hold off on any immediate action.

hs-apotell, 2022-06-04:
@alaindargelas Looks like this is still happening and doesn't seem like large diffs are the problem since it is happening for successful builds as well. From whatever logs I could gather from past few builds, I have narrowed down the issue to two tests - Rsd & Google. Either of these two tests always show up as still running in the logs. I would like to temporarily disable these test to see if that fixes the problem. I would like to get at least 10 builds before toggling the tests again.

A thing I noticed about these two tests is that they both report high "large allocations" in the log. Do we know where this single 1.5GB+ allocation is coming from? Last failed build for Rsd test reports a single allocation of size 1505796096 bytes.

alaindargelas, 2022-06-04:
@hs-apotell, can you disable those tests in the CI only, but when I run the regression locally they still run?
I don't want to fly "blind", since those tests are important.

No idea what triggers the large allocation. @hzeller?

hs-apotell, 2022-06-04:
Yes, I can do that with the help of available environment variables in CI. I will post a PR with the change.

hs-apotell, 2022-06-06:
@alaindargelas @hzeller Found this in Rsd.log from a failed build. I am not familiar with kj libraries implementation. May be it means something to either of you.

```
tcmalloc: large alloc 1507762176 bytes == 0x11a434000 @  0x7f6cf3047680 0x7f6cf30672ec 0xa3402b 0xa3805e 0xa1b411 0x913052 0x92549d 0x926758 0x90cf23 0x736f02 0x68df79 0x46413d 0x418e95 0x4174ec 0x418565 0x7f6cf2add083 0x4171be

terminate called after throwing an instance of 'kj::ExceptionImpl'
  what():  kj/io.c++:53: failed: expected n >= minBytes; Premature EOF
stack: a7b7e6 a76a43 897869 418628 7f784a489082 41778d
```

hs-apotell, 2022-06-08:
Notes (attempt to narrow down root cause) -
* Turning off Rsd test fixed the posix+gcc but not posix+clang
* Tried running all tests sequentially - 3 consecutive builds succeeded
* Tried cherry-picking large tests and running only those sequentially - 3 consecutive failure, all posix+clang

Things to try -
* Check if disk is full in case of posix+clang
* Try shuffling the order in which tests are run

hs-apotell, 2022-06-08:
This was the last thing in the log and the test itself failed with segmentation fault. So, I would say this was more than a warning.

Also, on a machine with 7GB of RAM a 2GB allocation can be huge especially when running things in parallel. 

This is what I have found so far -
* This is not a disk space problem.
* Builds succeeds if I run all tests in parallel. Incidentally, the builds are no slower (1 vs. 2 cores doesn't make much difference especially given limited RAM). 
* I tried running small tests in parallel and large tests sequentially (any teat that takes longer than 80 seconds of CPU time is considered large). The builds still failed.

hs-apotell, 2022-06-08:
Forgot to mention, the issue doesn't happen in all builds. I going it in only 2 builds and both reported large allocations. Both builds were with Clang on Linux.

hzeller, 2022-06-08:
ok, so figuring out the source of the large allocation is a good next step. I'll see if I can set up something locally today.

hzeller, 2022-06-08:
Looking at the lifetime of the Rsd process, `capnp::MallocMessageBuilder::allocateSegment()` is the originator of a peak allocation of 2.8GiB (followed by `UHDM::FactoryT::Make()` with 1.38GiB).

Maybe UHDM could be written in chunks without it keeping the full serialized format in memory ?

hs-apotell, 2022-06-08:
Since you already have it setup, could you do a quick check to see what happens if either of these large allocations fail.

hs-apotell, 2022-06-08:
Are these single allocations? FacoryT allocations should be smaller chunks since it is responsible for creating different UHDM models.

hs-apotell, 2022-06-08:
> Maybe UHDM could be written in chunks without it keeping the full serialized format in memory ?

The bigger question is whether a machine with 7GB of RAM even a realistic environment for Surelog, especially if one is working with large designs like Rsd or Google. Memory-mapping is a good option but chunks seems overkill and unnecessary complication in my opinion.

hzeller, 2022-06-08:
Fair enough. Optimizations for unrealistically small machines should not be high on priority lists.

hs-apotell, 2022-06-08:
> Tried running all tests sequentially - 3 consecutive builds succeeded

Take that back. 2 of 5 builds failed even when running all tests sequentially. Unfortunately, with failed builds there is nothing to dig into because none of the logs are available from CI.

alaindargelas, 2022-07-04:
This is fixed

closed 2022-07-04 by alaindargelas