chipsalliance/Surelog #2966: Google test has a diff in CI (Can't explain yet)

alaindargelas, 2022-05-24:


alaindargelas, 2022-05-25:
The problem is with the python regression system, it does not clean correctly when running the Google test locally.
The tests/regression.tcl system does clean correctly and the test always gives the same answer.
With the python implementation, some cache files seem to persist and since this is a batch test (Many tests run in a single surelog batch test), the answer depend if you are running in a fresh checkout (CI) or a local run.

hs-apotell, 2022-05-25:
Regression script takes a snapshot of the individual test folder before running the test and restores the same state at the end of that test (by moving whatever extra files exists in the folder to the output folder). Unless there are files already existing in these test folders (may be from previous runs) there shouldn't be any stale state.

Either way, I have added a step in the regression script to just delete any slpp_all/slpp_unit folder in the test directories before running the tests. That should clean out any remnants from past failed runs.

hs-apotell, 2022-05-25:
The issue is that individual batch jobs write to this location and then the last one comes looking for them. Those will have to align for it to work. The script change is surely a patch. On top of that it ends up polluting the source test folders. 

Feel free to open a new issue. This surely needs a deeper investigation. For now, the script change would do.

alaindargelas, 2022-05-25:
Fixed by https://github.com/chipsalliance/Surelog/pull/2968

closed 2022-05-25 by alaindargelas