chipsalliance/Surelog #3007: Trailing spaces in multline macros fail to parse

hs-apotell, 2022-06-04:
Test:
```
`define MACRO(arg) \   
  call1(``arg``); \    
  call2(``arg``)   

module top();
  function f();
    `MACRO("here");
  endfunction
endmodule
```

Note: There are spaces at end of each of the line 1, 2, and 3 i.e. spaces after the '\'.

The parsing fails with the following error -
```
[SNT:PA0207] test1.sv:2:5: Syntax error: no viable alternative at input 'call1(',
call1(``arg``); #~@   #~@
     ^-- ...\slpp_all\work\_14695981039346656037\test1.sv:2:5:.
```

Thomasb81, 2022-06-04:
Hello

Page 676 Ieee1800-2017.1
[image]

I don't think trailer space after  backslash are legal. The standard specify : "If more than one line is necessary to specify the text, the nexline character shall be preceded by a blackslash ( \ )."

hs-apotell, 2022-06-04:
Appreciate the info.

closed 2022-06-04 by hs-apotell