chipsalliance/Surelog #3008: Include macro and macro_instance as part of non-elaborated tree

hs-apotell, 2022-06-04:
Macros get evaluated during preprocessing and are not available in the non-elaborated tree.

alaindargelas, 2022-06-04:
You need a completely new tree for that which is not a parsed tree. Do not modify the non-elaborated tree which is a parsed tree necessary as an intermediate step for elaboration.
What you need is a preprocessor AST. Surelog has that tree internally, you need to create a completely new schema in UHDM for preprocessor blobs (that reflects the preprocessor tree of Surelog). The starting point for studying would be the preprocessor grammar.

You cannot pass preprocessing directives to the parser. The parser grammar would explode. And SystemVerilog preprocessing is sufficiently difficult that it is most likely impossible to treat part of the parser-lexer. 
The UHDM non-elaborated and elaborated trees are parse-tree based.

alaindargelas, 2023-08-26:
@hs-apotell , I believe you addressed that.

closed 2023-08-26 by alaindargelas