chipsalliance/Surelog #3012: New vpiParent visitor makes debugging impossible

alaindargelas, 2022-06-06:
With the new explicit visit of vpiParent, it becomes impossible to dump objects while debugging, the traces are showing large amount of object parent data which is typically useless to debug real problems.
Please provide a way to disable visiting the parent by default when using UHDM::decompile and only dump the parent info with an explicit option.

Example of such a trace thet goes on and on about the parents.

```
decompile:
module: work@r5p_gpr (work@r5p_soc_de0_cv.soc.core.gpr), id:25678, file:hdl/rtl/core/r5p_core.sv, line:286:1, endln:307:3
|vpiParent:
\_module: work@r5p_core (work@r5p_soc_de0_cv.soc.core), id:25528, file:hdl/rtl/soc/r5p_soc_top.sv, line:88:1, endln:117:3
|vpiName:gpr
|vpiFullName:work@r5p_soc_de0_cv.soc.core.gpr
|vpiVariables:
\_logic_var: (work@r5p_soc_de0_cv.soc.core.gpr.wen), id:21013, line:44:18, endln:44:21
  |vpiParent:
  \_module: work@r5p_gpr (work@r5p_soc_de0_cv.soc.core.gpr), id:25678, file:hdl/rtl/core/r5p_core.sv, line:286:1, endln:307:3
  |vpiTypespec:
  \_logic_typespec: , id:21012, line:44:1, endln:44:6
  |vpiName:wen
  |vpiFullName:work@r5p_soc_de0_cv.soc.core.gpr.wen
  |vpiVisibility:1
|vpiVariables:
\_logic_var: (work@r5p_soc_de0_cv.soc.core.gpr.t_rs1), id:21027, line:45:18, endln:45:23
  |vpiParent:
  \_module: work@r5p_gpr (work@r5p_soc_de0_cv.soc.core.gpr), id:25678, file:hdl/rtl/core/r5p_core.sv, line:286:1, endln:307:3
  |vpiTypespec:
  \_logic_typespec: , id:21020, line:45:1, endln:45:17
    |vpiRange:
    \_range: , id:21014, line:45:7, endln:45:17
      |vpiLeftRange:
      \_constant: , id:21018, line:45:8, endln:45:12
        |vpiParent:
        \_range: , id:21014, line:45:7, endln:45:17
        |vpiDecompile:31
        |vpiSize:64
        |INT:31
        |vpiConstType:7
      |vpiRightRange:
      \_constant: , id:21019, line:45:15, endln:45:16
        |vpiParent:
        \_range: , id:21014, line:45:7, endln:45:17
        |vpiDecompile:0
        |vpiSize:64
        |UINT:0
        |vpiConstType:9
  |vpiName:t_rs1
  |vpiFullName:work@r5p_soc_de0_cv.soc.core.gpr.t_rs1
  |vpiVisibility:1
  |vpiRange:
  \_range: , id:21021, line:45:7, endln:45:17
    |vpiLeftRange:
    \_constant: , id:21025, line:45:8, endln:45:12
      |vpiParent:
      \_range: , id:21021, line:45:7, endln:45:17
      |vpiDecompile:31
      |vpiSize:64
      |INT:31
      |vpiConstType:7
    |vpiRightRange:
    \_constant: , id:21026, line:45:15, endln:45:16
      |vpiParent:
      \_range: , id:21021, line:45:7, endln:45:17
      |vpiDecompile:0
      |vpiSize:64
      |UINT:0
      |vpiConstType:9
|vpiVariables:
\_logic_var: (work@r5p_soc_de0_cv.soc.core.gpr.t_rs2), id:21041, line:46:18, endln:46:23
  |vpiParent:
  \_module: work@r5p_gpr (work@r5p_soc_de0_cv.soc.core.gpr), id:25678, file:hdl/rtl/core/r5p_core.sv, line:286:1, endln:307:3
[... 610 lines cut ...]
          |vpiRefFile:hdl/rtl/riscv/riscv_isa_pkg.sv
          |vpiRefLineNo:142
          |vpiRefColumnNo:3
          |vpiRefEndLineNo:142
          |vpiRefEndColumnNo:13
      |vpiName:ISA.spec.base.E
      |vpiActual:
      \_ref_obj: (ISA), id:15544, line:287:13, endln:287:16
        |vpiParent:
        \_hier_path: (ISA.spec.base.E), id:15543, line:287:13, endln:287:28
        |vpiName:ISA
        |vpiActual:
        \_operation: , id:14508, line:91:10, endln:91:13
          |vpiParent:
          \_param_assign: , id:14436, line:36:9, endln:36:47
          |vpiTypespec:
          \_struct_typespec: (riscv_isa_pkg::isa_t), id:14509, line:140:9, endln:140:15
            |vpiParent:
            \_operation: , id:14508, line:91:10, endln:91:13
            |vpiName:riscv_isa_pkg::isa_t
            |vpiInstance:
     ......
   

```

hs-apotell, 2022-06-06:
`vpiParent` visit is a shallow visit. I can add configuration options to the `visit_object` call but I am not sure what else I can do to improve debugging. This data in the logs has been very helpful in identifying problems.

Would configuration options to enable/disable the `vpiParent` visit help?

alaindargelas, 2022-06-06:
I use heavily UHDM::decompile(any* object) during gdb sessions.
If you can make that function only print the name and location of the parent and not go as far as it does, that will be perfect.

As you can see above, the traversal is not shallow.:

```
|vpiActual:
      \_ref_obj: (ISA), id:15544, line:287:13, endln:287:16
        |vpiParent:
        \_hier_path: (ISA.spec.base.E), id:15543, line:287:13, endln:287:28
        |vpiName:ISA
        |vpiActual:
        \_operation: , id:14508, line:91:10, endln:91:13
          |vpiParent:
          \_param_assign: , id:14436, line:36:9, endln:36:47
          |vpiTypespec:
          \_struct_typespec: (riscv_isa_pkg::isa_t), id:14509, line:140:9, endln:140:15
            |vpiParent:
            \_operation: , id:14508, line:91:10, endln:91:13
            |vpiName:riscv_isa_pkg::isa_t
            |vpiInstance:
```

alaindargelas, 2022-06-07:
Fixed by #3015

closed 2022-06-07 by alaindargelas