chipsalliance/Surelog #3067: Broken test CoresSweRVMP

alaindargelas, 2022-06-21:
cat build/regression/CoresSweRVMP/CoresSweRVMP.log 
[INF:CM0023] Creating log file ../../../build/regression/CoresSweRVMP/slpp_unit/surelog.log.

[WRN:CM0010] Command line argument "-Wno-UNOPTFLAT" ignored.

[INF:CM0024] Executing with 16 processes and 0 threads.

[INF:CM0020] Separate compilation-unit mode is on.   << Should not happen -fileunit used by mistake by regression script

[INF:PP0122] Preprocessing source file "../../UVM/1800.2-2017-1.0/src/uvm_pkg.sv".

[INF:PP0122] Preprocessing source file "snapshots/default/common_defines.vh".

[INF:PP0122] Preprocessing source file "design/include/swerv_types.sv".

[ERR:PP0102] design/include/swerv_types.sv:58:31: Unknown macro "RV_LSU_NUM_NBLOAD_WIDTH".   << This error cause by above wrong fileunit

[ERR:PP0102] design/include/swerv_types.sv:76:31: Unknown macro "RV_BTB_ADDR_HI".

[ERR:PP0102] design/include/swerv_types.sv:76:47: Unknown macro "RV_BTB_ADDR_LO".

hs-apotell, 2022-06-21:
I just ran both the tests locally and I am not seeing that 'Separate compilation-unit mode' statement in my logs. Both logs attached.

[CoresSweRVMP.log](https://github.com/chipsalliance/Surelog/files/8946593/CoresSweRVMP.log)
[CoresSweRV.log](https://github.com/chipsalliance/Surelog/files/8946595/CoresSweRV.log)

And both the tests are giving the same results. In fact, the log for CoresSweRVMP test in repo is different from CoresSweRV.
Here's the local run -
```
Starting Surelog Regression Tests @ 2022-06-20 23:51:00.608515
Environment:
      command-line: scripts/regression.py run --filters CoresSweRVMP CoresSweRV
   multi-threading: None
  multi-processing: None
          max-jobs: 2
         max-tests: 593
 blacklisted-tests: 7
    filtered-tests: 2

Running 2 tests ...
Running CoresSweRV ...
Running CoresSweRVMP ...
... CoresSweRV Completed.
... CoresSweRVMP Completed.

Results:
  +--------------+--------+-------+--------+--------+----------+------+----------+---------+---------+-----------+
  | TESTNAME     | STATUS | FATAL | SYNTAX |  ERROR |  WARNING | NOTE | CPU-TIME | VTL-MEM | PHY-MEM | ROUNDTRIP |
  +--------------+--------+-------+--------+--------+----------+------+----------+---------+---------+-----------+
  | CoresSweRV   | PASS   |     0 |      0 |      4 |      182 |   15 |    19.16 |    1493 |    1268 |       0/0 |
  | CoresSweRVMP | DIFF   |     0 |      0 | 4 (-1) | 182 (65) |   15 |    19.10 |    1493 |    1269 |       0/0 |
  +--------------+--------+-------+--------+--------+----------+------+----------+---------+---------+-----------+

Summary:
  +----------------+---------------------+
  | PASS           | 1                   |
  | DIFF           | 1                   |
  | FAIL           | 0                   |
  | FAILDUMP       | 0                   |
  | SEGFLT         | 0                   |
  | NOGOLD         | 0                   |
  | TOOLFAIL       | 0                   |
  |                |                     |
  | MAX CPU TIME   | 19.16 (CoresSweRV)  |
  | TOTAL CPU TIME | 38.26               |
  | MAX WALL TIME  | 64 (CoresSweRVMP)   |
  | MAX MEMORY     | 1269 (CoresSweRVMP) |
  +----------------+---------------------+

Surelog Regression Test Completed @ 2022-06-20 23:52:42.854754 in 102 seconds
```

hs-apotell, 2022-06-21:
> Should not happen -fileunit used by mistake by regression script

Regression script doesn't do anything whatsoever with -fileunit flag explicitly.

closed 2022-06-22 by alaindargelas