chipsalliance/Surelog #3081: Latest Antlr broken on Clang

alaindargelas, 2022-07-01:
https://github.com/chipsalliance/Surelog/runs/7143585776?check_suite_focus=true

@hzeller, @mithro, you might want to direct to the proper channels.

QuantamHD, 2022-07-01:
@alaindargelas what commit was antlr at?

hzeller, 2022-07-01:
Looks like the third_party/antlr4 submodule currently points to https://github.com/antlr/antlr4/tree/ff2acb19f399c882ab038bd944af1a3797b5c636

hzeller, 2022-07-01:
Is there a reason we're not following master in Antlr ( https://github.com/antlr/antlr4/tree/master ) ?
Looks like the sub-module follows the `dev` repo.

QuantamHD, 2022-07-01:
dev is the nightly equivalent for ANTLR. The stable releases of ANTLR go to master, and are pretty rare. It's up to @alaindargelas if you'd like to onboard new changes sooner. Catching issues like this early on in the process will make upgrading easier.

hs-apotell, 2022-07-01:
> No reason, I think @hs-apotell set it up to track "dev". Please change to master

There was a reason. At the time Surelog transitioned to using the pristine antlr, the required changes had not been merged to master yet - they were available only in dev branch. That has changed since then. There is still this PR https://github.com/antlr/antlr4/pull/3678 that is pending which would make clang possible for Windows as well. But for the current supported platforms and compilers for Surelog, we can safely transition to using master.

P.S. If anyone is motivated enough, appreciate some push to get the above mentioned PR approved. It's been pending since April 25th.

alaindargelas, 2022-07-02:
@QuantamHD, we can keep it to dev and bug them to fix as we catch these. I'm ok.

KvanTTT, 2022-07-02:
Hi. It looks like the failures related to this commit: https://github.com/antlr/antlr4/commit/67baf7b9487a18a8b25daaf16d44e545ee24cf1d#diff-eb4b31b28c26cdd50a34423808134155bb8e8836b348e4a85698c2ff330412fa But our tests are fine. Could anybody suggest a fix?

QuantamHD, 2022-07-02:
@KvanTTT 

```
/home/runner/work/Surelog/Surelog/third_party/antlr4/runtime/Cpp/runtime/src/atn/SemanticContext.h:95:43: warning: abstract class is marked 'final' [-Wabstract-final-class]
  class ANTLR4CPP_PUBLIC SemanticContext::Empty final : public SemanticContext{
                                          ^
/home/runner/work/Surelog/Surelog/third_party/antlr4/runtime/Cpp/runtime/src/atn/SemanticContext.h:40:18: note: unimplemented pure virtual method 'eval' in 'Empty'
    virtual bool eval(Recognizer *parser, RuleContext *parserCallStack) const = 0;
                 ^
/home/runner/work/Surelog/Surelog/third_party/antlr4/runtime/Cpp/runtime/src/atn/SemanticContext.h:62:20: note: unimplemented pure virtual method 'hashCode' in 'Empty'
    virtual size_t hashCode() const = 0;
                   ^
/home/runner/work/Surelog/Surelog/third_party/antlr4/runtime/Cpp/runtime/src/atn/SemanticContext.h:64:18: note: unimplemented pure virtual method 'equals' in 'Empty'
    virtual bool equals(const SemanticContext &other) const = 0;
                 ^
/home/runner/work/Surelog/Surelog/third_party/antlr4/runtime/Cpp/runtime/src/atn/SemanticContext.h:66:25: note: unimplemented pure virtual method 'toString' in 'Empty'
    virtual std::string toString() const = 0;
                        ^
```

It looks like this is related to the empty abstract class being marked final. I'm not sure if that's intentional. My C++ isn't strong enough to say

KvanTTT, 2022-07-02:
Yes, it looks like that. But neither Visual Studio nor clang don't report such errors.

QuantamHD, 2022-07-02:
Could be down to the clang version?

KvanTTT, 2022-07-02:
I have clang 13, but not sure it depends on a clang version, maybe it depends on some compiler flags.

alaindargelas, 2022-07-06:
@QuantamHD shall we point to main since it looks like dev is going to stay broken for a while?

hzeller, 2022-07-06:
if this hasn't resolved yet, I'll have a look today.

hzeller, 2022-07-06:
It looks like only clang10 and clang 11 complain about this situation.
Created PR https://github.com/antlr/antlr4/pull/3776

parrt, 2022-07-06:
Just merged @hzeller fix.

PR #3088 (2022-07-07): Update antlr submodule.
Fixes #3081
commit d8e7719a (2022-07-07): Update antlr submodule.

Fixes #3081
closed 2022-07-07 by alaindargelas