chipsalliance/Surelog #3115: C++ errors

yurivict, 2022-07-22:
Clang-13 finds these errors:

```
In file included from /disk-samsung/freebsd-ports/cad/surelog/work/Surelog-0.0-3950-g88ae53316/src/Cache/ParseCache.cpp:23:
In file included from /disk-samsung/freebsd-ports/cad/surelog/work/Surelog-0.0-3950-g88ae53316/src/Cache/ParseCache.h:27:
In file included from /disk-samsung/freebsd-ports/cad/surelog/work/Surelog-0.0-3950-g88ae53316/src/Cache/Cache.h:27:
In file included from /disk-samsung/freebsd-ports/cad/surelog/work/Surelog-0.0-3950-g88ae53316/src/Design/FileContent.h:31:
/disk-samsung/freebsd-ports/cad/surelog/work/Surelog-0.0-3950-g88ae53316/src/Design/DesignComponent.h:142:32: error: no type named 'import' in namespace 'UHDM'; did you mean 'port'?
  void addImportedSymbol(UHDM::import* i) { m_imported_symbols.push_back(i); }
                         ~~~~~~^~~~~~
                               port
/disk-samsung/freebsd-ports/cad/surelog/work/.build/third_party/UHDM/generated/uhdm/uhdm_forward_decl.h:178:7: note: 'port' declared here
class port;
      ^
In file included from /disk-samsung/freebsd-ports/cad/surelog/work/Surelog-0.0-3950-g88ae53316/src/Cache/ParseCache.cpp:23:
In file included from /disk-samsung/freebsd-ports/cad/surelog/work/Surelog-0.0-3950-g88ae53316/src/Cache/ParseCache.h:27:
In file included from /disk-samsung/freebsd-ports/cad/surelog/work/Surelog-0.0-3950-g88ae53316/src/Cache/Cache.h:27:
In file included from /disk-samsung/freebsd-ports/cad/surelog/work/Surelog-0.0-3950-g88ae53316/src/Design/FileContent.h:31:
/disk-samsung/freebsd-ports/cad/surelog/work/Surelog-0.0-3950-g88ae53316/src/Design/DesignComponent.h:143:34: error: expected expression
  const std::vector<UHDM::import*>& getImportedSymbols() const {
                                 ^
/disk-samsung/freebsd-ports/cad/surelog/work/Surelog-0.0-3950-g88ae53316/src/Design/DesignComponent.h:143:27: error: no member named 'import' in namespace 'UHDM'
  const std::vector<UHDM::import*>& getImportedSymbols() const {
                    ~~~~~~^
/disk-samsung/freebsd-ports/cad/surelog/work/Surelog-0.0-3950-g88ae53316/src/Design/DesignComponent.h:188:28: error: expected expression
  std::vector<UHDM::import*> m_imported_symbols;
                           ^
/disk-samsung/freebsd-ports/cad/surelog/work/Surelog-0.0-3950-g88ae53316/src/Design/DesignComponent.h:188:21: error: no member named 'import' in namespace 'UHDM'
  std::vector<UHDM::import*> m_imported_symbols;
              ~~~~~~^
/disk-samsung/freebsd-ports/cad/surelog/work/Surelog-0.0-3950-g88ae53316/src/Design/DesignComponent.h:63:3: error: no matching member function for call to 'AsOfTypeRecurse'
  SURELOG_IMPLEMENT_RTTI(DesignComponent, ValuedComponentI)
  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/disk-samsung/freebsd-ports/cad/surelog/work/Surelog-0.0-3950-g88ae53316/src/Common/RTTI.h:13:3: note: expanded from macro 'SURELOG_IMPLEMENT_RTTI'
  UHDM_IMPLEMENT_RTTI(classType, baseType)
  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/disk-samsung/freebsd-ports/cad/surelog/work/.build/third_party/UHDM/generated/uhdm/RTTI.h:186:85: note: expanded from macro 'UHDM_IMPLEMENT_RTTI'
    { return (tid == thistype_t::kTypeId) ? static_cast<void *>(this) : basetype_t::AsOfTypeRecurse(tid); }       \
                                                                        ~~~~~~~~~~~~^~~~~~~~~~~~~~~
/disk-samsung/freebsd-ports/cad/surelog/work/Surelog-0.0-3950-g88ae53316/src/Design/ValuedComponentI.h:40:3: note: candidate function not viable: no known conversion from 'SURELOG::DesignComponent' to 'SURELOG::ValuedComponentI' for object argument
  SURELOG_IMPLEMENT_RTTI(ValuedComponentI, RTTI)
  ^
/disk-samsung/freebsd-ports/cad/surelog/work/Surelog-0.0-3950-g88ae53316/src/Common/RTTI.h:13:3: note: expanded from macro 'SURELOG_IMPLEMENT_RTTI'
  UHDM_IMPLEMENT_RTTI(classType, baseType)
  ^
/disk-samsung/freebsd-ports/cad/surelog/work/.build/third_party/UHDM/generated/uhdm/RTTI.h:185:18: note: expanded from macro 'UHDM_IMPLEMENT_RTTI'
    inline void *AsOfTypeRecurse(typeid_t tid)                                                                    \
                 ^
/disk-samsung/freebsd-ports/cad/surelog/work/Surelog-0.0-3950-g88ae53316/src/Design/ValuedComponentI.h:40:3: note: candidate function not viable: no known conversion from 'SURELOG::DesignComponent' to 'const SURELOG::ValuedComponentI' for object argument
/disk-samsung/freebsd-ports/cad/surelog/work/Surelog-0.0-3950-g88ae53316/src/Common/RTTI.h:13:3: note: expanded from macro 'SURELOG_IMPLEMENT_RTTI'
  UHDM_IMPLEMENT_RTTI(classType, baseType)
  ^
/disk-samsung/freebsd-ports/cad/surelog/work/.build/third_party/UHDM/generated/uhdm/RTTI.h:187:24: note: expanded from macro 'UHDM_IMPLEMENT_RTTI'
    inline const void *AsOfTypeRecurse(typeid_t tid) const                                                        \
                       ^
In file included from /disk-samsung/freebsd-ports/cad/surelog/work/Surelog-0.0-3950-g88ae53316/src/Cache/ParseCache.cpp:23:
In file included from /disk-samsung/freebsd-ports/cad/surelog/work/Surelog-0.0-3950-g88ae53316/src/Cache/ParseCache.h:27:
In file included from /disk-samsung/freebsd-ports/cad/surelog/work/Surelog-0.0-3950-g88ae53316/src/Cache/Cache.h:27:
In file included from /disk-samsung/freebsd-ports/cad/surelog/work/Surelog-0.0-3950-g88ae53316/src/Design/FileContent.h:31:
/disk-samsung/freebsd-ports/cad/surelog/work/Surelog-0.0-3950-g88ae53316/src/Design/DesignComponent.h:63:3: error: no matching member function for call to 'AsOfTypeRecurse'
  SURELOG_IMPLEMENT_RTTI(DesignComponent, ValuedComponentI)
  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/disk-samsung/freebsd-ports/cad/surelog/work/Surelog-0.0-3950-g88ae53316/src/Common/RTTI.h:13:3: note: expanded from macro 'SURELOG_IMPLEMENT_RTTI'
  UHDM_IMPLEMENT_RTTI(classType, baseType)
  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/disk-samsung/freebsd-ports/cad/surelog/work/.build/third_party/UHDM/generated/uhdm/RTTI.h:188:91: note: expanded from macro 'UHDM_IMPLEMENT_RTTI'
    { return (tid == thistype_t::kTypeId) ? static_cast<const void *>(this) : basetype_t::AsOfTypeRecurse(tid); } \
                                                                              ~~~~~~~~~~~~^~~~~~~~~~~~~~~
/disk-samsung/freebsd-ports/cad/surelog/work/Surelog-0.0-3950-g88ae53316/src/Design/ValuedComponentI.h:40:3: note: candidate function not viable: no known conversion from 'const SURELOG::DesignComponent' to 'SURELOG::ValuedComponentI' for object argument
  SURELOG_IMPLEMENT_RTTI(ValuedComponentI, RTTI)
  ^
/disk-samsung/freebsd-ports/cad/surelog/work/Surelog-0.0-3950-g88ae53316/src/Common/RTTI.h:13:3: note: expanded from macro 'SURELOG_IMPLEMENT_RTTI'
  UHDM_IMPLEMENT_RTTI(classType, baseType)
  ^
/disk-samsung/freebsd-ports/cad/surelog/work/.build/third_party/UHDM/generated/uhdm/RTTI.h:185:18: note: expanded from macro 'UHDM_IMPLEMENT_RTTI'
    inline void *AsOfTypeRecurse(typeid_t tid)                                                                    \
                 ^
/disk-samsung/freebsd-ports/cad/surelog/work/Surelog-0.0-3950-g88ae53316/src/Design/ValuedComponentI.h:40:3: note: candidate function not viable: no known conversion from 'const SURELOG::DesignComponent' to 'const SURELOG::ValuedComponentI' for object argument
/disk-samsung/freebsd-ports/cad/surelog/work/Surelog-0.0-3950-g88ae53316/src/Common/RTTI.h:13:3: note: expanded from macro 'SURELOG_IMPLEMENT_RTTI'
  UHDM_IMPLEMENT_RTTI(classType, baseType)
  ^
```

FreeBSD 13.1

alaindargelas, 2022-07-23:
Ooops, is "import" a new C++ keyword?

yurivict, 2022-07-23:
I think compiler is trying to suggest some variable name.