chipsalliance/Surelog #3132: Evaluating function in module

aolofsson, 2022-08-02:
This code simulats ok in icarus, but I get the following message in Surelog. This is nastier than my usual code, so I may be missing something in the standard here.

The snippet is inside a module...

```sv
 function automatic [7:0] offset;
      input [7:0] i;
      input [63:0] vector;
      integer 	    ii;
      begin
	 offset = 0;
	 for(ii=0; ii<i; ii=ii+1)
	   offset = offset + vector[ii*8+:8];
      end
   endfunction

 for(i=0;i<SECTIONS;i=i+1)
     begin: ila_section
	// figure out how many bits came before
	localparam [7:0] START = offset(i,NIO)
	localparam [7:0] N = NPIO[8*i+:8]
   

```

**Surelog Error:**

ERR:CP0337] <REDACTED> Undefined user task/function "offset".

alaindargelas, 2022-08-03:
@aolofsson, same comment, I'm trying to reverse engineer a workable testcase from the snippet, can you provide a complete snippet that compiles with all variables/params declared, that is encompassed with module/endmodule?

alaindargelas, 2022-08-04:
@aolofsson how about this one? Is is fixed too?

aolofsson, 2022-08-04:
I found a way around that construct so the code is "gone". I recommend closing it.

alaindargelas, 2022-08-04:
OK, thanks

closed 2022-08-04 by alaindargelas