chipsalliance/Surelog #3133: Parameter poartial select

aolofsson, 2022-08-02:
The following code has been verified to work in icarus. Is it possible that there is some issue with extracting 
vector fields from a packed parameter? Should be legal right?

```verilog
parameter [63:0]  N          = 64'h0807060504030201,  

genvar i;
   generate
      for(i=0;i<SECTIONS;i=i+1)
	begin: i0
           la_ioside #(.N(N[8*i+:8]),
```

**SURELOG ERROR:**

NTE:EL0531] ../rtl/la_iosection.v:45:13: Negative value in instance "work@la_ioside.ila_iosection[0].i0"
             text:     output [N-1:0]     z, // output to core
             value: INT:-1.

mithro, 2022-08-03:
FYI - @kgugala / @hzeller

mithro, 2022-08-03:
@alaindargelas - We should make this a priority to fix given that @aolofsson and @siliconcompiler are a pretty big / important user of Surelog.

mithro, 2022-08-03:
@aolofsson - We should land an example of this into sv-tests.

alaindargelas, 2022-08-03:
@aolofsson, is it possible to get a complete snipet that compiles. For instance the value of SECTION plays a big role in the answer and it's not in the testcase.
Ideally, can you provide a module .... endmodule self contained testcase that replicate the error?

aolofsson, 2022-08-03:
How about this? The concern is the negative index below.

```verilog
module param_testcase
  #(parameter [4:0]  SECTIONS  =  1,
    parameter [7:0]  NSIDE     =  8'h8,
    parameter [63:0] N         =  64'h0404,
    parameter [63:0] NSTART    =  64'h00
     )
   (
    input [NSIDE-1:0] a // input from core
    );

   genvar i;
   generate
      for(i=0;i<SECTIONS;i=i+1)
	begin: ila_ioside
           la_iosection #(.N(N[8*i+:8]))
	   i0(.a (a[NSTART[i*8+:8]+:N[i*8+:8]]));
	end
   endgenerate
endmodule // param_testcase

```

```verilog
module la_iosection
  #(parameter [7:0]  N         = 0     // total pads
    )
   (input [N-1:0]      a
    );

   initial
     begin
	$display("%d", N);
     end

endmodule // la_ioside
```

```bash
$ surelog param_testcase.v -y . -parse -top param_testcase +libext+.sv+.v
```

[INF:CM0023] Creating log file ./slpp_all/surelog.log.

[WRN:PA0205] param_testcase.v:1: No timescale set for "param_testcase".

[WRN:PA0205] la_iosection.v:1: No timescale set for "la_iosection".

[INF:CP0300] Compilation...

[INF:CP0303] la_iosection.v:1: Compile module "work@la_iosection".

[INF:CP0303] param_testcase.v:1: Compile module "work@param_testcase".

[INF:CP0302] builtin.sv:4: Compile class "work@mailbox".

[INF:CP0302] /builtin.sv:33: Compile class "work@process".

[INF:CP0302] builtin.sv:58: Compile class "work@semaphore".

[INF:EL0526] Design Elaboration...

[INF:CP0335] param_testcase.v:13: Compile generate block "work@param_testcase.ila_ioside[0]".

[NTE:EL0503] param_testcase.v:1: Top level module "work@param_testcase".

[NTE:EL0531] /la_iosection.v:4:12: Negative value in instance "work@param_testcase.ila_ioside[0].i0"
             text:    (input [N-1:0]      a
             value: INT:-1.

[NTE:EL0508] Nb Top level modules: 1.

[NTE:EL0509] Max instance depth: 3.

[NTE:EL0510] Nb instances: 2.

[NTE:EL0511] Nb leaf instances: 0.

[INF:UH0706] Creating UHDM Model...

[INF:UH0708] Writing UHDM DB: ./slpp_all//surelog.uhdm...

```

alaindargelas, 2022-08-03:
Perfect! Thanks.

alaindargelas, 2022-08-03:
@aolofsson, 
I created the same setup with 2 files. I can't see the error. 
It looks like you are using a Surelog version that is several months old. Can you update to the latest?

```
[INF:CM0023] Creating log file ./slpp_all/surelog.log.

LIB:  work
FILE: tests/ScratchPad.sv
n<> u<0> t<_INVALID_> f<0> l<0:0>
n<> u<1> t<Null_rule> p<263> s<262> l<1:1> el<1:0>
n<> u<2> t<Module_keyword> p<116> s<3> l<1:1> el<1:7>
n<param_testcase> u<3> t<StringConst> p<116> s<92> l<1:8> el<1:22>
n<4> u<4> t<IntConst> p<5> l<2:16> el<2:17>
n<> u<5> t<Primary_literal> p<6> c<4> l<2:16> el<2:17>
n<> u<6> t<Constant_primary> p<7> c<5> l<2:16> el<2:17>
n<> u<7> t<Constant_expression> p<12> c<6> s<11> l<2:16> el<2:17>
n<0> u<8> t<IntConst> p<9> l<2:18> el<2:19>
n<> u<9> t<Primary_literal> p<10> c<8> l<2:18> el<2:19>
n<> u<10> t<Constant_primary> p<11> c<9> l<2:18> el<2:19>
n<> u<11> t<Constant_expression> p<12> c<10> l<2:18> el<2:19>
n<> u<12> t<Constant_range> p<13> c<7> l<2:16> el<2:19>
n<> u<13> t<Packed_dimension> p<14> c<12> l<2:15> el<2:20>
n<> u<14> t<Data_type_or_implicit> p<24> c<13> s<23> l<2:15> el<2:20>
n<SECTIONS> u<15> t<StringConst> p<22> s<21> l<2:22> el<2:30>
n<1> u<16> t<IntConst> p<17> l<2:35> el<2:36>
n<> u<17> t<Primary_literal> p<18> c<16> l<2:35> el<2:36>
n<> u<18> t<Constant_primary> p<19> c<17> l<2:35> el<2:36>
n<> u<19> t<Constant_expression> p<20> c<18> l<2:35> el<2:36>
n<> u<20> t<Constant_mintypmax_expression> p<21> c<19> l<2:35> el<2:36>
n<> u<21> t<Constant_param_expression> p<22> c<20> l<2:35> el<2:36>
n<> u<22> t<Param_assignment> p<23> c<15> l<2:22> el<2:36>
n<> u<23> t<List_of_param_assignments> p<24> c<22> l<2:22> el<2:36>
n<> u<24> t<Parameter_declaration> p<25> c<14> l<2:5> el<2:36>
n<> u<25> t<Parameter_port_declaration> p<92> c<24> s<47> l<2:5> el<2:36>
n<7> u<26> t<IntConst> p<27> l<3:16> el<3:17>
n<> u<27> t<Primary_literal> p<28> c<26> l<3:16> el<3:17>
n<> u<28> t<Constant_primary> p<29> c<27> l<3:16> el<3:17>
n<> u<29> t<Constant_expression> p<34> c<28> s<33> l<3:16> el<3:17>
n<0> u<30> t<IntConst> p<31> l<3:18> el<3:19>
n<> u<31> t<Primary_literal> p<32> c<30> l<3:18> el<3:19>
n<> u<32> t<Constant_primary> p<33> c<31> l<3:18> el<3:19>
n<> u<33> t<Constant_expression> p<34> c<32> l<3:18> el<3:19>
n<> u<34> t<Constant_range> p<35> c<29> l<3:16> el<3:19>
n<> u<35> t<Packed_dimension> p<36> c<34> l<3:15> el<3:20>
n<> u<36> t<Data_type_or_implicit> p<46> c<35> s<45> l<3:15> el<3:20>
n<NSIDE> u<37> t<StringConst> p<44> s<43> l<3:22> el<3:27>
n<8'h8> u<38> t<IntConst> p<39> l<3:35> el<3:39>
n<> u<39> t<Primary_literal> p<40> c<38> l<3:35> el<3:39>
n<> u<40> t<Constant_primary> p<41> c<39> l<3:35> el<3:39>
n<> u<41> t<Constant_expression> p<42> c<40> l<3:35> el<3:39>
n<> u<42> t<Constant_mintypmax_expression> p<43> c<41> l<3:35> el<3:39>
n<> u<43> t<Constant_param_expression> p<44> c<42> l<3:35> el<3:39>
n<> u<44> t<Param_assignment> p<45> c<37> l<3:22> el<3:39>
n<> u<45> t<List_of_param_assignments> p<46> c<44> l<3:22> el<3:39>
n<> u<46> t<Parameter_declaration> p<47> c<36> l<3:5> el<3:39>
n<> u<47> t<Parameter_port_declaration> p<92> c<46> s<69> l<3:5> el<3:39>
n<63> u<48> t<IntConst> p<49> l<4:16> el<4:18>
n<> u<49> t<Primary_literal> p<50> c<48> l<4:16> el<4:18>
n<> u<50> t<Constant_primary> p<51> c<49> l<4:16> el<4:18>
n<> u<51> t<Constant_expression> p<56> c<50> s<55> l<4:16> el<4:18>
n<0> u<52> t<IntConst> p<53> l<4:19> el<4:20>
n<> u<53> t<Primary_literal> p<54> c<52> l<4:19> el<4:20>
n<> u<54> t<Constant_primary> p<55> c<53> l<4:19> el<4:20>
n<> u<55> t<Constant_expression> p<56> c<54> l<4:19> el<4:20>
n<> u<56> t<Constant_range> p<57> c<51> l<4:16> el<4:20>
n<> u<57> t<Packed_dimension> p<58> c<56> l<4:15> el<4:21>
n<> u<58> t<Data_type_or_implicit> p<68> c<57> s<67> l<4:15> el<4:21>
n<N> u<59> t<StringConst> p<66> s<65> l<4:22> el<4:23>
n<64'h0404> u<60> t<IntConst> p<61> l<4:35> el<4:43>
[... 1050 lines cut ...]
            |vpiParent:
            \_initial: , id:198, line:7:4, endln:10:9
            |vpiFullName:work@param_testcase.ila_ioside[0].i0
            |vpiStmt:
            \_sys_func_call: ($display), id:200, line:9:2, endln:9:19
              |vpiParent:
              \_begin: (work@param_testcase.ila_ioside[0].i0), id:199, line:8:6, endln:10:9
              |vpiArgument:
              \_constant: , id:11, line:9:11, endln:9:15
              |vpiArgument:
              \_ref_obj: (work@param_testcase.ila_ioside[0].i0.N), id:201, line:9:17, endln:9:18
                |vpiParent:
                \_sys_func_call: ($display), id:200, line:9:2, endln:9:19
                |vpiName:N
                |vpiFullName:work@param_testcase.ila_ioside[0].i0.N
                |vpiActual:
                \_parameter: (work@param_testcase.ila_ioside[0].i0.N), id:171, line:2:22, endln:2:23
              |vpiName:$display
===================
[  FATAL] : 0
[ SYNTAX] : 0
[  ERROR] : 0
[WARNING] : 2
[   NOTE] : 9
```

aolofsson, 2022-08-04:
Great news, updating to version 1.35 solved the problem! Bad news, I don't get to feel good about finding an issue that helps the surelog project.:smile:

alaindargelas, 2022-08-04:
You'll get your chance :-)

closed 2022-08-04 by alaindargelas