chipsalliance/Surelog #3134: Running PreProcMacro with valgrind reveals a read from non-allocated memory

hzeller, 2022-08-04:
To reproduce
```bash
(make debug ; cd tests/PreProcMacro ; valgrind ../../dbuild/bin/surelog $(cat PreProcMacro.sl))
```

It looks like that in [SV3_1aPpTreeShapeListener.cpp:1163](https://github.com/chipsalliance/Surelog/blob/932777626024448702ddaa422ffef24758f3cc15/src/SourceCompile/SV3_1aPpTreeShapeListener.cpp#L1163) maybe an element is accessed in the stack that does not exist ?

Should stack-emptiness be checked ?

```patch
--- a/src/SourceCompile/SV3_1aPpTreeShapeListener.cpp
+++ b/src/SourceCompile/SV3_1aPpTreeShapeListener.cpp
@@ -1158,7 +1158,7 @@ void SV3_1aPpTreeShapeListener::enterEndif_directive(
     if (ctx->One_line_comment()) {
       addLineFiller(ctx);
     }
-    while (unroll) {
+    while (unroll && !stack.empty()) {
       PreprocessFile::IfElseItem &item = stack.back();
       switch (item.m_type) {
         case PreprocessFile::IfElseItem::IFDEF:
```

alaindargelas, 2022-08-04:
Right! I fixed in the wrong place. There is no core dump anymore, but there is still a UMR.
I'll fix this.

alaindargelas, 2022-08-04:
Fixed by #3136

closed 2022-08-04 by alaindargelas