chipsalliance/Surelog #3142: "Vendor mode" for command line arguments

udif, 2022-08-09:
I see that the current command line argument flags are geared towards accepting parameters from a specific large EDA vendor, whose name begins with an "S" 😃 
Just to name a few:
```
+incdir+<path>
+define+<name>=<value>
+liborder
+librescan
```
etc.

Given that another EDA vendor (whose name begins with a "C") command line arguments use very similar, but slightly different names:
```
-y <path>
-incdir <path>
-define <name>=<value>
-liborder
-librescan
```
Would you be accepting a patch that would allow something like:
`surelog <surelog-specific-arguments>  --vendorname -f <vendorname-args-file>`
I would not suggest accepting both, because from what I've seen, some of the single letter generic command line options may mean different things with different tools (for example `-I`).

Can we use real vendor names, or are they going to send their legal hound dogs at us? 😄

alaindargelas, 2022-08-11:
We cannot use vendor name. 
-y works for all vendors.
I suggest accepting both types, I'll accept the PR. For -I and other conflicting args maybe having a -style M/C/S placed prior the arguments to help disambiguate would be acceptable.

udif, 2022-08-11:
I can prepare a pull request.

Another option I thought about was to have an option like:
`-remap oldargname newargname`
with minimal templating that is powerful enough to translate (for example) `+define+A=3` into `-define A=3`
By doing this you are distancing yourself from any vendor name or arguments.
It's up to the user to prepare his own translation file that can be called with `-f translate_my_vendor_args_to_surelog`
Unfortunately, It's not only a matter of regex, but you need a sane syntax that will allow you map 2 consecutive arguments into 1 or vice versa.
If you think this is an overkill, I'll drop it and stick to the '-style' option above.

hzeller, 2022-08-11:
I like the remap idea.

udif, 2022-08-12:
I created a preliminary pull request for review, #3149 .

I've decided not to implement my 'remap' idea because I realized that the vendor tool in question is not replacing the old flag, but rather adding new synonyms for the existing flags, while still supporting the old ones.
If I would have remapped the existing flag, I would run into problems with legacy build scripts that uses both styles. By adding an explicit style flag, I can check both notations when it is set.

The whole point I'm working on this pull request is to let me take a legacy build flow that contains many small file lists, defines, include paths and library references, and run it through Surelog without needing to change it, other than adding a single '-style' flag on the command line.

udif, 2022-08-15:
Well, I opened the pull request **for reviews**, and you went ahead and merged it...
In fact, **it had several issues (only if you use the new features - if you ignore the -style option it has no effect)**:
* It forces you to fill the Surelog code with literally hundreds of commands if you really want to be able to 100% parse a vendor's command line.
* You may not want to copy all those commands as-is for the same reasons we don't name the styles (copyright? trademark?).
* The original implementation also had bugs.

So I went back to the drawing board, threw away the original implementation, and came back to the remap idea.
I've added 3 commands:

**`-cmd_ign <flag> <argcount>`**
This will add <flag> to the list of dropped commands, with <argcount> arguments.
Examples:
`-cmd_ign -xyz 1`
Will drop any command named `-xyz` found later on the command line ,and 1 argument following it.
`-cmd_ign +abc 0`
Will drop any command named `+abc` found later on the command line. (it has no arguments, so only the command will be dropped).

**`-cmd_ren <orig_flag> <new_flag>`**
This will add <orig_flag> to the list of renamed commands, with <new_flag> as its new name.
Example:
`-cmd_ren -abc -xyz`
Will cause any vendor command named `-abc`,  found later on the command line, to be renamed into the surelog `-xyz` command (assuming there is a surelog command named `-xyz`).

**`-cmd_mrg <orig_flag> <new_flag>`**
This will add <orig_flag> to the list of merged commands, with <new_flag> as its new name. When later the vendor command <orig_flag> will be encountered, it will be renamed into <new_flag>, appended a '+' and the following command line argument.
Example:
`cmd_mrg -incdir +incdir`
Will cause the following command line argument sequence `-incdir ./src` to be renamed into `+incdir+./src`

Advantages of the new approach:
1. Surelog's code no  longer needs to be cluttered with hundreds of useless vendor command strings
2. No need to risk copying a vendor's full command line interface into the Surelog codebase.
3. Every user of this feature can create his own set of commands that needs to be ignored, renamed or merged and put their definition in a single '-f` file, as usually users needs to recognize only a very small subset of the vendor's command line interface.

e.g. `surelog -f vendor_commands_im_using_in_my_legacy_file_list.f -f my_legacy_file_list.f`
With `vendor_commands_im_using_in_my_legacy_file_list.f` containing something along the lines of:
```
-cmd_ign -vendor_cmd1 0
-cmd_ign -vendor_cmd2 0
-cmd_ign -vendor_cmd3 1
-cmd_ren -vendor_cmd4 +surelog_cmd
-cmd_mrg -vendor_cmd5 +surelog_other_cmd
...
```

The new code is ready, and seems to pass trivial manual tests. I need to find how to create a `CommandLineParser_test.cpp` file that will get called.
As the last PR is already closed, should I open a new PR? My current code is rebased on top of `master` and takes care of removing the old PR.

alaindargelas, 2022-08-15:
Please submit a new PR

   

This will add to the list of dropped commands, with arguments.

closed 2022-08-20 by alaindargelas