chipsalliance/Surelog #3145: macros followed by parenthesis are always interpreted as macro parameters

udif, 2022-08-10:
Consider the following testcase:
```
`define D #1
module test(input clk);
reg d;
always @(posedge clk)
    d <= `D (~d);
endmodule
```
Running `surelog test.v` yields:
```
[INF:CM0023] Creating log file ./slpp_all/surelog.log.

[ERR:PP0107] test.v:5: Too many arguments (1) for macro "D",
             test.v:1:9: macro definition takes 0.

[  FATAL] : 0
[ SYNTAX] : 0
[  ERROR] : 1
[WARNING] : 0
[   NOTE] : 0
```

Verilator, Yosys and Xcelium handles this code with no warnings.

udif, 2022-08-15:
I've originally tried to fix this myself, in PreprocessFile.cpp, around lines 800-846, but then I realized that each time a macro invocation is followed by a '(', it is automatically considered as part of the macro parameters, regardless of the actual macro definition, and so it must be fixed in the lexer/parser.

The lexer/parser should not consume the '(' as part of the macro invocation unless the macro is already defined with one or more parameters.
I've looked at the grammar/SV3_1aPpLexer.g4, grammar/SV3_1aPpParser.g4, grammar/SV3_1aLexer.g4 and grammar/SV3_1aParser.g4 ,  but I'm not sure which rule exactly is handling this.

udif, 2022-08-15:
This what I came up with so far:

```
diff --git a/grammar/SV3_1aPpParser.g4 b/grammar/SV3_1aPpParser.g4
index e04845ed3..78ca8e974 100644
--- a/grammar/SV3_1aPpParser.g4
+++ b/grammar/SV3_1aPpParser.g4
@@ -16,6 +16,16 @@

 parser grammar SV3_1aPpParser;

+@parser::members {
+  //
+  bool isParamMacro() {
+    const std::string macro_name = getCurrentToken()->getText().substr(1);
+    bool is_macro_with_param = (macro_name != "D"); // just a dummy test for the moment
+    //std::cout << (is_macro_with_param ? "param\n" : "no_param\n");
+    return is_macro_with_param;
+  }
+}
+
 options { tokenVocab = SV3_1aPpLexer; }

 top_level_rule : null_rule source_text EOF ; // SV files
@@ -99,7 +109,7 @@ description :

 escaped_identifier : Escaped_identifier;

-macro_instance : (Macro_identifier | Macro_Escaped_identifier) Spaces* PARENS_OPEN macro_actual_args PARENS_CLOSE # MacroInstanceWithArgs
+macro_instance : {isParamMacro()}? (Macro_identifier | Macro_Escaped_identifier) Spaces* PARENS_OPEN macro_actual_args PARENS_CLOSE # MacroInstanceWithArgs
                | (Macro_identifier | Macro_Escaped_identifier)                                                    # MacroInstanceNoArgs
```

I've added a semantic predicate to the `macro_instance` rule part that parses macros with parameters.
I wrote a skeleton function to detect if a macro has parameters or not.
I can extract the macro name, and in the skeleton example I check if it does not begins with the letter "D".
This works OK (The original `test.v` is now parsed correctly because a macro named D is assumed to have no params, but a macro named E will be assumed to have parameters and fail to parse the `test.v` file, modified to use a parameter E instead of D).

Now all that is left is to replace the dummy check with a real one based on `getDefineList()` . However, I don't know how to access this method within the isParamMacro() method in the ANTLR PpParser class.

Any idea how to call `getDefineList()` from the PpParser class?

alaindargelas, 2022-08-15:
There is no back pointer from PpParser (antlr generated) to the
Preprocessor class. The PpParser class would have to be made a subclass of
some class with such pointer using multiple inheritance as it has its own
antlr parent class.
Grammar predicates are also slow and make the grammar target language
dependant.
The grammar can be executed in Java too today which is convenient for debug
and profiling.

Maybe the grammar needs a more profound change not to hastily absorb the
parenthesis or simply take the space in between the macro invocation and
the parenthesis as a deal breaker (end of macro instance)

shash, 2022-08-16:
> Maybe the grammar needs a more profound change not to hastily absorb the parenthesis or simply take the space in between the macro invocation and the parenthesis as a deal breaker (end of macro instance)

According to the Verizon-2005 standard this is indeed the correct behaviour: "The left parenthesis shall follow the text macro name immediately, with no space in between." (Page 351, section 19.3.1 `define).

udif, 2022-08-16:
OK, I was making some progress with semantic predicates (code still not working due to getMacro() below returning NULL)

```
@header {
#include <Surelog/SourceCompile/CompileSourceFile.h>
#include <Surelog/SourceCompile/PreprocessFile.h>
#include <Surelog/CommandLine/CommandLineParser.h>
#include <Surelog/SourceCompile/MacroInfo.h>
}
@parser::members {
  //
  PreprocessFile *pp;
  CommandLineParser *clp;
  void setPreprocessFile(PreprocessFile *pp) {
    this->pp = pp;
    this->clp = pp->getCompileSourceFile()->getCommandLineParser();
  }
  bool isParamMacro() {
    const std::string macro_name = getCurrentToken()->getText().substr(1);
    SymbolId macroId = pp->registerSymbol(macro_name);
    // if it's inm the +define override list, it has no macro arguments
    if (clp->getDefineList().count(macroId))
      return false;
    MacroInfo* m = pp->getMacro(macro_name);
    if (m->m_type == MacroInfo::Type::NO_ARGS)
      return false;
    return true;
  }
}
```

But given your suggestion I'll drop it and change this according to @shash 's comment above.

Also, as part of debugging the code above, I was surprised to learn that `+define+NAME=val` definitions **override** `` `define`` declarations in the code. I was always under the assumption `+define` is only used to assign an initial value to the macro that can be overridden in the code.
`getMacro()` is looking first in `m_define` and only if not found, it is looking in the macro list.

udif, 2022-08-16:
> According to the Verizon-2005 standard this is indeed the correct behaviour: "The left parenthesis shall follow the text macro name immediately, with no space in between." (Page 351, section 19.3.1 [...]

Not so fast...

Upon checking the relevant quote (ieee1364-2005), it seems the "no space between macro name and left parenthesis" rule applies to macro **definition**, not to macro usage. In fact, for macro usage, the explicit quote on the same page is:  

*To use a macro defined with arguments, the name of the text macro shall be followed by a list of actual
arguments in parentheses, separated by commas.*  **White space shall be allowed between the text macro name
and the left parenthesis.**  

The exact same wording also appears in IEEE1800-2017, section 22.5.1 (pages 676-677).

According to the standard, whitespaces may or may not appear between the macro name and the macro arguments left parenthesis, precluding any use of whitespaces for deciding if this macro has arguments or not.

This leaves me no choice other than using a semantic predicate.

udif, 2022-08-18:
I've been trying to solve this for a few days, but I'm not making progress, and would appreciate if someone took a look at it.
The PR contains 3 variations I've tried , but none work.
I've put debug prints around `getMacroInfo()` and `registerMacroInfo()`, and I'm getting  two calls to `getMacroInfo()` before the call to `registerMacroInfo()` and I don't know why.

alaindargelas, 2022-08-20:
This approach cannot work.

The predicate is called during parsing:

[image]

The macro is recorded during walking the parse tree (Too late, walking is done AFTER parsing):

[image]

I guess that is a limitation of the approach I took to write the preprocessor, predicates depending on walking the parse tree cannot work by construction.

Thomasb81, 2022-08-20:
@udif, @alaindargelas 

In https://github.com/Nic30/hdlConvertor we have use antlr mode feature to address this specific issue.
This allow to switch to specific grammar interpretation on specific part of the source code.

With this approach the grammar is no longer language agnostic, because verilog/SystemVerilog is so complex that some FSM is required to control in the lexer which token are emit but also mode evolution.

Antlr is a fantastic tool but I there not enough language agnostic options to address complex language as C++ or systemVerilog without adding language specific code ... Leading later to difficulties to only work on parser aspect of the tool.

Following python script using hdlConvertor produce the preprocessed code of your example:
```
from hdlConvertor import HdlConvertor
from hdlConvertorAst.language import Language

c = HdlConvertor()
 
test_result = c.verilog_pp("test.v",Language.VERILOG, [])
print(test_result)
```
Result:
```

module test(input clk);
reg d;
always @(posedge clk)
    d <= #1 (~d);
endmodule

```

alaindargelas, 2022-08-20:
Thanks @Thomasb81 for the pointer.

In Surelog case, since all the preprocessing magic happens in the walker, this is where I just had to relax a bogus rule.
Fixed with #3167

closed 2022-08-20 by alaindargelas

udif, 2022-08-20:
@alaindargelas thanks for solving this, I'm on to the next issues with my codebase (trying to understand why a certain path is not added to the `include file search path).
I always prefer coming up with a solution rather than just an issue. I hope next time I'll have a more useful PR.