chipsalliance/Surelog #3146: Deep recursion when running LoopParamOver test ?

hzeller, 2022-08-10:
The `LoopParamOver` test crashes when running with address sanitizer due to a very deep stack, which looks like a super-deep recursion (see below). It might only show up with address sanitizer as crash due to more limited resources, but given that the input is

```SystemVerilog
module Foo ();
 parameter P1 = 10;
 parameter P2 = P1 + P2;
 
endmodule

module top();
  parameter P3 = P1;
  parameter P2 = P3;
  parameter P1 = P2;
  Foo #(.P1(P2)) sub();
endmodule
```

... it is unclear where the deep recursion is coming from

```
*** SIGSEGV (@0x7fffda3ee400) ***
PC: @     0x56438cb460a5  (unknown)  UHDM::ExprEval::reduceExpr()
    @     0x564390160598        256  FailureSignalHandler()
    @     0x7fdae39eb1c0  (unknown)  (unknown)
    @     0x56438c459f7d       1056  SURELOG::CompileHelper::reduceExpr()
    @     0x56438c45c7d3       1840  SURELOG::CompileHelper::getValue()
    @     0x56438c48a380         80  std::__u::__function::__policy_invoker<>::__call_impl<>()
    @     0x56438cb4475f        352  UHDM::ExprEval::getValue()
    @     0x56438cb468f0      12352  UHDM::ExprEval::reduceExpr()
    @     0x56438c459f7d       1056  SURELOG::CompileHelper::reduceExpr()
    @     0x56438c45c7d3       1840  SURELOG::CompileHelper::getValue()
    @     0x56438c48a380         80  std::__u::__function::__policy_invoker<>::__call_impl<>()
    @     0x56438cb4475f        352  UHDM::ExprEval::getValue()
    @     0x56438cb468f0      12352  UHDM::ExprEval::reduceExpr()
    @     0x56438c459f7d       1056  SURELOG::CompileHelper::reduceExpr()
    @     0x56438c45c7d3       1840  SURELOG::CompileHelper::getValue()
    @     0x56438c48a380         80  std::__u::__function::__policy_invoker<>::__call_impl<>()
    @     0x56438cb4475f        352  UHDM::ExprEval::getValue()
    @     0x56438cb468f0      12352  UHDM::ExprEval::reduceExpr()
    @     0x56438c459f7d       1056  SURELOG::CompileHelper::reduceExpr()
    @     0x56438c45c7d3       1840  SURELOG::CompileHelper::getValue()
    @     0x56438c48a380         80  std::__u::__function::__policy_invoker<>::__call_impl<>()
    @     0x56438cb4475f        352  UHDM::ExprEval::getValue()
    @     0x56438cb468f0      12352  UHDM::ExprEval::reduceExpr()
    @     0x56438c459f7d       1056  SURELOG::CompileHelper::reduceExpr()
    @     0x56438c45c7d3       1840  SURELOG::CompileHelper::getValue()
    @     0x56438c48a380         80  std::__u::__function::__policy_invoker<>::__call_impl<>()
    @     0x56438cb4475f        352  UHDM::ExprEval::getValue()
    @     0x56438cb468f0      12352  UHDM::ExprEval::reduceExpr()
    @     0x56438c459f7d       1056  SURELOG::CompileHelper::reduceExpr()
    @     0x56438c45c7d3       1840  SURELOG::CompileHelper::getValue()
    @     0x56438c48a380         80  std::__u::__function::__policy_invoker<>::__call_impl<>()
    @     0x56438cb4475f        352  UHDM::ExprEval::getValue()
    @     0x56438cb468f0      12352  UHDM::ExprEval::reduceExpr()
    @     0x56438c459f7d       1056  SURELOG::CompileHelper::reduceExpr()
    @     0x56438c45c7d3       1840  SURELOG::CompileHelper::getValue()
    @     0x56438c48a380         80  std::__u::__function::__policy_invoker<>::__call_impl<>()
    @     0x56438cb4475f        352  UHDM::ExprEval::getValue()
    @     0x56438cb468f0      12352  UHDM::ExprEval::reduceExpr()
    @     0x56438c459f7d       1056  SURELOG::CompileHelper::reduceExpr()
    @     0x56438c45c7d3       1840  SURELOG::CompileHelper::getValue()
    @     0x56438c48a380         80  std::__u::__function::__policy_invoker<>::__call_impl<>()
    @     0x56438cb4475f        352  UHDM::ExprEval::getValue()
    @     0x56438cb468f0      12352  UHDM::ExprEval::reduceExpr()
    @     0x56438c459f7d       1056  SURELOG::CompileHelper::reduceExpr()
    @     0x56438c45c7d3       1840  SURELOG::CompileHelper::getValue()
    @     0x56438c48a380         80  std::__u::__function::__policy_invoker<>::__call_impl<>()
    @     0x56438cb4475f        352  UHDM::ExprEval::getValue()
    @     0x56438cb468f0      12352  UHDM::ExprEval::reduceExpr()
    @     0x56438c459f7d       1056  SURELOG::CompileHelper::reduceExpr()
    @     0x56438c45c7d3       1840  SURELOG::CompileHelper::getValue()
    @     0x56438c48a380         80  std::__u::__function::__policy_invoker<>::__call_impl<>()
    @     0x56438cb4475f        352  UHDM::ExprEval::getValue()
    @     0x56438cb468f0      12352  UHDM::ExprEval::reduceExpr()
    @     0x56438c459f7d       1056  SURELOG::CompileHelper::reduceExpr()
    @     0x56438c45c7d3       1840  SURELOG::CompileHelper::getValue()
    @     0x56438c48a380         80  std::__u::__function::__policy_invoker<>::__call_impl<>()
    @     0x56438cb4475f        352  UHDM::ExprEval::getValue()
    @     0x56438cb468f0      12352  UHDM::ExprEval::reduceExpr()
    @     0x56438c459f7d       1056  SURELOG::CompileHelper::reduceExpr()
    @     0x56438c45c7d3       1840  SURELOG::CompileHelper::getValue()
    @     0x56438c48a380         80  std::__u::__function::__policy_invoker<>::__call_impl<>()
    @     0x56438cb4475f        352  UHDM::ExprEval::getValue()
    @     0x56438cb468f0      12352  UHDM::ExprEval::reduceExpr()
    @     0x56438c459f7d       1056  SURELOG::CompileHelper::reduceExpr()
    @ ... and at least 1000 more frames
```

hs-apotell, 2022-08-10:
This recursion comes from the evaluation logic. The logic to identify circular dependence is based on it reaching 1000 attempts without success.

hzeller, 2022-08-10:
That sounds dangerous and problematic. Doesn't the usual 'no change in the last iteration' be sufficient ?

hs-apotell, 2022-08-10:
https://github.com/chipsalliance/Surelog/blob/103bf155d8c90c962e9f9695ffe59750d0b2722c/src/DesignCompile/CompileHelper.cpp#L78-L82

I have to reduce the limit down to 100 for Windows since the stack doesn't grow automatically as it does for Linux. The current stack size on Windows is 64 MB and with that limit only 100 iterations were possible. The situation is worse for Clang since the stack requirement for Clang is larger than both cl and gcc.

hs-apotell, 2022-08-10:
> Doesn't the usual 'no change in the last iteration' be sufficient ?

Yes, but I am just stating the facts. The current implementation relies on _number of attempts_ to evaluate. Sure, there is lot of room to improve on this logic.

alaindargelas, 2022-08-11:
The correct fix requires way too many changes in all the functions involved in expression evaluation both on the UHDM and Surelog side. Just for the sake of detecting an error condition that is far from the norm. Many limiting to 100 when running valgrind would be enough? We are attempting here to run valgrind on a negative test that is built to create a corner case behavior. Not the typical usage...

alaindargelas, 2022-08-30:
Create 2 log files, that is what is typically done. One per OS when there is OS specific diffs.