chipsalliance/Surelog #3156: Missing information on port connected to concatenation of nets

rkapuscik, 2022-08-18:
In a test case submitted [here](https://github.com/antmicro/yosys-systemverilog/issues/989) a module uses explicitly named ports connected to concatenation of internal nets:
```verilog
module gnl_8000_08_7_80_80
   (.op({\op[79] ,\op[78] ,\op[77] ,\op[76] ,\op[75] ,\op[74] ,
           (..and so on, up to..) 
           \op[0] }),
    clk,
    .ip({\ip[79] ,\ip[78] ,\ip[77] ,\ip[76] ,\ip[75] ,\ip[74] ,
         (..and so on, up to..)
         \ip[0] }));
```
but information in UHDM seems to be missing:
```
   |vpiPort:                                                                                                                                                                  
   \_port: , line:69:5, endln:69:1041                                                                                                                                         
     |vpiParent:                                                                                                                                                              
     \_module: work@gnl_8000_08_7_80_80 (work@gnl_8000_08_7_80_80), file:gnl_4_8_7_2.0_gnl_8000_08_7_80_80.v, line:68:1, endln:113449:10                                                                                                                                     
     |vpiDirection:3                                                                                                                                                          
   |vpiPort:                                                                                                                                                                  
   \_port: (clk), line:70:5, endln:70:8                                                                                                                                       
     |vpiParent:                                                                                                                                                              
     \_module: work@gnl_8000_08_7_80_80 (work@gnl_8000_08_7_80_80), file:gnl_4_8_7_2.0_gnl_8000_08_7_80_80.v, line:68:1, endln:113449:10                                                                                                                                     
     |vpiName:clk                                                                                                                                                             
     |vpiDirection:1                                                                                                                                                          
     |vpiLowConn:                                                                                                                                                             
     \_ref_obj:~                                                                                                                                                              
       |vpiActual:                                                                                                                                                            
       \_logic_net: (work@gnl_8000_08_7_80_80.clk), line:236:8, endln:236:11                                                                                                  
   |vpiPort:                                                                                                                                                                  
   \_port: , line:71:5, endln:71:1041                                                                                                                                         
     |vpiParent:                                                                                                                                                              
     \_module: work@gnl_8000_08_7_80_80 (work@gnl_8000_08_7_80_80), file:gnl_4_8_7_2.0_gnl_8000_08_7_80_80.v, line:68:1, endln:113449:10                                                                                                                                     
     |vpiDirection:1                                                                                                                                                          
```
Apparently the name for the port is missing, triggering an assert in later processing, but I'm not sure if this is converted properly and if so, how to access that information.

alaindargelas, 2022-08-19:
#3158 is partially addressing this issue.
3 more fixes need to happen to complete this case:
  - Remove the warnings about type and direction of the ports
  - Fix incorrect port declaration (incorrect range, should be bit select)
  - Remove the ports bits from the port list if they are used in the port expression

alaindargelas, 2022-08-20:
This is now fixed.
The port object has a lowConn relation that points to a concat operation with all the reb_obj to individual nets.

closed 2022-08-20 by alaindargelas