chipsalliance/Surelog #3169: -cmd_mrg not working as expected (fixed by PR#3168)

udif, 2022-08-21:
A minor bug in #3153 caused the `-cmd_mrg` command, intended to convert vendor commands into the `+cmd+arg` form, to work not work as expected.  
The `-cmd_mrg <oldcmd> <newcmd>` command with its 2 arguments are correctly parsed, but when encountering `<oldcmd> <arg>` , it is converted internally to `<newcmd>+arg` as expected, but not pushed back to the command line.

It is fixed in #3168 .

_Example:_

Testfile `test7.v`
```
`include "d.vh"
module test(input clk);
reg d;
always @(posedge clk)
    d <= `D (~d);
endmodule```
```

Include file `t/d.vh`
```
`define D #1
```

Normal mode:
```
localhost:~/git/Surelog$ surelog -verbose  +incdir+t test7.v
[INF:CM0023] Creating log file ./slpp_all/surelog.log.

[INF:PP0122] Preprocessing source file "test7.v".

[INF:PP0123] Preprocessing include file "t/d.vh".

[  FATAL] : 0
[ SYNTAX] : 0
[  ERROR] : 0
[WARNING] : 0
[   NOTE] : 0
```

The command below should have worked, converting `-incdir t` to `+incdir+t`, but in fact `+incdir+t` is never appended to the command line:
```
localhost:~/git/Surelog$ surelog -verbose -cmd_mrg -incdir +incdir -incdir t test7.v
[INF:CM0023] Creating log file ./slpp_all/surelog.log.

[INF:PP0122] Preprocessing source file "test7.v".

[INF:PP0123] Preprocessing include file "d.vh".

[ERR:PP0101] test7.v:1: Cannot open include file "d.vh".

[ERR:PP0102] test7.v:5:10: Unknown macro "D".
```

After bugfix:
```
localhost:~/git/Surelog$ build/bin/surelog -verbose -cmd_mrg -incdir +incdir -incdir t test7.v
[INF:CM0023] Creating log file ./slpp_all/surelog.log.

[INF:PP0122] Preprocessing source file "test7.v".

[INF:PP0123] Preprocessing include file "t/d.vh".

[  FATAL] : 0
[ SYNTAX] : 0
[  ERROR] : 0
[WARNING] : 0
[   NOTE] : 0
```

udif, 2022-08-21:
After this minor bug fix, I'm happy to report that I was able to run our entire top level design+dv hierarchy (3000+ files, including lots of 3rd party commercial IP),  with zero errors!

I haven't changed a single line in the top level compile script (other than calling a 20 line conversion script in the surelog command line, a script that maps or drops vendor commands to surelog syntax).

The only warnings encountered are `define redefinitions and `undef on non existing defines.

alaindargelas, 2022-08-21:
Awesome!

Regarding "The only warnings encountered are `define redefinitions and `undef on non existing defines." Is it possible that this entire project should be compiled with the "Compilation Unit" semantic of Verilog? Meaning that all the files are not meant to be compiled in the same macro space?

Surelog by default uses the "Interpreter Semantic" of Verilog, the macro space in the preprocessor is global to all the files, but there is an option (-fileunit) that makes each file it's own Compilation Unit, and only the macros define in that file and included files are used to compile that file, for the next .v the preprocessor is reset (no previous macro definition are used).

Also Surelog supports separate compilation, you can break down the list of files into compilation units that can be compiled separately (And in parallel), then link the libraries together in the final elaboration step (Also a familiar concept used by EDA vendors), see section in the README.

Finally, for the "Compilation Unit" or "Interpreted" semantics, you can use either multithreaded (-mt <n>) or multiprocess (-mp <n>) to speed up compilation (If that is a problem on a large project like this one).

alaindargelas, 2022-08-21:
How do you plan to use Surelog/UHDM next (if that can be disclosed)? 
- Use the Yosys plugin ?
- Analyse the design using UHDM ?
- ...

udif, 2022-08-21:
> Awesome!
>
> Regarding "The only warnings encountered are `define redefinitions and `undef on non existing defines." Is it possible that this entire project should be compiled with the "Compilation [...]

No, it is merely because the design is built gradually over 12 years with multiple generations of tape-outs, and multiple designers phasing in and out, and because unfortunately, strict methodology is not enforced. Most of the redefinitions is due to module owners adding their own `` `include`` with common defines at the block level, with these `` `define``s repeated at the top level.

Since both the DV and PD methodologies are based on building file lists hierarchically (i.e. `-f` at the chip/subchip level calling sub `-f` filelists of individual blocks and 3rd part IPs), I keep it that way.

udif, 2022-08-21:
> How do you plan to use Surelog/UHDM next (if that can be disclosed)?
> 
> * Use the Yosys plugin ?
> * Analyse the design using UHDM ?
> * ...

I'm just beginning to study UHDM and still overwhelmed by the README. About 10 years ago I evaluated a product from a small startup called Invionics that allowed me to parse an entire Verilog design (using a commercial parser, I think it was cheetah, not Verific), and gave me a Python API to traverse the design tree, modules, instances, ports, etc.

Anyway, this is not an official project, I'm just trying the stuff.
My first objective was to build an HTML design browser.  I already hacked one in python, which, given a Verilog design, it would build a static HTML file (no server needed) where I could fold the module hierarchy with instance names, and by clicking on a module, it would load it on the right side of the window.
Given its limited scope (only trace hierarchy, not signals), and the fact that it was more like a glorified regex beast that a real parser, it was OK. It did handle preprocessor defines, file lists, but it can't really handle stuff like `if generate` so it obviously parsed **too much**. It became too limiting and I wanted something better.
So the first project is to replicate it based on `util/uhdm-hier.cpp` as a starting point.

I'm also missing a tool that will give me a fully elaborated view of a design to make sure I got all the `generate` stuff correctly.

Another long-tem goal is to write a tool to track all the possible transaction paths inside the SoC, it has dozens of masters and slaves, using a simple company internal bus, and I would like to construct a table of all paths from each master to every slave connected to it, with the full path across the various design hierarchies, as well as the latency of each path (in clock cycles).

So while I started taking a look at UHDM, I see very little in term of user documentation and examples. Many of the source files mentioned in the UHDM README are not stand-alone programs, making it somewhat hard to see how they are actually used.
Perhaps by adding a github "Discussions" section to the UHDM repo will give a place where questions of "how to" could be asked and answered, which would be more appropriate than opening an Issue .

alaindargelas, 2022-08-22:
Opened https://github.com/chipsalliance/UHDM/discussions/782

closed 2022-08-22 by alaindargelas