chipsalliance/Surelog #3196: Port type for concatenation is missing

rkapuscik, 2022-08-31:
In a test case simplified from [this issue](https://github.com/antmicro/yosys-systemverilog/issues/989):
```verilog
module top
  (.op({\op[1] ,\op[0] }),
    clk,
    .ip({\ip[1] ,\ip[0] }));

  input  clk;
  output \op[0] ;
  output \op[1] ;
  input  \ip[0] ;
  input  \ip[1] ;

endmodule
```
`op` and `ip` are missing typedefs, nets or any other information about size of the port:
```
|vpiPort:
\_port: (op), line:2:4, endln:2:34
  |vpiParent:
  \_module: work@top (work@top), file:../wire_dec.sv, line:1:1, endln:12:10
  |vpiName:op
  |vpiDirection:3
  |vpiLowConn:
  \_operation: , line:2:9, endln:2:20
    |vpiOpType:33
    |vpiOperand:
    \_ref_obj: (op[1])
      |vpiName:op[1]
    |vpiOperand:
    \_ref_obj: (op[0])
      |vpiName:op[0]
|vpiPort:
\_port: (clk), line:3:5, endln:3:8
  |vpiParent:
  \_module: work@top (work@top), file:../wire_dec.sv, line:1:1, endln:12:10
  |vpiName:clk
  |vpiDirection:1
  |vpiLowConn:
  \_ref_obj: (work@top.clk), line:3:5, endln:3:8
    |vpiName:clk
    |vpiFullName:work@top.clk
    |vpiActual:
    \_logic_net: (work@top.clk), line:3:5, endln:3:8
  |vpiTypedef:
  \_logic_typespec: , line:6:10, endln:6:10
|vpiPort:
\_port: (ip), line:4:5, endln:4:35
  |vpiParent:
  \_module: work@top (work@top), file:../wire_dec.sv, line:1:1, endln:12:10
  |vpiName:ip
  |vpiDirection:1
  |vpiLowConn:
  \_operation: , line:4:10, endln:4:21
    |vpiOpType:33
    |vpiOperand:
    \_ref_obj: (ip[1])
      |vpiName:ip[1]
    |vpiOperand:
    \_ref_obj: (ip[0])
      |vpiName:ip[0]
```
`uhdmallModules` contains even less info, lacking port name:
```
|vpiPort:
\_port: , line:4:5, endln:4:35
  |vpiParent:
  \_module: work@top (work@top), file:../wire_dec.sv, line:1:1, endln:12:10
  |vpiDirection:1
```

alaindargelas, 2022-09-01:
You should only look at uhdmtopModules for information, uhdmallModules does not do binding and these ports with no name are the result of the port being an expression. Those ports get properly processed in uhdmtopModules.

You have the typedefs of all the individual bits in the ports:

```
 |vpiPort:
  \_port: (ip[1]), id:17, line:10:10, endln:10:21
    |vpiParent:
    \_module: work@top (work@top), id:59, file:tests/ScratchPad.sv, line:1:1, endln:12:10
    |vpiName:ip[1]
    |vpiDirection:1
    |vpiLowConn:
    \_ref_obj: (work@top.ip[1]), id:33, line:10:10, endln:10:21
      |vpiName:ip[1]
      |vpiFullName:work@top.ip[1]
      |vpiActual:
      \_logic_net: (work@top.ip[1]), id:26, line:10:10, endln:10:21
    |vpiTypedef:
    \_logic_typespec: , id:18, line:10:10, endln:10:10
```

And you have the composed ports:

```
|vpiPort:
  \_port: (op), id:0, line:2:4, endln:2:34
    |vpiParent:
    \_module: work@top (work@top), id:59, file:tests/ScratchPad.sv, line:1:1, endln:12:10
    |vpiName:op
    |vpiDirection:3
    |vpiLowConn:
    \_operation: , id:1, line:2:9, endln:2:20
      |vpiOpType:33
      |vpiOperand:
      \_ref_obj: (op[1]), id:2
        |vpiName:op[1]
      |vpiOperand:
      \_ref_obj: (op[0]), id:3
        |vpiName:op[0]
```

You should send to Yosys the composed ports and any port that is not part of a composed port. You have all the information you need.

 
```
|uhdmtopModules:
\_module: work@top (work@top), id:59, file:tests/ScratchPad.sv, line:1:1, endln:12:10
  |vpiName:work@top
  |vpiDefName:work@top
  |vpiTop:1
  |vpiNet:
  \_logic_net: (work@top.op[0]), id:20, line:7:10, endln:7:21
    |vpiParent:
    \_module: work@top (work@top), id:59, file:tests/ScratchPad.sv, line:1:1, endln:12:10
    |vpiTypespec:
    \_logic_typespec: , id:19, line:7:10, endln:7:10
    |vpiName:op[0]
    |vpiFullName:work@top.op[0]
  |vpiNet:
  \_logic_net: (work@top.op[1]), id:22, line:8:10, endln:8:21
    |vpiParent:
    \_module: work@top (work@top), id:59, file:tests/ScratchPad.sv, line:1:1, endln:12:10
    |vpiTypespec:
    \_logic_typespec: , id:21, line:8:10, endln:8:10
    |vpiName:op[1]
    |vpiFullName:work@top.op[1]
  |vpiNet:
  \_logic_net: (work@top.ip[0]), id:24, line:9:10, endln:9:21
    |vpiParent:
    \_module: work@top (work@top), id:59, file:tests/ScratchPad.sv, line:1:1, endln:12:10
[... 97 lines cut ...]
    |vpiName:ip[0]
    |vpiDirection:1
    |vpiLowConn:
    \_ref_obj: (work@top.ip[0]), id:32, line:9:10, endln:9:21
      |vpiName:ip[0]
      |vpiFullName:work@top.ip[0]
      |vpiActual:
      \_logic_net: (work@top.ip[0]), id:24, line:9:10, endln:9:21
    |vpiTypedef:
    \_logic_typespec: , id:16, line:9:10, endln:9:10
  |vpiPort:
  \_port: (ip[1]), id:17, line:10:10, endln:10:21
    |vpiParent:
    \_module: work@top (work@top), id:59, file:tests/ScratchPad.sv, line:1:1, endln:12:10
    |vpiName:ip[1]
    |vpiDirection:1
    |vpiLowConn:
    \_ref_obj: (work@top.ip[1]), id:33, line:10:10, endln:10:21
      |vpiName:ip[1]
      |vpiFullName:work@top.ip[1]
      |vpiActual:
      \_logic_net: (work@top.ip[1]), id:26, line:10:10, endln:10:21
    |vpiTypedef:
    \_logic_typespec: , id:18, line:10:10, endln:10:10
```

alaindargelas, 2022-09-08:
Fixed by #3215

closed 2022-09-08 by alaindargelas