chipsalliance/Surelog #3203: Caching issue with macros

dpetrisko, 2022-09-02:
Hi,

Trying to parse https://github.com/black-parrot/black-parrot with Surelog in deferred mode in https://github.com/antmicro/yosys-systemverilog and believe I've found a surelog problem.

To get the code:
```
git clone git@github.com:black-parrot/black-parrot
git submodule update --init --recursive
```

To reproduce,
```
surelog black-parrot/bp_be/src/v/bp_be_calculator/bp_be_pipe_int.sv -sverilog -Iblack-parrot/external/basejump_stl/bsg_misc -Iblack-parrot/external/basejump_stl/bsg_noc -Iblack-parrot/external/HardFloat/source -Iblack-parrot/external/HardFloat/source/RISCV -Iblack-parrot/bp_common/src/include -Iblack-parrot/bp_be/src/include

surelog black-parrot/bp_be/src/v/bp_be_calculator/bp_be_csr.sv -sverilog -Iblack-parrot/external/basejump_stl/bsg_misc -Iblack-parrot/external/basejump_stl/bsg_noc -Iblack-parrot/external/HardFloat/source -Iblack-parrot/external/HardFloat/source/RISCV -Iblack-parrot/bp_common/src/include -Iblack-parrot/bp_be/src/include

```

Produces this output:
```
...
[ERR:PP0102] black-parrot/bp_be/src/v/bp_be_calculator/bp_be_csr.sv:578:39: Unknown macro "BSG_SIGN_EXTEND".

[ERR:PP0102] black-parrot/bp_be/src/v/bp_be_calculator/bp_be_csr.sv:593:38: Unknown macro "BSG_SIGN_EXTEND".

[ERR:PP0102] black-parrot/bp_be/src/v/bp_be_calculator/bp_be_csr.sv:596:39: Unknown macro "BSG_SIGN_EXTEND".

[ERR:PP0102] black-parrot/bp_be/src/v/bp_be_calculator/bp_be_csr.sv:608:28: Unknown macro "BSG_SIGN_EXTEND".

[ERR:PP0102] black-parrot/bp_be/src/v/bp_be_calculator/bp_be_csr.sv:643:27: Unknown macro "BSG_SIGN_EXTEND".

[  FATAL] : 0
[ SYNTAX] : 0
[  ERROR] : 47
[WARNING] : 50
[   NOTE] : 0
```

However, this works:
```
surelog black-parrot/bp_be/src/v/bp_be_calculator/bp_be_pipe_int.sv -sverilog -Iblack-parrot/external/basejump_stl/bsg_misc -Iblack-parrot/external/basejump_stl/bsg_noc -Iblack-parrot/external/HardFloat/source -Iblack-parrot/external/HardFloat/source/RISCV -Iblack-parrot/bp_common/src/include -Iblack-parrot/bp_be/src/include

rm -rf slpp_all;

surelog black-parrot/bp_be/src/v/bp_be_calculator/bp_be_csr.sv -sverilog -Iblack-parrot/external/basejump_stl/bsg_misc -Iblack-parrot/external/basejump_stl/bsg_noc -Iblack-parrot/external/HardFloat/source -Iblack-parrot/external/HardFloat/source/RISCV -Iblack-parrot/bp_common/src/include -Iblack-parrot/bp_be/src/include

```

This workaround will not work in deferred mode, but hopefully they are the same problem. Thanks!

dpetrisko, 2022-09-02:
Related to https://github.com/chipsalliance/Surelog/issues/3008 perhaps?

dpetrisko, 2022-09-02:
I am using commit:

```
commit ab317007e7313ecd8643a65e35e12c88c058d6f0 (HEAD, origin/master, origin/HEAD, master)
Merge: 922672d47 6c58db95a
Author: alaindargelas <63669492+alaindargelas@users.noreply.github.com>
Date:   Wed Aug 31 13:09:02 2022 -0700

    Merge pull request #3197 from hs-apotell/platform-logs

    Add support for platform specific logs for regression tests
```

Would be happy to try another one

dpetrisko, 2022-09-02:
Gotcha, note that my example here does not use -sepcomp, so I believe it should be two independent compilations

alaindargelas, 2022-09-03:
The problem can be seen with a single file:

surelog black-parrot/bp_be/src/v/bp_be_calculator/bp_be_csr.sv -sverilog -Iblack-parrot/external/basejump_stl/bsg_misc -Iblack-parrot/external/basejump_stl/bsg_noc -Iblack-parrot/external/HardFloat/source -Iblack-parrot/external/HardFloat/source/RISCV -Iblack-parrot/bp_common/src/include -Iblack-parrot/bp_be/src/include

SAVING FILE: "./slpp_all/cache/work/bp_be_calculator_7954026238918316119/bp_be_csr.sv.slpp"
SAVING FILE: "./slpp_all/cache/work/include_15106086597427109405/bp_common_defines.svh.slpp"
SAVING MACRO: BP_COMMON_DEFINES_SVH
SAVING FILE: "./slpp_all/cache/work/bsg_misc_12438084844581288670/bsg_defines.v.slpp"
SAVING MACRO: BSG_ABSTRACT_MODULE
SAVING MACRO: BSG_CDIV
SAVING MACRO: BSG_COUNTONES_SYNTH
SAVING MACRO: BSG_DEFINES_V
SAVING MACRO: BSG_DISCONNECTED_IN_SIM
SAVING MACRO: BSG_GET_BIT
SAVING MACRO: BSG_HIDE_FROM_VERILATOR
SAVING MACRO: BSG_INV_PARAM
SAVING MACRO: BSG_IS_POW2
SAVING MACRO: BSG_MAX
SAVING MACRO: BSG_MIN
SAVING MACRO: BSG_SAFE_CLOG2
SAVING MACRO: BSG_SAFE_MINUS
SAVING MACRO: BSG_SIGN_EXTEND    <<< One of the error macro is seen here being saved
SAVING MACRO: BSG_STRINGIFY
SAVING MACRO: BSG_SYNTH_MUST_HARDEN
SAVING MACRO: BSG_UNDEFINED_IN_SIM
SAVING MACRO: BSG_VIVADO_SYNTH_FAILS
SAVING MACRO: BSG_WIDTH
SAVING MACRO: BSG_ZERO_EXTEND
SAVING MACRO: rpgroup

surelog black-parrot/bp_be/src/v/bp_be_calculator/bp_be_csr.sv -sverilog -Iblack-parrot/external/basejump_stl/bsg_misc -Iblack-parrot/external/basejump_stl/bsg_noc -Iblack-parrot/external/HardFloat/source -Iblack-parrot/external/HardFloat/source/RISCV -Iblack-parrot/bp_common/src/include -Iblack-parrot/bp_be/src/include

RESTORING FILE: "./slpp_all/cache/work/bp_be_calculator_7954026238918316119/bp_be_csr.sv.slpp"
RESTORING FILE: "./slpp_all/cache/work/include_15106086597427109405/bp_common_defines.svh.slpp"
RESTORING MACRO: BP_COMMON_DEFINES_SVH
RESTORING FILE: "./slpp_all/cache/work/bsg_misc_12438084844581288670/bsg_defines.v.slpp"

<<<< No macro gets restored

RESTORING FILE: "./slpp_all/cache/work/include_15106086597427109405/bp_common_addr_defines.svh.slpp"
RESTORING MACRO: BP_COMMON_ADDR_DEFINES_SVH
RESTORING MACRO: declare_bp_memory_map

@hs-apotell you have been modifying this code extensively, please check if your latest changes did not impact this.

alaindargelas, 2022-09-03:
@dpetrisko,  Can you try Surelog version prior to https://github.com/chipsalliance/Surelog/pull/3159

dpetrisko, 2022-09-04:
In our BP+Surelog CI we use -fileunit, which may be masking this issue. I can try without this flag on our current working version and see if that breaks

alaindargelas, 2022-09-04:
Is the intend of BP source code to be in interpreted semantic (all files in the same macro space) or compilation unit per file? It looks like it was the latter since you used the -fileunit option. Please try the -sepcomp -fileunit combination
As pointed above there is a multiple inclusion that does not work as expected

dpetrisko, 2022-09-05:
> Is the intend of BP source code to be in interpreted semantic (all files in the same macro space) or compilation unit per file? It looks like it was the latter since you used the -fileunit option. Please try the -sepcomp -fileunit combination
As pointed above there is a multiple inclusion that does not work as expected

Thanks, I'll give that a shot when I get a chance. We intend to support compilation unit per file since that seems to work across all tools, whereas global compilation unit has complicated ordering rules depending on tool implementation. 

That said, we expect that if unit per file works, then global compilation should always work expect perhaps in the presence of undefs (which we intentionally do not use)

alaindargelas, 2022-09-05:
@alaindargelas Question for you. What's the expected behavior for this scenario?

Consider three files a.sv, b.vh, c.vh

// a.sv:
`include "c.vh"
`include "b.vh"

// b.vh:
`include "c.vh"
`undef ADD

// c.vh:
`define ADD +
Note that c.vh is included by both a and b.

Surelog run -

surelog a.sv -nobuiltin
The output -

[INF:CM0023] Creating log file slpp_all\surelog.log.

Saving File: "slpp_all\\cache\\work\\_14695981039346656037\\a.sv.slpp"
Saving File: "slpp_all\\cache\\work\\_14695981039346656037\\c.vh.slpp"
Saving macro: ADD
Saving File: "slpp_all\\cache\\work\\_14695981039346656037\\b.vh.slpp"
Saving File: "slpp_all\\cache\\work\\_14695981039346656037\\c.vh.slpp"
[  FATAL] : 0
[ SYNTAX] : 0
[  ERROR] : 0
[WARNING] : 0
[   NOTE] : 0
Few problems here -

1) Cache file for c.vh got written twice. In fact, there are two instances of PreprocessFile and the file gets processed twice.

It is expected to be processed twice as within a compilation unit (All the files included from the .sv) all macros are in the same name space. Multiple inclusion could result in different macro results (A define or undefine could change the outcome of an include), Example:

// a.sv:
`define OP +
`include "b.vh"
assign o1 = a `ADD b;
`define OP -
`include "b.vh"
assign o2 = a `ADD b;

// b.vh:
`include "c.vh"

// c.vh:
`define ADD `OP

results in:
assign o1 = a + b;
assign o2 = a - b;

The undef in your example does what it is supposed to do, it does it is the last action concerning ADD and any use of ADD in a.sv will be faced with undefined macro, Surelog is correct there too.
 
2) Undef directives deletes macro that wasn't even defined in its own file.

That is what is expected from the preprocessor, the macro does not have a **file scope,** but it has a **compilation unit scope** (All the files compiled/included).

3)  For instance here, ADD wasn't defined in b.vh but undef-ing removes it from a.vh. When the latter gets written to disk, it has lost the entry.

**Now, this is the real problem,** the cache mechanism does not support multiple inclusion when the outcome of an included file can be different through the multiple inclusions. The preprocessor does the right thing, the cache does not.

4) This is a simpler version of root cause of this issue. I am not sure how you planned this to work. Are multiple instances of PreprocessFile (and preprocessing the same file multiple times) for the same file expected behavior?

Yes, it is for the preprocessor. For the cache I realize that does not work. Essentially, the preprocessor cache should be restored for the .sv and not for the .h, it should not even be saved for the .h in the first place to be exact as this only works when all inclusion have the same outcome.
The guarding in the files:
`ifndef BSG_DEFINES_V
`define BSG_DEFINES_V

prevent the subsequent inclusion to re-create the correct content since BSG_DEFINES_V is defined during the first inclusion.
For the preprocessor, it is fine, but it is not for the cache.

5) Optionally, the second save of c.vh.slpp - shouldn't that be before the save of b.vh.slpp i.e. before b gets a chance to modify the content of c?

We should disable the cache saving for include files to be on the safe side until we find a better solution with some hashing of the include file content in the file signature maybe?

alaindargelas, 2022-09-05:
@dpetrisko  #3208 fixes this

closed 2022-09-05 by alaindargelas

dpetrisko, 2022-09-07:
>@dpetrisko https://github.com/chipsalliance/Surelog/pull/3208 fixes this

Thanks @alaindargelas, can confirm this fix works for me in both this test case and my more complicated one