chipsalliance/Surelog #3206: Build fails on i386: error: conversion function cannot be redeclared

yurivict, 2022-09-04:
```
In file included from /wrkdirs/usr/ports/cad/surelog/work/Surelog-1.36/src/Library/ParseLibraryDef.cpp:25:
In file included from /wrkdirs/usr/ports/cad/surelog/work/Surelog-1.36/include/Surelog/Config/ConfigSet.h:28:
In file included from /wrkdirs/usr/ports/cad/surelog/work/Surelog-1.36/include/Surelog/Config/Config.h:29:
/wrkdirs/usr/ports/cad/surelog/work/Surelog-1.36/include/Surelog/Common/NodeId.h:27:3: error: conversion function cannot be redeclared
  operator std::size_t() const { return id; }
  ^
/wrkdirs/usr/ports/cad/surelog/work/Surelog-1.36/include/Surelog/Common/NodeId.h:26:3: note: previous definition is here
  operator RawNodeId() const { return id; }
  ^
1 error generated.
```

Version: 1.36
clang-14
FreeBSD 13.1

yurivict, 2022-09-05:
Compiler is clang-14. i386 *is* a 32-bit platform.

FreeBSD ports are built on a variety of platforms, i386 is one of them.

yurivict, 2022-09-05:
> Even if I fix the specific issue identified here, you might still run into issues with caching and such which aren't verified for 32 bit platform.

That's fine, I'll report further issues then. This is the way with most other ports that don't even have any CI.

Thomasb81, 2022-09-05:
- Surelog is not end-user application. 
- 32 bit platform are limited to 4Go of memory. (How many test in the current regression  testsuite exceed the 4096Mo of memory used ?)

From my point of view nobody is using anymore i386 platform for hardware development. 

What could be the effort interest other than being able to brand " This application work on freebsd i386" ?

yurivict, 2022-09-05:
On the other hand this is just a C++ code defect when 2 type conversion operators are defined for 2 very similar types. It doesn't take much to fix this place. If something is convertible to NodeId it probably can't also have the "size" meaning.

alaindargelas, 2022-09-06:
I don't need 32 bit support.

alaindargelas, 2022-09-06:
@hzeller will check this particular error.

hzeller, 2022-09-06:
Ah, I have seen this one before and did a local hack to fix, but it was on an ephemeral machine. Will recreate and send a PR.

PR #3211 (2022-09-06): Make it possible to compile NodeId on 32 bit systems.
Fixes #3206
commit f9275d56 (2022-09-06): Make it possible to compile NodeId on 32 bit systems.

Fixes #3206
closed 2022-09-06 by hzeller