chipsalliance/Surelog #3236: Nested foreach loop reports outer loop's index variable as its own in UHDM tree

mglb, 2022-09-22:
It looks like variables from non-last [...] are treated as loop variables.
Reference:
[1]: IEEE Std 1800-2021 "IEEE Standard for SystemVerilog", 12.7. Loop statements

Syntax:
```
foreach '(' ps_or_hierarchical_array_identifier '[' loop_variables ']' ')' statement
```

Example (https://github.com/chipsalliance/Surelog/blob/master/tests/ForeachSquare/dut.sv):
```systemverilog
module dut;

    int array[16][16];

    initial begin
        foreach(array[i])
            foreach(array[i][j])
                array[i][j] = i * j;
    end

endmodule
```

```
\_foreach_stmt: (work@dut), line:7:13, endln:7:20
  |vpiParent:
  \_foreach_stmt: (work@dut), line:6:9, endln:6:16
  |vpiFullName:work@dut
  |vpiVariables:
  \_ref_var: (work@dut.array), line:7:21, endln:7:26  ─┒
    |vpiParent:                                        ┃
    \_foreach_stmt: (work@dut), line:7:13, endln:7:20  ┃ Should refer to
    |vpiTypespec:                                      ┃ `dut.array[i]`?
    \_array_typespec:                                  ┃
    |vpiName:array                                     ┃
    |vpiFullName:work@dut.array                       ─┚
  |vpiLoopVars:                                       ─┒
  \_ref_var: (work@dut.i), line:7:27, endln:7:28       ┃
    |vpiParent:                                        ┃ This is a loop variable
    \_foreach_stmt: (work@dut), line:7:13, endln:7:20  ┃ of the outer `foreach`,
    |vpiTypespec:                                      ┃ not this one.
    \_int_typespec: , line:3:5, endln:3:8              ┃
    |vpiName:i                                         ┃
    |vpiFullName:work@dut.i                           ─┚
  |vpiLoopVars:
  \_int_var: (j), line:7:30, endln:7:31
    |vpiTypespec:
    \_int_typespec: , line:3:5, endln:3:8
      |vpiSigned:1
    |vpiName:j
  |vpiStmt:
  ...
```

alaindargelas, 2022-09-23:
This is non - synthesizable code. Let's focus on synthesizable code for the Yosys plugin.

alaindargelas, 2022-09-23:
Please provide a synthesizable example we can formally verify

mglb, 2022-09-24:
This one should be syntesizable:
```systemverilog
module dut(input clk);
    logic [31:0] array[4][4];
    always @(posedge clk) begin
        foreach(array[i])
            foreach(array[i][j])
                array[i][j] = i * j;
    end
endmodule
```

Relevant fragments of UHDM tree (command: `surelog -parse -d uhdm`):
```
...
|vpiStmt:
\_foreach_stmt: (work@dut), line:4:9, endln:4:16
  |vpiParent:
  \_begin: (work@dut), line:3:27, endln:7:8
  |vpiFullName:work@dut
  |vpiVariables:
  \_ref_var: (work@dut.array), line:4:17, endln:4:22
    |vpiParent:
    \_foreach_stmt: (work@dut), line:4:9, endln:4:16
    |vpiTypespec:
    \_array_typespec:
    |vpiName:array
    |vpiFullName:work@dut.array
  |vpiLoopVars:
  \_logic_var: (i), line:4:23, endln:4:24
    |vpiTypespec:
    \_logic_typespec: , line:2:5, endln:2:17
      |vpiRange:
      \_range: , line:2:11, endln:2:17
        |vpiLeftRange:
        \_constant: , line:2:12, endln:2:14
          |vpiParent:
          \_range: , line:2:11, endln:2:17
          |vpiDecompile:31
          |vpiSize:64
          |UINT:31
          |vpiConstType:9
        |vpiRightRange:
        \_constant: , line:2:15, endln:2:16
          |vpiParent:
          \_range: , line:2:11, endln:2:17
          |vpiDecompile:0
          |vpiSize:64
          |UINT:0
          |vpiConstType:9
    |vpiName:i
  |vpiStmt:
  \_foreach_stmt: (work@dut), line:5:13, endln:5:20
    |vpiParent:
    \_foreach_stmt: (work@dut), line:4:9, endln:4:16
    |vpiFullName:work@dut
    |vpiVariables:                                        ─┒
    \_ref_var: (work@dut.array), line:5:21, endln:5:26     ┃
      |vpiParent:                                          ┃
      \_foreach_stmt: (work@dut), line:5:13, endln:5:20    ┃ Should refer to
      |vpiTypespec:                                        ┃ `dut.array[i]`?
      \_array_typespec:                                    ┃
      |vpiName:array                                       ┃
      |vpiFullName:work@dut.array                         ─┚
    |vpiLoopVars:                                         ─┒
    \_ref_var: (work@dut.i), line:5:27, endln:5:28         ┃
      |vpiParent:                                          ┃ This is a loop variable
      \_foreach_stmt: (work@dut), line:5:13, endln:5:20    ┃ of the outer `foreach`,
      |vpiTypespec:                                        ┃ not this one.
      \_logic_typespec: , line:2:5, endln:2:17             ┃
      |vpiName:i                                           ┃
      |vpiFullName:work@dut.i                             ─┚
    |vpiLoopVars:
    \_logic_var: (j), line:5:30, endln:5:31
      |vpiTypespec:
      \_logic_typespec: , line:2:5, endln:2:17
        |vpiRange:
        \_range: , line:2:11, endln:2:17
          |vpiLeftRange:
          \_constant: , line:2:12, endln:2:14
            |vpiParent:
            \_range: , line:2:11, endln:2:17
            |vpiDecompile:31
            |vpiSize:64
            |UINT:31
            |vpiConstType:9
          |vpiRightRange:
          \_constant: , line:2:15, endln:2:16
            |vpiParent:
            \_range: , line:2:11, endln:2:17
            |vpiDecompile:0
            |vpiSize:64
            |UINT:0
            |vpiConstType:9
      |vpiName:j
    |vpiStmt:
    ...
```

alaindargelas, 2022-09-26:
I changed the testcase into to make it useful for Synthesis comparison (Observable):

```
module dut(input clk, output logic [31:0] data[4][4]);
    always @(posedge clk) begin
        foreach(data[i])
            foreach(data[i][j])
                data[i][j] = i * j;
    end
endmodule
```

I made a fix in #3243 to get the right typespec for data when it is a port (Not a variable).

You need to check the elaborated model to get more accurate binding. I think the elaborated model contains the correct bindings for the variable and the loop vars.

To the question: Should refer to  `dut.array[i]`?
my answer is no, the variable is array, the loop var is i.

closed 2022-10-07 by alaindargelas