chipsalliance/Surelog #3346: Calling a func inside a range doesn't work properly

mandrys, 2022-11-17:
In the example below, the range of `xc` is miscalculated because of using a function call.

```
module func_width_scope_top(inp, out1, out2);
    input wire inp;

    localparam WIDTH_A = 5;
    function automatic [WIDTH_A-1:0] func1;
        input reg [WIDTH_A-1:0] s;
        func1 = ~s; 
    endfunction
    wire [func1(0)-1:0] xc;
    assign xc = 1'sb1;

    output wire [1023:0] out1, out2;
    wire [WIDTH_A-1:0] xn; 
    assign xn = func1(inp);
    assign out1 = xn; 
    assign out2 = xc; 

endmodule
```
As a result, we get negative `vpiDecompile` value and wrong INT value:

```
|vpiNet:
\_logic_net: (work@func_width_scope_top.xc), line:9:22, endln:9:24
  |vpiParent:
  \_module: work@func_width_scope_top (work@func_width_scope_top), file:/yosys-systemverilog/yosys-systemverilog/yosys/tests/simple/func_width_scope_trimmed.v, line:1:1, endln:19:10
  |vpiTypespec:
  \_logic_typespec: , line:9:2, endln:9:21
    |vpiRange:
    \_range: , line:9:7, endln:9:21
      |vpiLeftRange:
      \_constant: , line:9:8, endln:9:16
        |vpiParent:
        \_range: , line:9:7, endln:9:21
        |vpiDecompile:-2
        |vpiSize:64
        |INT:-2
        |vpiConstType:7
      |vpiRightRange:
      \_constant: , line:9:19, endln:9:20
        |vpiParent:
        \_range: , line:9:7, endln:9:21
        |vpiDecompile:0
        |vpiSize:64
        |UINT:0
        |vpiConstType:9
  |vpiName:xc 
  |vpiFullName:work@func_width_scope_top.xc
  |vpiNetType:1
```

 If we use a constant instead of a func call, like `wire [31-1:0] xc;`, the example works fine.

alaindargelas, 2022-11-18:
The function return value range is ignored. No matter what is the call context. That's the problem.

alaindargelas, 2022-11-21:
Fixed by #3347

closed 2022-11-21 by alaindargelas