chipsalliance/Surelog #3348: Antlr v4.11.1 is not working for Surelog

alaindargelas, 2022-11-21:
I updated locally Antlr v4.11.1, 
I replaced the jar file in the CMakeLists.txt:
set(ANTLR_JAR_LOCATION
    ${PROJECT_SOURCE_DIR}/third_party/antlr4_bin/antlr-4.11.1-complete.jar)

Compilation fails during the execution of compiling the UVM packages, after many warnings:

```
make
...
/home/alain/Surelog/build/generated/src/parser/SV3_1aParser.cpp: In member function ‘SURELOG::SV3_1aParser::Package_scopeContext* SURELOG::SV3_1aParser::package_scope()’:
/home/alain/Surelog/build/generated/src/parser/SV3_1aParser.cpp:76312:80: warning: suggest parentheses around ‘&&’ within ‘||’ [-Wparentheses]
76312 |     if (!(_la == SV3_1aParser::DOLLAR_UNIT || (((_la - 326) & ~ 0x3fULL) == 0) &&
      |                                               ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~
76313 |       ((1ULL << (_la - 326)) & 4672924483589) != 0)) {
      |       ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~                              
/home/alain/Surelog/build/generated/src/parser/SV3_1aParser.cpp: In member function ‘SURELOG::SV3_1aParser::Ps_identifierContext* SURELOG::SV3_1aParser::ps_identifier()’:
/home/alain/Surelog/build/generated/src/parser/SV3_1aParser.cpp:76420:80: warning: suggest parentheses around ‘&&’ within ‘||’ [-Wparentheses]
76420 |     if (!(_la == SV3_1aParser::DOLLAR_UNIT || (((_la - 326) & ~ 0x3fULL) == 0) &&
      |                                               ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~
76421 |       ((1ULL << (_la - 326)) & 4672924483589) != 0)) {
      |       ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~                              
[ 98%] Linking CXX static library lib/libsurelog.a
gmake[3]: Leaving directory '/home/alain/Surelog/build'
[ 98%] Built target surelog
gmake[3]: Entering directory '/home/alain/Surelog/build'
gmake[3]: Entering directory '/home/alain/Surelog/build'
gmake[3]: Entering directory '/home/alain/Surelog/build'
gmake[3]: Entering directory '/home/alain/Surelog/build'
Scanning dependencies of target surelog-bin
Scanning dependencies of target hellouhdm
Scanning dependencies of target roundtrip
Scanning dependencies of target hellodesign
gmake[3]: Leaving directory '/home/alain/Surelog/build'
gmake[3]: Entering directory '/home/alain/Surelog/build'
[ 98%] Building CXX object CMakeFiles/surelog-bin.dir/src/main.cpp.o
gmake[3]: Leaving directory '/home/alain/Surelog/build'
gmake[3]: Leaving directory '/home/alain/Surelog/build'
gmake[3]: Leaving directory '/home/alain/Surelog/build'
gmake[3]: Entering directory '/home/alain/Surelog/build'
gmake[3]: Entering directory '/home/alain/Surelog/build'
gmake[3]: Entering directory '/home/alain/Surelog/build'
[ 98%] Building CXX object CMakeFiles/hellodesign.dir/src/hellodesign.cpp.o
[ 98%] Building CXX object CMakeFiles/hellouhdm.dir/src/hellouhdm.cpp.o
[ 98%] Building CXX object CMakeFiles/roundtrip.dir/src/roundtrip.cpp.o
[ 98%] Linking CXX executable bin/surelog
       Staging for precompiled packages ...
       ... staging completed.
gmake[3]: Leaving directory '/home/alain/Surelog/build'
[ 98%] Built target surelog-bin
gmake[3]: Entering directory '/home/alain/Surelog/build'
gmake[3]: Entering directory '/home/alain/Surelog/build'
Scanning dependencies of target PrecompileUVM
Scanning dependencies of target PrecompileOVM
gmake[3]: Leaving directory '/home/alain/Surelog/build'
gmake[3]: Leaving directory '/home/alain/Surelog/build'
gmake[3]: Entering directory '/home/alain/Surelog/build'
gmake[3]: Entering directory '/home/alain/Surelog/build'
[ 98%] Generating bin/pkg/work/uvm_pkg.sv.slpa
[ 98%] Generating bin/pkg/work/ovm_pkg.sv.slpa
       Creating OVM precompiled package...
       Creating UVM precompiled package...
gmake[3]: *** [CMakeFiles/PrecompileOVM.dir/build.make:82: bin/pkg/work/ovm_pkg.sv.slpa] Error 139
gmake[3]: Leaving directory '/home/alain/Surelog/build'
gmake[2]: *** [CMakeFiles/Makefile2:880: CMakeFiles/PrecompileOVM.dir/all] Error 2
gmake[2]: *** Waiting for unfinished jobs....
gmake[3]: *** [CMakeFiles/PrecompileUVM.dir/build.make:82: bin/pkg/work/uvm_pkg.sv.slpa] Error 139
gmake[3]: Leaving directory '/home/alain/Surelog/build'
gmake[2]: *** [CMakeFiles/Makefile2:1227: CMakeFiles/PrecompileUVM.dir/all] Error 2
[100%] Linking CXX executable bin/hellodesign
[... 36 lines cut ...]
#11 0x0000555555af1639 in antlr4::atn::ParserATNSimulator::closure_(std::shared_ptr<antlr4::atn::ATNConfig> const&, antlr4::atn::ATNConfigSet*, std::unordered_set<std::shared_ptr<antlr4::atn::ATNConfig>, antlr4::atn::ATNConfig::Hasher, antlr4::atn::ATNConfig::Comparer, std::allocator<std::shared_ptr<antlr4::atn::ATNConfig> > >&, bool, bool, int, bool) ()
#12 0x0000555555af1639 in antlr4::atn::ParserATNSimulator::closure_(std::shared_ptr<antlr4::atn::ATNConfig> const&, antlr4::atn::ATNConfigSet*, std::unordered_set<std::shared_ptr<antlr4::atn::ATNConfig>, antlr4::atn::ATNConfig::Hasher, antlr4::atn::ATNConfig::Comparer, std::allocator<std::shared_ptr<antlr4::atn::ATNConfig> > >&, bool, bool, int, bool) ()
#13 0x0000555555af1639 in antlr4::atn::ParserATNSimulator::closure_(std::shared_ptr<antlr4::atn::ATNConfig> const&, antlr4::atn::ATNConfigSet*, std::unordered_set<std::shared_ptr<antlr4::atn::ATNConfig>, antlr4::atn::ATNConfig::Hasher, antlr4::atn::ATNConfig::Comparer, std::allocator<std::shared_ptr<antlr4::atn::ATNConfig> > >&, bool, bool, int, bool) ()
#14 0x0000555555af1639 in antlr4::atn::ParserATNSimulator::closure_(std::shared_ptr<antlr4::atn::ATNConfig> const&, antlr4::atn::ATNConfigSet*, std::unordered_set<std::shared_ptr<antlr4::atn::ATNConfig>, antlr4::atn::ATNConfig::Hasher, antlr4::atn::ATNConfig::Comparer, std::allocator<std::shared_ptr<antlr4::atn::ATNConfig> > >&, bool, bool, int, bool) ()
#15 0x0000555555aed559 in antlr4::atn::ParserATNSimulator::computeReachSet(antlr4::atn::ATNConfigSet*, unsigned long, bool) ()
#16 0x0000555555ae9f59 in antlr4::atn::ParserATNSimulator::computeTargetState(antlr4::dfa::DFA&, antlr4::dfa::DFAState*, unsigned long) ()
#17 0x0000555555ae8c7d in antlr4::atn::ParserATNSimulator::execATN(antlr4::dfa::DFA&, antlr4::dfa::DFAState*, antlr4::TokenStream*, unsigned long, antlr4::ParserRuleContext*) ()
#18 0x0000555555aef066 in antlr4::atn::ParserATNSimulator::adaptivePredict(antlr4::TokenStream*, unsigned long, antlr4::ParserRuleContext*) ()
#19 0x00005555558d3626 in SURELOG::SV3_1aPpParser::endif_directive() ()
#20 0x00005555558dc8d5 in SURELOG::SV3_1aPpParser::description() ()
#21 0x00005555558dccde in SURELOG::SV3_1aPpParser::source_text() ()
#22 0x00005555558dce03 in SURELOG::SV3_1aPpParser::top_level_rule() ()
#23 0x0000555555748edd in SURELOG::PreprocessFile::preprocess() ()
#24 0x0000555555753ae8 in SURELOG::SV3_1aPpTreeShapeListener::enterInclude_directive(SURELOG::SV3_1aPpParser::Include_directiveContext*) ()
--Type <RET> for more, q to quit, c to continue without paging-- 
#25 0x0000555555b29cd6 in antlr4::tree::IterativeParseTreeWalker::walk(antlr4::tree::ParseTreeListener*, antlr4::tree::ParseTree*) const ()
#26 0x000055555574855b in SURELOG::PreprocessFile::preprocess() ()
#27 0x0000555555a65e7a in SURELOG::CompileSourceFile::preprocess_() ()
#28 0x000055555573423c in SURELOG::Compiler::compileFileSet_(SURELOG::CompileSourceFile::Action, bool, std::vector<SURELOG::CompileSourceFile*, std::allocator<SURELOG::CompileSourceFile*> >&) ()
#29 0x0000555555737e29 in SURELOG::Compiler::compile() ()
#30 0x00005555556dfb26 in SURELOG::start_compiler(SURELOG::CommandLineParser*) ()
#31 0x00005555556dc6fd in executeCompilation(int, char const**, bool, bool, SURELOG::ErrorContainer::Stats*) ()
#32 0x00005555556d5dda in main ()

```

alaindargelas, 2022-11-23:
@parrt is that a known issue ?

parrt, 2022-11-23:
Yes that was fixed but I introduce another bug but that has also been fixed haha. If you grab the latest from dev branch, the upcoming 4.11.2, you will have the fixes. I think you want to patch 

https://github.com/antlr/antlr4/pull/3958

and

https://github.com/antlr/antlr4/pull/3978

parrt, 2022-11-23:
by the way I also fixed that parentheses thing...

alaindargelas, 2022-11-24:
Thanks @parrt!

alaindargelas, 2022-11-24:
Fixed by https://github.com/chipsalliance/Surelog/pull/3354

closed 2022-11-24 by alaindargelas