chipsalliance/Surelog #3370: Compilation error on Windows with MSYS2 (MINGW64 environment)

RustamC, 2022-12-08:
Hi!

I've tried to build the [systemverilog plugin](https://github.com/chipsalliance/yosys-f4pga-plugins/tree/main/systemverilog-plugin) for yosys on Windows using [msys2/mingw64](https://www.msys2.org/docs/environments/). It links to the [Surelog](https://github.com/chipsalliance/Surelog/) and UHDM static libraries. So I built Surelog which also builds UHDM.

All went well, but during compilation I ran into a problem with Surelog.

1) While building Surelog on Windows, I encountered that the Cmake `create_symlink` command didn't work (even though I used the latest Cmake version) and I had to replace it with `copy`/`copy_directory`. 
    * In the last Cmake command, `build/compile_commands.json` is copied into the source directory:
https://github.com/chipsalliance/Surelog/blob/b08d86c5baa54f6cdb03ad9dcf141f338f94e254/CMakeLists.txt#L851-L854
On Windows I had to change it to `${PROJECT_SOURCE_DIR}/build/compile_commands.json` to make it work (of course, after replacing `create_symlink`)
2) When I tried to compile systemverilog plugin, I'd got an error in this line:
https://github.com/chipsalliance/Surelog/blob/b08d86c5baa54f6cdb03ad9dcf141f338f94e254/include/Surelog/ErrorReporting/ErrorDefinition.h#L36
`ERROR` is already defined in Windows API, so I had to add `#undef ERROR` before class declaration to avoid this problem. From my point of view, this is not the most elegant solution. I'm not familiar with the codebase & perhaps it'll be better just to rename `ERROR` to smth like `SURELOG_ERROR`, for example.

alaindargelas, 2022-12-09:
@hs-apotell who uses the Windows build all the time can comment.

RustamC, 2022-12-12:
1. CMake version:
```powershell
PS C:\Users\user1> cmake --version
cmake version 3.25.0

CMake suite maintained and supported by Kitware (kitware.com/cmake).
PS C:\Users\user1> whereis cmake
cmake: /usr/share/cmake /mingw64/bin/cmake.exe
```
2. CMake Error:
```powershell
PS D:\sources> cd SurelogOriginal
PS D:\sources\SurelogOriginal> cmake -G "MinGW Makefiles" -DCMAKE_BUILD_TYPE=Release -DCMAKE_INSTALL_PREFIX=D:/opt/Surelog
 -DCMAKE_POSITION_INDEPENDENT_CODE=ON -S . -B build
PS D:\sources\SurelogOriginal> cmake --build build
[ 14%] Building CXX object third_party/flatbuffers/CMakeFiles/flatc.dir/src/idl_gen_dart.cpp.obj
[ 14%] Building CXX object third_party/antlr4/runtime/Cpp/runtime/CMakeFiles/antlr4_static.dir/src/NoViableAltException.cpp.obj
[ 14%] Building CXX object third_party/antlr4/runtime/Cpp/runtime/CMakeFiles/antlr4_static.dir/src/Parser.cpp.obj
[ 14%] Building CXX object third_party/antlr4/runtime/Cpp/runtime/CMakeFiles/antlr4_static.dir/src/ParserInterpreter.cpp.obj
CMake Error: failed to create symbolic link 'D:/sources/SurelogOriginal/compile_commands.json': ������ �� �������� ���������� �������.

mingw32-make[2]: *** [CMakeFiles\link_target.dir\build.make:69: CMakeFiles/link_target] Error 1
mingw32-make[1]: *** [CMakeFiles\Makefile2:1179: CMakeFiles/link_target.dir/all] Error 2
mingw32-make[1]: *** Waiting for unfinished jobs....
```

3. As for ERROR in Windows API, it occurs when I try to compile [yosys](https://github.com/RustamC/yosys/tree/win-static-build-024) with [SystemVerilog plugin](https://github.com/RustamC/yosys/tree/win-static-build-024/frontends/systemverilog) (taken from [here](https://github.com/chipsalliance/yosys-f4pga-plugins/tree/main/systemverilog-plugin)):
```powershell
PS D:\sources\yosys> make -j4
[Makefile.conf] CONFIG := msys2-64
[Makefile.conf] PREFIX := D:/opt/y++
[Makefile.conf] ENABLE_GHDL := 1
[Makefile.conf] GHDL_PREFIX := D:/msys64/mingw64
[Makefile.conf]
[Makefile.conf] ENABLE_SV := 1
[Makefile.conf] UHDM_INSTALL_DIR := D:/opt/Surelog
couldn't read file "/proc/1134/fd/63": no such file or directory
couldn't read file "/proc/1138/fd/63": no such file or directory
[  0%] Building kernel/version_da1197127.cc
[  0%] Building kernel/driver.o
[  0%] Building techlibs/common/simlib_help.inc
[  0%] Building techlibs/common/simcells_help.inc
[  1%] Building kernel/rtlil.o
....
[ 15%] Building frontends/systemverilog/UhdmAst.o
[ 15%] Building frontends/systemverilog/UhdmAstUpstream.o
[ 16%] Building frontends/systemverilog/uhdmastfrontend.o
[ 16%] Building frontends/systemverilog/uhdmcommonfrontend.o
[ 16%] Building frontends/systemverilog/uhdmsurelogastfrontend.o
In file included from D:/msys64/mingw64/include/windows.h:71,
                 from ./kernel/yosys.h:109,
                 from ./kernel/rtlil.h:20,
                 from ./frontends/ast/ast.h:32,
                 from frontends/systemverilog/UhdmAst.h:4,
                 from frontends/systemverilog/uhdmsurelogastfrontend.cc:20:
D:/opt/Surelog/include/Surelog/ErrorReporting/ErrorDefinition.h:36:39: error: expected identifier before numeric constant
   36 |   enum ErrorSeverity { FATAL, SYNTAX, ERROR, WARNING, INFO, NOTE };
      |                                       ^~~~~
D:/opt/Surelog/include/Surelog/ErrorReporting/ErrorDefinition.h:36:39: error: expected '}' before numeric constant
In file included from D:/opt/Surelog/include/Surelog/ErrorReporting/Error.h:28,
                 from D:/opt/Surelog/include/Surelog/surelog.h:25,
                 from frontends/systemverilog/uhdmsurelogastfrontend.cc:35:
D:/opt/Surelog/include/Surelog/ErrorReporting/ErrorDefinition.h:36:22: note: to match this '{'
   36 |   enum ErrorSeverity { FATAL, SYNTAX, ERROR, WARNING, INFO, NOTE };
      |                      ^
D:/opt/Surelog/include/Surelog/ErrorReporting/ErrorDefinition.h:36:39: error: expected unqualified-id before numeric constant
   36 |   enum ErrorSeverity { FATAL, SYNTAX, ERROR, WARNING, INFO, NOTE };
      |                                       ^~~~~
D:/opt/Surelog/include/Surelog/ErrorReporting/ErrorDefinition.h:253:28: error: expected ')' before 'severity'
  253 |     ErrorInfo(ErrorSeverity severity, ErrorCategory category,
[... 85 lines cut ...]
      |                        ^~~~~~~~~~~
D:/opt/Surelog/include/Surelog/ErrorReporting/ErrorDefinition.h:277:15: warning: 'void SURELOG::setSeverity(ErrorType, int)' declared 'static' but never defined [-Wunused-function]
  277 |   static void setSeverity(ErrorType type, ErrorSeverity severity);
      |               ^~~~~~~~~~~
D:/opt/Surelog/include/Surelog/ErrorReporting/ErrorDefinition.h:278:15: warning: 'void SURELOG::rec(ErrorType, int, ErrorCategory, std::string_view, std::string_view)' declared 'static' but never defined [-Wunused-function]
  278 |   static void rec(ErrorType type, ErrorSeverity severity,
      |               ^~~
D:/opt/Surelog/include/Surelog/ErrorReporting/ErrorDefinition.h:288:20: warning: 'SURELOG::ErrorMap* SURELOG::mutableGlobalErrorInfoMap()' declared 'static' but never defined [-Wunused-function]
  288 |   static ErrorMap* mutableGlobalErrorInfoMap();
      |                    ^~~~~~~~~~~~~~~~~~~~~~~~~
D:/opt/Surelog/include/Surelog/ErrorReporting/ErrorDefinition.h:268:48: warning: 'const std::map<SURELOG::ErrorType, SURELOG::ErrorInfo>& SURELOG::getErrorInfoMap()' defined but not used [-Wunused-function]
  268 |   static const std::map<ErrorType, ErrorInfo>& getErrorInfoMap() {
[... cut ...] ~~~~~~
make: *** [Makefile:767: frontends/systemverilog/uhdmsurelogastfrontend.o] Error 1
make: *** Waiting for unfinished jobs....

```
4. I'm building with gcc:
```powershell
PS D:\sources\yosys> x86_64-w64-mingw32-g++ --version
x86_64-w64-mingw32-g++.exe (Rev6, Built by MSYS2 project) 12.2.0
Copyright (C) 2022 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

RustamC, 2022-12-14:
1. I've tried building under an administrator console and it **worked**. Although I have administrator account and have all the permissions except for the special ones. I can copy files but can't create links ¯\_(ツ)_/¯
2. Same error.
3. I think, it's due to ru-RU locale.
4. Nothing has changed.

RustamC, 2022-12-14:
Yes, it's disabled :-) So mystery solved.

Thank you, @hs-apotell!