chipsalliance/Surelog #3371: Surelog verilog parser enhancement request

shakouri, 2022-12-09:
An RTL that a commercial synthesis tool (Synopsys Design Compiler) is able to read surelog is not able to parse.

1.  Allow parameter passing w/o use of parantheses ('-' denotes original RTL, and '+' denotes changes to make yosys read it in):
[image]

2. Allow use of logic as instance name('-' denotes original RTL, and '+' denotes changes to make yosys read it in):
[image]

Thomasb81, 2022-12-09:
Hello

If DC support 1)  I would say that DC is not properly supporting verilog and/or systemVerilog standard...
The problem to use non standard construction is that they are non-standard ...
There is many place where this syntax `#( ) `is used to specify parameters of object. What are the exact syntax to support ? How does it look like with multiple variation to specialize parameter ? Would you be able to fully specify what you want (because saying like "this specific tool", is not very helpful in the perspective of doing a change) ? ... That's an huge work to do. Not sure someone will specify and implement that for free.

On 2)`logic` is valid identifier for verilog but not for systemVerilog, see Table B.1 of SystemVerilog standard revision 2017. SystemVerilog is more restrictive than verilog on identifier choice because it has more keyword.
SystemVerilog (Chapter 22.14) specifies that you can use directive _begin_keywords_ and _end_keywords_ to work-arround reserved identifier for a code section.
I am not sure _begin_keyword_/_end_keyword_ is actually supported by Surelog.
I don't think Surelog can specify to be to consider a specific file as verilog source code.

shakouri, 2022-12-09:
The issue is with older IPs that commercial tools support and newer tools are not supporting when ported.

alaindargelas, 2022-12-10:
@shakouri please upload small .v files that are self contained with these example codes.

shakouri, 2022-12-10:
Here is a small .v file with example codes:
[dffr.v.gz](https://github.com/chipsalliance/Surelog/files/10200888/dffr.v.gz)

alaindargelas, 2022-12-30:
Fixed by #3409

closed 2022-12-30 by alaindargelas