chipsalliance/Surelog #3373: incorrect [ERR:CP0317] undefined type

Thomasb81, 2022-12-10:
Hello

On following code the tool report:
 [ERR:CP0317] nettype.sv:6:13: Undefined type "my_real".
```
nettype real my_real;

module dut ();

    my_real my_real_net;

endmodule
```

Ieee1800-2017 chapter 6.6.7 define how we can declare an user-define nettype, it say exactly:
"This nettype is similar to a typedef in some ways, but shall only be used in declaring a net."

alaindargelas, 2022-12-10:
Completely not supported as of now.

alaindargelas, 2022-12-10:
I can get to it in a while, after current issues.

alaindargelas, 2023-01-09:
Fixed by #3425

Thomasb81, 2023-01-09:
Thanks for the progress.

I have an additional question.

If I have :
`nettype bit net_bit;`

capture as
```
 ====== UHDM =======
design: (unnamed)
|vpiName:unnamed
|vpiTypedef:
\_bit_typespec: (net_bit), line:1:9, endln:1:12
  |vpiParent:
  \_design: (unnamed)
  |vpiName:net_bit
  |vpiInstance:
  \_design: (unnamed)
  |vpiSigned:1
===================
```

Why 
```
nettype bit net_bit with my_function;

function automatic bit my_function();
endfunction
```

is captured has an array:
```
====== UHDM =======
design: (unnamed)
|vpiName:unnamed
|vpiTypedef:
\_array_typespec: (net_bit)
  |vpiParent:
  \_design: (unnamed)
  |vpiName:net_bit
  |vpiInstance:
  \_design: (unnamed)
  |vpiElemTypespec:
  \_bit_typespec: , line:1:9, endln:1:12
    |vpiSigned:1
|vpiTaskFunc:
\_function: (my_function), line:3:1, endln:4:12
  |vpiParent:
  \_design: (unnamed)
  |vpiName:my_function
  |vpiMethod:1
  |vpiVisibility:1
  |vpiAutomatic:1
  |vpiReturn:
  \_bit_var: , line:3:20, endln:3:23
    |vpiTypespec:
    \_bit_typespec: , line:3:20, endln:3:23
      |vpiSigned:1
  |vpiParent:
  \_design: (unnamed)
===================
```

Suddenly the datatype is promoted as an array ... this does not look as example provide chapter 37.23
Also I do not understand very well 37.16. in particular regarding chapter 6.6.7 : how to retrieve the resolution function ?

alaindargelas, 2023-01-13:
Fixed by #3428, I extended the schema to have a function pointer in the typespec

closed 2023-01-13 by alaindargelas