chipsalliance/Surelog #3379: Incorrect [ERR:UH0705]  Unsupported data type / interconnect

Thomasb81, 2022-12-12:
On following code:
```
module a(
input interconnect b
	);
endmodule
```

The tool with following command:
`surelog -parse -d uhdm  ../ipxact-import/build/interconnect.sv`
produce :
`[ERR:UH0705] interconnect.sv:2:20: Unsupported data type "<n<> u<5> t<Implicit_data_type> p<6> l<2:20> el<2:20>> input interconnect b".`
interconnect is not a data type, see chapter: **6.6.8 Generic interconnect**

Surprisingly with ` -nobuiltin ` the error disappear... This behavior is very odd to me. According to the online help I won't expect this, because it is not buildin classes (array...) 

By going through the LRM, I note:
- (**23.2.2.1**) It shall be illegal to specify signed for a port declared as an interconnect port. (The grammar allow it)
- (**23.2.2.3**) If the data type is omitted, it shall default to logic, except for interconnect ports, which have no data type.
- (**23.2.2.3**) If the previous port was an interconnect port, this port shall also be an interconnect port.
- (**23.2.2.4**) A default shall not be specified for a port declared as interconnect 

There is other notes but they refer to elaboration step.
 
additional legal examples:
```
module a( n);
  input interconnect n; 
endmodule

module b( n);
  input n;
  interconnect n; 
endmodule

module c( input interconnect n,m);
endmodule

module d(n,m);
  input n,m;
  interconnect n;
  interconnect m;
endmodule

```

Illegal examples:

```
module a(
  input interconnect signed n
);
endmodule

module b(
  input interconnect n = 1'b1
);
endmodule

module c(n);
  input n;
  interconnect signed n;
endmodule

module d(n);
  input interconnect n = 1'b1;
endmodule

module d(n);
  input n;
  interconnect n = 1'b1;
endmodule

```

alaindargelas, 2022-12-13:
It is unsupported. It is parsed but not modeled. I welcome a PR!

Thomasb81, 2022-12-13:
I look at that but for the moment I do not understand the code...

Thomasb81, 2022-12-13:
I have difficulties to understand the way the antlr ast is used. Apparently you did differently than what is done in hdlConvertor for example.

Thomasb81, 2022-12-14:
```
module dut( input interconnect net);
endmodule
```

```
design: (work@dut)
|vpiName:work@dut
|uhdmallModules:
\_module: work@dut (work@dut), interconnect.sv, line:1:1, endln:2:10
  |vpiParent:
  \_design: (work@dut)
  |vpiFullName:work@dut
  |vpiDefName:work@dut
  |vpiNet:
  \_logic_net: (work@dut.net), line:1:32, endln:1:35
    |vpiParent:
    \_module: work@dut (work@dut), interconnect.sv, line:1:1, endln:2:10
    |vpiName:net
    |vpiFullName:work@dut.net
    |vpiNetType:12
  |vpiPort:
  \_port: (net), line:1:32, endln:1:35
    |vpiParent:
    \_module: work@dut (work@dut), interconnect.sv, line:1:1, endln:2:10
    |vpiName:net
    |vpiDirection:1
    |vpiLowConn:
    \_ref_obj: 
      |vpiActual:
      \_logic_net: (work@dut.net), line:1:32, endln:1:35
```

Is this data structure correct  for provided SV code ? `vpiNetType:12` correspond to `#define vpiNone` which is actually missing net type in generated code.

alaindargelas, 2023-01-14:
Yes, that is acceptable. I'm working on it.

alaindargelas, 2023-01-15:
Fixed by #3432

closed 2023-01-15 by alaindargelas