chipsalliance/Surelog #3399: Side effects of functions are ignored in uhdmtopModules

mandrys, 2022-12-21:
Since we get a constant value in _vpiConstAssign_ instead of a func call in _uhdmtopModules_, like here:

```
   |vpiContAssign:
   \_cont_assign: , line:23:9, endln:23:32
     |vpiRhs:
     \_constant: 
       |vpiParent:
       \_cont_assign: , line:23:9, endln:23:32
       |vpiDecompile:10
       |vpiSize:4
       |UINT:10
       |vpiConstType:9
     |vpiLhs:
     \_ref_obj: (out1), line:23:9, endln:23:13
       |vpiParent:
       \_cont_assign: , line:23:9, endln:23:32
       |vpiName:out1
```

side assignments done in the function, that don't take part in the func result calculation, are not performed. Here it goes an example:

```
module const_fold_func_top(
    input wire [3:0] inp,
    output wire [3:0] out1,
    output reg [3:0] out2
);

    function automatic [3:0] flip;
        input [3:0] inp;
        flip = ~inp;
    endfunction

  function automatic [3:0] pow_flip_b;
    input [3:0] base, exp;
    begin
        out2[exp] = base & 1;
        pow_flip_b = 1;
        if (exp > 0)
            pow_flip_b = base * pow_flip_b(flip(base), exp - 1); 
    end 
  endfunction

    assign out1 = pow_flip_b(2, 2); 
endmodule
```
In this case, `out2` assignment isn't performed every time we call the function, but it should be.
It is well visible in the diff of dumps after the yosys-systemverilog plugin between the case in which we read only _uhdmtopModules_ (left side) and the case in which we read both, _uhdmallModules_ and _uhdmtopModules_ (right side):
[image]

alaindargelas, 2022-12-23:
Functions with side effects should not be reduced then.

alaindargelas, 2023-01-03:
Fixed by #3411

closed 2023-01-03 by alaindargelas

alaindargelas, 2023-01-04:
And #3413