chipsalliance/Surelog #3449: Mark final statement as non-synthesizable

robertszczepanski, 2023-01-26:
Documentation describes `final` statement as a simulation only keyword:
_**9.2.3 Final procedures**_
_The final procedure is like an initial procedure, defining a procedural block of statements, except that it occurs at the end of simulation time and executes without delays. A final procedure is typically used to display statistical information about the simulation.
The only statements allowed inside a final procedure are those permitted inside a function declaration, so that they execute within a single simulation cycle._

There are no logic assignments inside and it has no purpose outside a simulation hence it should be a part of `nonSynthesizableObjects` set.

Currently it's not passed through a shared class.

alaindargelas, 2023-01-30:
@robertszczepanski, Please add a testcase

robertszczepanski, 2023-01-30:
@alaindargelas - The simplest design would be:
```verilog
module top();
	final begin
		$display("Finished");
	end
endmodule
```
In Yosys with Systemverilog plugin it doesn't fall into [non-synthesizable check](https://github.com/chipsalliance/yosys-f4pga-plugins/blob/d843c714ee77860901ce57ea039f0c744008108e/systemverilog-plugin/UhdmAst.cc#L4274-L4280).

```
yosys> plugin -i systemverilog

yosys> read_systemverilog top.sv 
1. Executing Verilog with UHDM frontend.
[INF:CM0023] Creating log file /hdd/yosys-systemverilog/slpp_all/surelog.log.

[WRN:PA0205] /hdd/yosys-systemverilog/top.sv:1:1: No timescale set for "top".

[INF:CP0300] Compilation...

[INF:CP0303] /hdd/yosys-systemverilog/top.sv:1:1: Compile module "work@top".

[INF:CP0302] Compile class "work@mailbox".

[INF:CP0302] Compile class "work@process".

[INF:CP0302] Compile class "work@semaphore".

[INF:EL0526] Design Elaboration...

[NTE:EL0503] /hdd/yosys-systemverilog/top.sv:1:1: Top level module "work@top".

[NTE:EL0508] Nb Top level modules: 1.

[NTE:EL0509] Max instance depth: 1.

[NTE:EL0510] Nb instances: 1.

[NTE:EL0511] Nb leaf instances: 0.

[INF:UH0706] Creating UHDM Model...

[  FATAL] : 0
[ SYNTAX] : 0
[  ERROR] : 0
[WARNING] : 1
[   NOTE] : 5
Warning: /hdd/yosys-systemverilog/top.sv:1: Skipping non-synthesizable object of type 'module_inst'
ERROR: :0: Encountered unhandled object '' of type 'final_stmt'
```

alaindargelas, 2023-01-31:
There is something wrong with your setup, as you can see below, the error is issued that this is a non-synthesizable construct.
Are you missing the -synth option for Surelog?

COMMAND: -parse tests/ScratchPad.sv -d inst -d ast -d uhdm -d vpi_ids -replay -nocache -nobuiltin **-synth**

n<> u<0> t<_INVALID_> f<0> l<0:0>
n<> u<1> t<Null_rule> p<31> s<30> l<2:1> el<1:2>
n<> u<2> t<Module_keyword> p<6> s<3> l<2:1> el<2:7>
n<top> u<3> t<StringConst> p<6> s<5> l<2:8> el<2:11>
n<> u<4> t<Port> p<5> l<2:12> el<2:12>
n<> u<5> t<List_of_ports> p<6> c<4> l<2:11> el<2:13>
n<> u<6> t<Module_nonansi_header> p<28> c<2> s<27> l<2:1> el<2:14>
n<> u<7> t<Dollar_keyword> p<14> s<8> l<4:3> el<4:4>
n<display> u<8> t<StringConst> p<14> s<13> l<4:4> el<4:11>
n<"Finished"> u<9> t<StringLiteral> p<10> l<4:12> el<4:22>
n<> u<10> t<Primary_literal> p<11> c<9> l<4:12> el<4:22>
n<> u<11> t<Primary> p<12> c<10> l<4:12> el<4:22>
n<> u<12> t<Expression> p<13> c<11> l<4:12> el<4:22>
n<> u<13> t<List_of_arguments> p<14> c<12> l<4:12> el<4:22>
n<> u<14> t<Subroutine_call> p<15> c<7> l<4:3> el<4:23>
n<> u<15> t<Subroutine_call_statement> p<16> c<14> l<4:3> el<4:24>
n<> u<16> t<Statement_item> p<17> c<15> l<4:3> el<4:24>
n<> u<17> t<Statement> p<18> c<16> l<4:3> el<4:24>
n<> u<18> t<Statement_or_null> p<20> c<17> s<19> l<4:3> el<4:24>
n<> u<19> t<End> p<20> l<5:1> el<5:4>
n<> u<20> t<Seq_block> p<21> c<18> l<3:7> el<5:4>
n<> u<21> t<Statement_item> p<22> c<20> l<3:7> el<5:4>
n<> u<22> t<Statement> p<23> c<21> l<3:7> el<5:4>
n<> u<23> t<Final_construct> p<24> c<22> l<3:1> el<5:4>
n<> u<24> t<Module_common_item> p<25> c<23> l<3:1> el<5:4>
n<> u<25> t<Module_or_generate_item> p<26> c<24> l<3:1> el<5:4>
n<> u<26> t<Non_port_module_item> p<27> c<25> l<3:1> el<5:4>
n<> u<27> t<Module_item> p<28> c<26> l<3:1> el<5:4>
n<> u<28> t<Module_declaration> p<29> c<6> l<2:1> el<6:10>
n<> u<29> t<Description> p<30> c<28> l<2:1> el<6:10>
n<> u<30> t<Source_text> p<31> c<29> l<2:1> el<6:10>
n<> u<31> t<Top_level_rule> c<1> l<2:1> el<8:1>
[WRN:PA0205] /home/alain/Surelog/tests/ScratchPad.sv:2:1: No timescale set for "top".

[INF:CP0300] Compilation...

[INF:CP0303] /home/alain/Surelog/tests/ScratchPad.sv:2:1: Compile module "work@top".

[INF:EL0526] Design Elaboration...

Instance tree:
[TOP] work@top work@top
[SCO] work@top.genblk1 work@top.genblk1

[NTE:EL0503] /home/alain/Surelog/tests/ScratchPad.sv:2:1: Top level module "work@top".

[NTE:EL0508] Nb Top level modules: 1.

[NTE:EL0509] Max instance depth: 1.

[NTE:EL0510] Nb instances: 1.

[NTE:EL0511] Nb leaf instances: 0.

[NTE:EL0523] /home/alain/Surelog/tests/ScratchPad.sv:2:1: Instance "work@top".

[NTE:EL0522] /home/alain/Surelog/tests/ScratchPad.sv:3:7: Scope "work@top.genblk1".

[INF:UH0706] Creating UHDM Model...

decompile:
final_stmt: , id:0
|vpiParent:
\_module_inst: work@top (work@top), id:5, file:/home/alain/Surelog/tests/ScratchPad.sv, line:2:1, endln:6:10
|vpiStmt:
\_begin: (work@top), id:1, line:3:7, endln:5:4
  |vpiParent:
  \_final_stmt: , id:0
  |vpiFullName:work@top
  |vpiStmt:
  \_sys_func_call: ($display), id:2, line:4:3, endln:4:23
    |vpiParent:
    \_begin: (work@top), id:1, line:3:7, endln:5:4
    |vpiArgument:
    \_constant: , id:3, line:4:12, endln:4:22
      |vpiParent:
      \_sys_func_call: ($display), id:2, line:4:3, endln:4:23
      |vpiDecompile:Finished
      |vpiSize:64
      |STRING:Finished
      |vpiConstType:6
    |vpiName:$display

**[ERR:UH0720] /home/alain/Surelog/tests/ScratchPad.sv:2:1: Non-synthesizable construct "".**

[INF:UH0708] Writing UHDM DB: /home/alain/Surelog/slpp_all/surelog.uhdm ...

[INF:UH0709] Writing UHDM Html Coverage: /home/alain/Surelog/slpp_all/checker/surelog.chk.html ...

[INF:UH0710] Loading UHDM DB: /home/alain/Surelog/slpp_all/surelog.uhdm ...

[INF:UH0711] Decompiling UHDM...

====== UHDM =======
design: (work@top), id:4
|vpiName:work@top
|uhdmallModules:
\_module_inst: work@top (work@top), id:5, file:/home/alain/Surelog/tests/ScratchPad.sv, line:2:1, endln:6:10
  |vpiParent:
  \_design: (work@top), id:4
  |vpiFullName:work@top
  |vpiDefName:work@top
  |vpiProcess:
  \_final_stmt: , id:0
    |vpiParent:
    \_module_inst: work@top (work@top), id:5, file:/home/alain/Surelog/tests/ScratchPad.sv, line:2:1, endln:6:10
    |vpiStmt:
    \_begin: (work@top), id:1, line:3:7, endln:5:4
      |vpiParent:
      \_final_stmt: , id:0
      |vpiFullName:work@top
      |vpiStmt:
      \_sys_func_call: ($display), id:2, line:4:3, endln:4:23
        |vpiParent:
        \_begin: (work@top), id:1, line:3:7, endln:5:4
        |vpiArgument:
        \_constant: , id:3, line:4:12, endln:4:22
          |vpiParent:
          \_sys_func_call: ($display), id:2, line:4:3, endln:4:23
          |vpiDecompile:Finished
          |vpiSize:64
          |STRING:Finished
          |vpiConstType:6
        |vpiName:$display
|uhdmtopModules:
\_module_inst: work@top (work@top), id:6, file:/home/alain/Surelog/tests/ScratchPad.sv, line:2:1, endln:6:10
  |vpiName:work@top
  |vpiDefName:work@top
  |vpiTop:1
  |vpiTopModule:1
  |vpiProcess:
  \_final_stmt: , id:0
===================
[  FATAL] : 0
[ SYNTAX] : 0
[  ERROR] : 1
[WARNING] : 1
[   NOTE] : 7

alaindargelas, 2023-01-31:
Fixed by #3455

closed 2023-01-31 by alaindargelas