chipsalliance/Surelog #3458: Q: Which diffs in logs matter ?

hzeller, 2023-02-03:
(low prio)

Running the tests and then manually looking at the logs, often time they differ, yet the regression table considers them PASS. Are the VObject prints just ignored in the diff output ? Should the logs be updated ?

Here an example of the first test, comparing the checked-in test logs with the build output:

```diff
diff -u ./tests/1364_2005/1364_2005.log ./build/regression/1364_2005/1364_2005.log 
--- ./tests/1364_2005/1364_2005.log     2023-01-06 09:40:06.308924001 -0800
+++ ./build/regression/1364_2005/1364_2005.log  2023-02-02 15:46:06.924048552 -0800
@@ -3,12 +3,12 @@
 LIB:  work
 FILE: ${SURELOG_DIR}/tests/1364_2005/dut.v
 n<> u<0> t<_INVALID_> f<0> l<0:0>
-n<> u<1> t<Null_rule> p<88> s<87> l<2:1> el<1:2>
+n<> u<1> t<Null_rule> p<89> s<88> l<2:1> el<1:2>
 n<> u<2> t<Top_directives> p<3> l<2:1> el<2:28>
-n<> u<3> t<Description> p<87> c<2> s<84> l<2:1> el<2:28>
-n<> u<4> t<Module_keyword> p<6> s<5> l<3:1> el<3:7>
+n<> u<3> t<Description> p<88> c<2> s<85> l<2:1> el<2:28>
```

alaindargelas, 2023-02-04:
Logs don't get updated with each PR if the regression says there is no DIFF (I do, but other contributors don't).
The regression is not checking for the AST changes. It checks for messages, UHDM diffs and roundtrip output changes, only then it says there is a DIFF.
When I update the grammar, the AST changes can be huge, and I update the all the logs, but typically, only a few tests show a DIFF (# of messages, UHDM tree dump change or rountrip change).

When running "python3 scripts/regression.py update" I think it only updates the tests that have a diff, and some logs don't get systematically updated if they only have an AST diff.
We should ask @hs-apotell to add an option to his script to force the update.

hzeller, 2023-02-08:
so if the AST changes don't matter (the `n<> u<1> t<Null_rule> p<88> s<87> l<2:1> el<1:2>` kind), wouldn't it be a good idea to not check them in in the first place to reduce churn (and headaches that triggered my original question) ?

closed 2023-03-03 by alaindargelas