chipsalliance/Surelog #3460: Can not assign localparam array with function and $info with arguments !

forthyen, 2023-02-07:
module top #(
)(
);
    typedef int unsigned ASSIGN_VADDR_RET_T[2];
    function static ASSIGN_VADDR_RET_T ASSIGN_VADDR();
        for (int i = 0; i < 2; i++) begin
             ASSIGN_VADDR[i] = 5;
        end
    endfunction

    localparam int unsigned VADDR[2] = ASSIGN_VADDR();

	if (VADDR[0] != 5) begin
		$info("--[0] (%d) should be 5",VADDR[0]);
	end;
	if (VADDR[1] != 5) begin
		$info("--[1] (%d) should be 5",VADDR[1]);
	end;
	
endmodule

module main;
    top #() top1;
endmodule
[test.zip](https://github.com/chipsalliance/Surelog/files/10671423/test.zip)

alaindargelas, 2023-02-07:
First problem, you need to change this line:
top #() top1;
Into:
top #() top1 ();

So this is interpreted as an instantiation.

forthyen, 2023-02-07:
Thanks,
I surelog -parse test.sv, and get 

.....
[INF:EL0549] /mnt/e/yosys-systemverilog/test.sv:14:3: Elaboration info "--[0] (%d) should be 5".

[INF:EL0549] /mnt/e/yosys-systemverilog/test.sv:17:3: Elaboration info "--[1] (%d) should be 5".
........

===============
1. VADDR[0] should be 5, but I get 1
2. $info("%d",VADDR[0]) sould be "Elaboration info 5", but I get "Elaboration info %d"
=========
Thanks,

alaindargelas, 2023-02-11:
#3464 addresses function returning an array and the generate statement evaluation.
It does not address the printf string expansion in the $info func yet.

forthyen, 2023-02-11:
Thank you very much for your support !

forthyen, 2023-02-11:
Would you please test this code ?

`module top #(
)(
);
	localparam int unsigned CNT    = 2;
//	localparam int unsigned V[CNT] = '{3,3};
	localparam int unsigned V[CNT] = '{defaut: 3};
	
    typedef int unsigned ASSIGN_VADDR_RET_T[CNT];
    function static ASSIGN_VADDR_RET_T ASSIGN_VADDR();
        for (int i = 0; i < CNT; i++) begin
             ASSIGN_VADDR[i] = V[i];
        end
    endfunction

    localparam int unsigned VADDR[CNT] = ASSIGN_VADDR();

	if (VADDR[0] != 5) begin
		$info("--[0] (%d) should be 5",VADDR[0]);
	end;
	if (VADDR[1] != 5) begin
		$info("--[1] (%d) should be 5",VADDR[1]);
	end;
	
endmodule

module main;
    top #() top1();
endmodule
`

localparam int unsigned V[CNT] = '{defaut: 3}; // not equal '{3,3}

[test.zip](https://github.com/chipsalliance/Surelog/files/10713818/test.zip)

alaindargelas, 2023-02-13:
ocalparam int unsigned V[CNT] = '{defaut: 3}; is 
Fixed by #3467 
Note the testcase I checked in in the PR is slightly different than what you asked.

alaindargelas, 2023-02-13:
As I said above, this is low priority. If you need it fixed soon, please submit a PR, else it might be weeks before I fix this.

forthyen, 2023-02-13:
It is not related to "printf" problem but the accessibility about param V! 

------
localparam int unsigned V[CNT] = '{default: 3};

if (V[0] == 5) begin
   $info("V=5");
end
if (V[0] != 5) begin
   $info("V!=5");
end
------

I can get "-Info: test.sv:21:9: V!=5" info message!

but

I can not get 
   "Elaboration info "V!=5"
or
  "Elaboration info "V==5"

forthyen, 2023-02-13:
V[CNT] = '{default: 3};

I can get info message in verilator bit not in surelog !

---
change to 
V[CNT] = '{3,3};

then 
I can get info message in surelog now !

alaindargelas, 2023-02-14:
Previous PR was working for Packages.
This PR #3470 solves it for modules.

Your testcase has a typo, you misspelled default, it cannot work with the typo:

localparam int unsigned V[CNT] = '{defaut: 3}; // not equal '{3,3}
Should be
localparam int unsigned V[CNT] = '{defau**l**t: 3}; // not equal '{3,3}

forthyen, 2023-02-14:
Yes, I find this typing error! But I correct it, this error also!

I  use yosys-systemverilog ...
  yosys
  > plugin -i systemverilog
  > read_systemverilog test.sv

then I get

---
ERROR: Couldn't find parameter type for node: \V
---

Maybe \V is undefined in UHDM file?

How can i check it !

alaindargelas, 2023-02-14:
Please open that issue in the Yosys plugin github repo, not here. Different developers will help you there.

alaindargelas, 2023-02-15:
There is another issue (from the plugin on above) in Surelog. I'm working on it.

alaindargelas, 2023-02-16:
https://github.com/chipsalliance/Surelog/pull/3471 fixes the issue when the parameter is declared in a module instead of a package.

closed 2023-02-16 by alaindargelas

forthyen, 2023-02-16:
Thank you very much !  It's better now!

Wish you have more time to see the next condition !

`
module top #(
)(
);
	localparam int unsigned CNT    = 2;
//	localparam int unsigned V[CNT] = '{3,3};
	localparam int unsigned V[CNT] = '{default: 3};
	
	typedef int unsigned ASSIGN_VADDR_RET_T[CNT];
	function static ASSIGN_VADDR_RET_T ASSIGN_VADDR();
	   for (int i = 0; i < CNT; i++) begin
		ASSIGN_VADDR[i] = V[i];
           end
	endfunction

	localparam int unsigned VADDR[CNT] = ASSIGN_VADDR();
	if (VADDR[0] == 5) begin
	   $info("VADDR[0] == 5");
	end
	if (VADDR[0] != 5) begin
	   $info("VADDR[0] != 5");
	end
	
    function static int unsigned MAX_VADDR_CNT(int unsigned SRC[CNT], int unsigned ICNT);
        MAX_VADDR_CNT = 0;
        for (int i = 0; i < ICNT; i++) begin
            if (SRC[i] > MAX_VADDR_CNT) begin
                MAX_VADDR_CNT = SRC[i];
            end
        end
    endfunction

	localparam int unsigned MAX_VADDR = MAX_VADDR_CNT(VADDR, CNT);	
	if (MAX_VADDR == 5) begin
	   $info("MAX_VADDR == 5");
	end
	if (MAX_VADDR != 5) begin
	   $info("MAX_VADDR != 5");
	end
	
endmodule

module main;
    top #() top1();
endmodule
`

MAX_VADDR can not be $info'ed now !

Thanks,

[test.zip](https://github.com/chipsalliance/Surelog/files/10752419/test.zip)

alaindargelas, 2023-02-19:
Fixed by #3473.
If you have new failing testcases, please create a new issue instead of appending to this issue.