chipsalliance/Surelog #3465: vpiDecompile on Expression

Thomasb81, 2023-02-11:
Hello

on
```
module dut #(parameter SIZE) (
  input wire [SIZE-1:0] wire_i
	);
endmodule
```
the UHDM tree for the left part of range `[SIZE-1:0]`  produce:
```
        |vpiLeftRange:
        \_operation: , line:2:13, endln:2:19
          |vpiParent:
          \_range: , line:2:12, endln:2:22
          |vpiOpType:11
          |vpiOperand:
          \_ref_obj: (work@dut.wire_i.SIZE), line:2:13, endln:2:17
            |vpiParent:
            \_port: (wire_i), line:2:23, endln:2:29
            |vpiName:SIZE
            |vpiFullName:work@dut.wire_i.SIZE
          |vpiOperand:
          \_constant: , line:2:18, endln:2:19
            |vpiParent:
            \_operation: , line:2:13, endln:2:19
            |vpiDecompile:1
            |vpiSize:64
            |UINT:1
            |vpiConstType:9
```

While I would have expect to see print in object tree in addition:
vpiDecompile:SIZE-1
then
vpiecompile:SIZE

for intermediate expression structure.

According to chapter 37.57, expr should allow to retrieve an equivalent to source code string with vpiDecompile.

I try to add the raw expression coming from antlr rule context to UHDM db by amending VObject but cache mechanic (https://github.com/chipsalliance/Surelog/blob/079abd2e6d114fee8f0bcf058aac2b07a5b8a25f/src/Cache/Cache.cpp#L222) let my think it was not plan to work this way.

An other strategy would be for vpiDecompile on expression to "re-elaborate" an expression with children data. I struggle to understand vpi object model to achieve that ...

alaindargelas, 2023-02-11:
@Thomasb81 The dump you show above is not the vpiDecompile. It is the UHDM tree dump used to debug UHDM model. It contains much more information than the standard vpiDecompile.

To achieve what you want (Decompilation from UHDM/VPI model), you need to work on https://github.com/chipsalliance/UHDM/blob/10663bd899d5e7cff7b741bdc346ed47d26fcf4b/templates/ExprEval.cpp#L733
The first thing would be to move this function in it's own file as it will become large very quickly. This is essentially pretty printing the UHDM model back into Verilog. The current sample code is alredy used to vpiDecompile simple expressions.

Thomasb81, 2023-02-12:
> @Thomasb81 The dump you show above is not the vpiDecompile. It is the UHDM tree dump used to debug UHDM model. It contains much more information than the standard vpiDecompile.

The text from the uhdm-dump come from this method of generated vpi_visitor.cpp:
```
static void visit_expr(vpiHandle obj_h, int indent, const char *relation, VisitedContainer* visited, std::ostream& out, bool shallowVisit) {
  if (const char* s = vpi_get_str(vpiDecompile, obj_h))
    stream_indent(out, indent) << "|vpiDecompile:" << s << "\n";
  if (const int n = vpi_get(vpiSize, obj_h))
    stream_indent(out, indent) << "|vpiSize:" << n << "\n";
  s_vpi_value value;
  vpi_get_value(obj_h, &value);
  if (value.format) {
    std::string val = visit_value(&value);
    if (!val.empty()) {
      stream_indent(out, indent) << val;
    }
  }
  if (vpiHandle itr = vpi_handle(vpiTypespec, obj_h)) {
    visit_object(itr, indent + kLevelIndent, "vpiTypespec", visited, out, false);
    release_handle(itr);
  }
}
```

> To achieve what you want (Decompilation from UHDM/VPI model), you need to work on https://github.com/chipsalliance/UHDM/blob/10663bd899d5e7cff7b741bdc346ed47d26fcf4b/templates/ExprEval.cpp#L733
> The [...]

I do not understand what this code does and how to use it. Could you provide additional pointer on where it is used ?

But yes, the `switch (opType)` case can be significantly long.

In my sandbox I achieve to sketch a visitor class based on the recursive mechanic generate in `generated/src/vpi_visitor.cpp`. I change a couple of return value to `std::string_view`/`std::string` that allow me to re-compile what I think `vpiDecompile` should return in `visit_expr` function provide above.

I am now able from this `input wire [SIZE-1:0] wire_i` statement in uhdm db from my initial example to produce
```
            <ipxact:port>
                <ipxact:name>wire_i</ipxact:name>
                <ipxact:wire>
                    <ipxact:direction>input</ipxact:direction>
                    <ipxact:vectors>
                        <ipxact:vector>
                            <ipxact:left>SIZE-1</ipxact:left>
                            <ipxact:right>0</ipxact:right>
                        </ipxact:vector>
                    </ipxact:vectors>
                </ipxact:wire>
            </ipxact:port>
```

alaindargelas, 2023-02-12:
Let's start from the top. It will be easier to guide you.
You are trying to write an IP Exact extractor from module definition ? 
- For all modules in a design?  
- For a single module?

Thomasb81, 2023-02-12:
> If the answer is for a single module (The top module of an IP), and no elaboration (Which I think is the right way to extract the IP exact definition), then

Top module of an IP for parameter and ports, but the full IP to get the list of files.
Can't escape to cover all the ast to grab all files...

> But if you complete it, it will write the expression as you have achieved ipxact:leftSIZE-1</ipxact:left>

Ok I will do.

Thomasb81, 2023-02-12:
I open a draft pull  request https://github.com/chipsalliance/UHDM/pull/867

alaindargelas, 2023-02-13:
> Top module of an IP for parameter and ports, but the full IP to get the list of files.
> Can't escape to cover all the ast to grab all files...

You can go through the list of uhdmAllModules and uhdmAllPackages and get the file list.
you don't need to go through uhdmTopModules or uhdmTopPackages.

alaindargelas, 2023-02-14:
Merged https://github.com/chipsalliance/UHDM/pull/867

Thomasb81, 2023-02-18:
Still for the intent to complete `exprEval::prettyPrint` to capture expression, I create following test case to capture` $sformatf("%d",b)` :
```
module dut();
localparam a = $sformatf("%d",b);
endmodule
```

The uhdm dump create following tree:
```
    |vpiRhs:
    \_sys_func_call: ($sformatf), line:2:16, endln:2:33
      |vpiArgument:
      \_constant: , line:2:26, endln:2:30
        |vpiParent:
        \_sys_func_call: ($sformatf), line:2:16, endln:2:33
        |vpiDecompile:%d
        |vpiSize:16
        |STRING:%d
        |vpiConstType:6
      |vpiArgument:
      \_ref_obj: (b), line:2:31, endln:2:32
        |vpiParent:
        \_sys_func_call: ($sformatf), line:2:16, endln:2:33
        |vpiName:b
        |vpiActual:
        \_logic_net: (b)
      |vpiName:$sformatf
```

I do not understand why my format constant string is capture as `%d` instead of `"%d"`...
Tentative to use `vpiDecompile("\"%d\"")` in my unittest produce for prettyPrint output `\"%d\"`. 

So I can produce ` $sformatf(%d,b)` or ` $sformatf(\"%d\",b)` depending of what how I fill `vpiDecompile` for my constant string.

It is not clear what happen here. I expect as any other constant a direct use : https://github.com/chipsalliance/UHDM/blob/7d0cd8dbaf3738559cddcf78bcdf04515067c311/templates/ExprEval.cpp#L741

I do not understand who has the responsibility to add the double quote in case of manipulation of vpiDecompile string constant or why surelog do not fill the db with a directly usable string constant. 

Regarding `vpiDecompile("\"%d\"")`
Why my backslash protection of double quote is being it self protected ? Is there C++/STL black magic here ?

alaindargelas, 2023-02-19:
Maybe @hs-apotell can help here.
It looks like the VpiDecompile method is printing as-is what was passed a an argument regardless of the type.
In the case of a constant string type, it should print "" around the value.

Thomasb81, 2023-02-19:
```
    case VObjectType::slStringLiteral: {
      UHDM::constant *c = s.MakeConstant();
      std::string_view value = StringUtils::unquoted(fC->SymName(child));
      c->VpiDecompile(value);
      c->VpiSize(value.length() * 8);
      c->VpiValue(StrCat("STRING:", value));
      c->VpiConstType(vpiStringConst);
      result = c;
      break;
    }
```

My believe is that:

- vpiDecompile should record the quoted string, so prettyPrint can work recursively without taking care of leaf element. At least that is `evalExpr::prettyPrint` is written today.

- vpiValue should record the unquoted string to allow recursive evaluation of expression ie string concatenation without to have any need of demangle  the value before use.

This could probably be apply to other kind of constant. Except that value for other kind of constant than string still need side attribute, because the kind is not c++ built-in. 

At this stage of the project it can have impact on application code: 
https://github.com/chipsalliance/Surelog/blob/0560506b3f9bcd4b6ff698503f5c062543ce1fdd/src/roundtrip.cpp#L388
https://github.com/chipsalliance/UHDM/blob/785fb0052bdd70362f62ca23fc8e128a95678673/templates/ExprEval.cpp#L1590

At least for me it seems very odd that I need to demangle the value for any use coming from the database for all application.

Thomasb81, 2023-02-19:
https://github.com/chipsalliance/Surelog/blob/0560506b3f9bcd4b6ff698503f5c062543ce1fdd/src/DesignCompile/CompileExpression.cpp#L712

Chapter 37.17 Details 9), page 992:
[image]

For `vpiStringConst` **vpiSize** should return the number of character

`c->VpiSize(value.length() * 8);`
should be
`c->VpiSize(value.length());`

Thomasb81, 2023-02-22:
Regarding this construction coming from  https://github.com/chipsalliance/Surelog/issues/3480 in the context of this ticket dealing with vpiDecompile
```
logic [31:0] data;
assign data = '{ default: 0 };
```

I figure out that there is no api to store `default` statement (see example given in chapter  37.57 6) the default value must be replace when the database is initially fill. So the associated vpiDecompile value must be fill with original source code token.
No way to build it by a later elaboration pass using the result of prettyPrint call.

Can we retrieve the source corresponding to the expression grammar rule in [Surelog/src/DesignCompile/CompileExpression.cpp](https://github.com/chipsalliance/Surelog/blob/0560506b3f9bcd4b6ff698503f5c062543ce1fdd/src/DesignCompile/CompileExpression.cpp) code, similarly to how we can retrieve it in antlr ast ?

alaindargelas, 2023-02-27:
@Thomasb81, can you open different issues with each topic discussed here? 
- Have you and @hs-apotell reached a conclusion for the string double-quotes?
- please open an issue regarding the string size
- please open an issue with the default statement so it does not get mingled in this case discussion (The answer being that the vpiAllModule contains the default original statement, and in the vpiTopModule it has been replaced by the resulting constant).

Thomasb81, 2023-02-27:
> @Thomasb81, can you open different issues with each topic discussed here?

I will do that

>     * Have you and @hs-apotell reached a conclusion for the string double-quotes?
>     * please open an issue regarding the string size

I need to have a deeper look at https://github.com/chipsalliance/Surelog/blob/0560506b3f9bcd4b6ff698503f5c062543ce1fdd/src/DesignCompile/CompileExpression.cpp#L710
See if a PR is achievable at my end or open an issue as you suggest.

>     * please open an issue with the default statement so it does not get mingled in this case discussion (The answer being that the vpiAllModule contains the default original statement, and in the vpiTopModule it has been replaced by the resulting constant).
I will open an issue on this one.

alaindargelas, 2023-03-03:
This particular issue is resolved.

closed 2023-03-03 by alaindargelas