chipsalliance/Surelog #3472: Q: What's the best way to compute the width of a uhdmoperation?

mysoreanoop, 2023-02-16:
Aim: To compute width of vpiCondition within a UHDM::uhdmcase_stmt

So far: I'm switching based on vpiCondition being part_sel, bit_sel, ref_obj, expr, operation, hier_path, etc., and computing the widths at the leaves and summing them at various intermediate nodes.

Seeking: Is there an existing function or better/simpler way to do this?

Thomasb81, 2023-02-16:
Are you trying to evaluate an expression ? if yes I would refer to chapter 11.6 Expression bit lengths of SV standard.

mysoreanoop, 2023-02-16:
Table 11-21 in there is what I was looking for, thanks!

Is there an existing function that I can leverage or modify for implementing this?

alaindargelas, 2023-02-17:
https://github.com/chipsalliance/UHDM/blob/7d0cd8dbaf3738559cddcf78bcdf04515067c311/templates/ExprEval.h#L60

I suppose you are using the VPI API.
From any VPI handle, you can get it's UHDM C++ object:
include https://github.com/chipsalliance/UHDM/blob/master/templates/vpi_uhdm.h

vpiHandle handle  = .....

any* uhdmHandle = ((uhdm_handle*)handle)->object;

(Someone has to submit a PR in UHDM to create a function that does that conversion)

Then use the UHDM:ExprEval::size( ..... ) method to evaluate the size of the condition which is an expression.

alaindargelas, 2023-02-27:
@mysoreanoop do you need more help with this issue?

mysoreanoop, 2023-02-28:
Oh right, sorry I wasn't thinking. It seems to be working, but sometimes incorrectly, thanks!

For example, when there's a case condition which is an `input logic [2:0]`, it seems to be printing 1 always (instead of 3). And, when a condition is a `logic enum [2:0] {...} state`, for example, it's printing 0 (incorrect).

closed 2023-02-28 by mysoreanoop