chipsalliance/Surelog #3481: can't compile in Linux, Windows or Cygwin

bluehalo-user, 2023-02-24:
I am sure I am doing something wrong, so this issue is maybe just a request for better documentation.

In Linux (Ubuntu 18.04), the first of many compile errors:
```
[ 13%] Building CXX object third_party/antlr4/runtime/Cpp/runtime/CMakeFiles/antlr4_static.dir/src/IntStream.cpp.o
In file included from /usr/include/c++/7/bits/hashtable.h:35:0,
                 from /usr/include/c++/7/unordered_map:47,
                 from /home/slittle/sandbox/Surelog/third_party/antlr4/runtime/Cpp/runtime/src/antlr4-common.h:31,
                 from /home/slittle/sandbox/Surelog/third_party/antlr4/runtime/Cpp/runtime/src/Exceptions.h:8,
                 from /home/slittle/sandbox/Surelog/third_party/antlr4/runtime/Cpp/runtime/src/RecognitionException.h:8,
                 from /home/slittle/sandbox/Surelog/third_party/antlr4/runtime/Cpp/runtime/src/NoViableAltException.h:8,
                 from /home/slittle/sandbox/Surelog/third_party/antlr4/runtime/Cpp/runtime/src/DefaultErrorStrategy.cpp:6:
/usr/include/c++/7/bits/hashtable_policy.h: In instantiation of ‘struct std::__detail::__is_noexcept_hash<std::pair<const antlr4::atn::PredictionContext*, const antlr4::atn::PredictionContext*>, std::hash<std::pair<const antlr4::atn::PredictionContext*, const antlr4::atn::PredictionContext*> > >’:
/usr/include/c++/7/type_traits:143:12:   required from ‘struct std::__and_<std::__is_fast_hash<std::hash<std::pair<const antlr4::atn::PredictionContext*, const antlr4::atn::PredictionContext*> > >, std::__detail::__is_noexcept_hash<std::pair<const antlr4::atn::PredictionContext*, const antlr4::atn::PredictionContext*>, std::hash<std::pair<const antlr4::atn::PredictionContext*, const antlr4::atn::PredictionContext*> > > >’
/usr/include/c++/7/type_traits:154:31:   required from ‘struct std::__not_<std::__and_<std::__is_fast_hash<std::hash<std::pair<const antlr4::atn::PredictionContext*, const antlr4::atn::PredictionContext*> > >, std::__detail::__is_noexcept_hash<std::pair<const antlr4::atn::PredictionContext*, const antlr4::atn::PredictionContext*>, std::hash<std::pair<const antlr4::atn::PredictionContext*, const antlr4::atn::PredictionContext*> > > > >’
/usr/include/c++/7/bits/unordered_map.h:103:66:   required from ‘class std::unordered_map<std::pair<const antlr4::atn::PredictionContext*, const antlr4::atn::PredictionContext*>, std::unique_ptr<antlr4::atn::PredictionContextMergeCache::Entry>, std::hash<std::pair<const antlr4::atn::PredictionContext*, const antlr4::atn::PredictionContext*> >, std::equal_to<std::pair<const antlr4::atn::PredictionContext*, const antlr4::atn::PredictionContext*> >, std::allocator<std::pair<const std::pair<const antlr4::atn::PredictionContext*, const antlr4::atn::PredictionContext*>, std::unique_ptr<antlr4::atn::PredictionContextMergeCache::Entry> > > >’
/home/slittle/sandbox/Surelog/third_party/antlr4/runtime/Cpp/runtime/src/atn/PredictionContextMergeCache.h:88:85:   required from here
/usr/include/c++/7/bits/hashtable_policy.h:87:34: error: no match for call to ‘(const std::hash<std::pair<const antlr4::atn::PredictionContext*, const antlr4::atn::PredictionContext*> >) (const std::pair<const antlr4::atn::PredictionContext*, const antlr4::atn::PredictionContext*>&)’
  noexcept(declval<const _Hash&>()(declval<const _Key&>()))>
```
In cygwin, the error looks like a cygwin path issue:
```
[ 60%] Generating generated/src/parser/SV3_1aLexer.h, generated/src/parser/SV3_1aParserBaseListener.h, generated/src/parser/SV3_1aParser.h, generated/src/parser/SV3_1aParserListener.h, generated/src/parser/SV3_1aLexer.cpp, generated/src/parser/SV3_1aParserBaseListener.cpp, generated/src/parser/SV3_1aParser.cpp, generated/src/parser/SV3_1aParserListener.cpp, generated/src/parser/SV3_1aPpLexer.h, generated/src/parser/SV3_1aPpParserBaseListener.h, generated/src/parser/SV3_1aPpParser.h, generated/src/parser/SV3_1aPpParserListener.h, generated/src/parser/SV3_1aPpLexer.cpp, generated/src/parser/SV3_1aPpParserBaseListener.cpp, generated/src/parser/SV3_1aPpParser.cpp, generated/src/parser/SV3_1aPpParserListener.cpp, generated/src/parser/generated-parsers.tstamp
---------- Parser generation...
Error: Unable to access jarfile /cygdrive/c/work/sandbox/surelog/Surelog/third_party/antlr4_bin/antlr-4.10-complete.jar
```
In Windows, I tried two things:
1. Using the command prompt (after installing all the required tools and verifying they are available from the command prompt, and cleaning all previous builds):
```
CUSTOMBUILD : CMake error : failed to create symbolic link 'C:/work/sandbox/surelog/Surelog/compile_commands.json': C annot create a file when that file already exists. [C:\work\sandbox\surelog\Surelog\build\surelog_link_target.vcxproj ]

C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\MSBuild\Microsoft\VC\v160\Microsoft.CppCommon.targets(2 41,5): error MSB8066: Custom build for 'C:\work\sandbox\surelog\Surelog\build\CMakeFiles\c764decd8bf274e0e22863a404e1 3538\surelog_link_target.rule;C:\work\sandbox\surelog\Surelog\CMakeLists.txt' exited with code 1. [C:\work\sandbox\su relog\Surelog\build\surelog_link_target.vcxproj]
make: *** [Makefile:23: release] Error 1
```
2. Generating a Visual Studio project from the CMAKE GUI:
```
Severity	Code	Description	Project	File	Line	Suppression State
Error		failed to create symbolic link 'C:/work/sandbox/surelog/Surelog/compile_commands.json': Cannot create a file when that file already exists.	surelog_link_target	C:\work\sandbox\surelog\Surelog\build\CUSTOMBUILD	1	
Error	MSB8066	Custom build for 'C:\work\sandbox\surelog\Surelog\build\CMakeFiles\c764decd8bf274e0e22863a404e13538\surelog_link_target.rule;C:\work\sandbox\surelog\Surelog\CMakeLists.txt' exited with code 1.	surelog_link_target	C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\MSBuild\Microsoft\VC\v160\Microsoft.CppCommon.targets	241	
Error	MSB8066	Custom build for 'C:\work\sandbox\surelog\Surelog\build\CMakeFiles\dd3b6a2d58fb89270cfec9e4dd7763df\UHDM.capnp.rule;C:\work\sandbox\surelog\Surelog\build\CMakeFiles\dd3b6a2d58fb89270cfec9e4dd7763df\UHDM.capnp.h.rule;C:\work\sandbox\surelog\Surelog\build\CMakeFiles\61c5986cd163dbe4264c453475e3d493\GenerateCode.rule;C:\work\sandbox\surelog\Surelog\third_party\UHDM\CMakeLists.txt' exited with code 9009.	GenerateCode	C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\MSBuild\Microsoft\VC\v160\Microsoft.CppCommon.targets	241	
Error	C1083	Cannot open include file: 'UHDM.capnp.h': No such file or directory	uhdm	C:\work\sandbox\surelog\Surelog\build\third_party\UHDM\generated\src\Serializer_restore.cpp	49	
```
I would sure like to test out Surelog for some things. Can someone tell me what I'm doing wrong?

bluehalo-user, 2023-02-24:
> EXTERNAL
>
> For unix, what version of gcc/clang are you using?
>
> On windows, make sure you have the developer mode turned on. Search for developer mode and you can find instructions on how to toggle it.
>
> -
> CAUTION: This email originated from outside of the organization. Do not click links, reply, or open attachments unless you recognize the sender (and sending email address) and know the content is safe.
>
> The information contained in this e-mail and any attachments from BlueHalo may contain confidential and/or proprietary information, and is intended only for the named recipient to whom it was originally addressed. If you are not the intended recipient, any disclosure, distribution, or copying of this e-mail or its attachments is strictly prohibited. If you have received this e-mail in error, please notify the sender immediately by return e-mail and permanently delete the e-mail and any attachments.

$ gcc-11 --version
gcc-11 (Ubuntu 11.1.0-1ubuntu1~18.04.1) 11.1.0
Copyright (C) 2021 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Obviously that might not be the executable that is being used by the build. I was thinking that CMAKE might be doing the right thing there.

I can research how to force the right executable when multiple versions of GCC are installed.

I will try developer mode on windows.

Thanks.

bluehalo-user, 2023-02-24:
Forcing GCC-11 on Linux helped. I am getting farther now, but I am now seeing this problem:
```
[ 78%] Building CXX object CMakeFiles/surelog.dir/src/DesignCompile/TestbenchElaboration.cpp.o
<path>/Surelog/src/DesignCompile/CompileHelper.cpp: In member function 'UHDM::any* SURELOG::CompileHelper::defaultPatternAssignment(const UHDM::typespec*, UHDM::any*, SURELOG::DesignComponent*, SURELOG::CompileDesign*, SURELOG::ValuedComponentI*)':
<path>/Surelog/src/DesignCompile/CompileHelper.cpp:2823:19: error: 'array_expr' was not declared in this scope
2823 |                   array_expr* array = s.MakeArray_expr();
      |                   ^~~~~~~~~~
<path>/Surelog/src/DesignCompile/CompileHelper.cpp:2823:31: error: 'array' was not declared in this scope; did you mean 'std::array'?
2823 |                   array_expr* array = s.MakeArray_expr();
      |                               ^~~~~
      |                               std::array
In file included from /usr/include/c++/11/tuple:39,
                 from /usr/include/c++/11/bits/hashtable_policy.h:34,
                 from /usr/include/c++/11/bits/hashtable.h:35,
                 from /usr/include/c++/11/unordered_set:46,
                 from /Surelog/build/third_party/UHDM/generated/uhdm/SymbolId.h:9,
                 from /Surelog/include/Surelog/Common/SymbolId.h:5,
                 from /Surelog/include/Surelog/Common/PathId.h:21,
                 from /Surelog/include/Surelog/CommandLine/CommandLineParser.h:28,
                 from /Surelog/src/DesignCompile/CompileHelper.cpp:24:
/usr/include/c++/11/array:95:12: note: 'std::array' declared here
   95 |     struct array
      |            ^~~~~
<path>/Surelog/src/DesignCompile/CompileHelper.cpp:2823:41: error: 'class UHDM::Serializer' has no member named 'MakeArray_expr'; did you mean 'MakeArray_var'?
2823 |                   array_expr* array = s.MakeArray_expr();
      |                                         ^~~~~~~~~~~~~~
      |                                         MakeArray_var
<path>/Surelog/src/DesignCompile/CompileHelper.cpp: In member function 'bool SURELOG::CompileHelper::compileParameterDeclaration(SURELOG::DesignComponent*, const SURELOG::FileContent*, SURELOG::NodeId, SURELOG::CompileDesign*, bool, SURELOG::ValuedComponentI*, bool, bool, bool)':
<path>/Surelog/src/DesignCompile/CompileHelper.cpp:3004:39: error: 'uhdmarray_expr' is not a member of 'UHDM'; did you mean 'uhdmarray_var'?
3004 |         if (expr && exprtype == UHDM::uhdmarray_expr) {
      |                                       ^~~~~~~~~~~~~~
      |                                       uhdmarray_var
```

I did a google search on this error before posting.

hzeller, 2023-02-24:
The UHDM errors look like your submodules might be old ? What happens after you do
```
git submodule update --init --recursive
```
?

hzeller, 2023-02-24:
Your Ubunutu from 2018 is somewhat ancient (close to EOL). Is it possible to upgrade to a recent one ?
(I know, in some circumstances it is not possible, but having something recent helps avoiding chasing issues that would not exist otherwise).

alaindargelas, 2023-02-25:
That error is definitely about a submodule (UHDM) not being updated:
 array_expr* array = s.MakeArray_expr();
      |                               ^~~~~
      |                               std::array

git submodule update --init --recursive
should take care of it.

alaindargelas, 2023-02-27:
@bluehalo-user, can we close this issue?

bluehalo-user, 2023-02-27:
I still haven't been able to compile on windows (I am waiting for corporate permission for admin rights to turn on development mode), but your suggestion did work for Linux. Thanks.

I would sure like to get this working on Cygwin and Windows.

EXTERNAL

alaindargelas, 2023-03-03:
Please reopen if you still have Windows compilation problems, but many users compile on Windows.

closed 2023-03-03 by alaindargelas