chipsalliance/Surelog #3487: Removing unused modules from UHDM file + dependencies

StefanoBettelli, 2023-02-27:
I translate SystemVerilog files to UHDM with surelog.
Let us say that I want to create a design with top module MyModule:

```
>> /usr/local/bin/surelog --version
   VERSION: 1.47
   BUILT  : Jan 24 2023
>> /usr/local/bin/surelog -parse -synth -timescale=1ns/1ps
    -I$(EDIR) -I$(IDIR) +libext+.sv -y $(MDIR) -top MyModule MyModule.sv
```

In this call, EDIR is a directory with .sv files containing enumerations, IDIR is a directory with .sv files containing interfaces, and MDIR is a directory containing **a large collection** of SystemVerilog modules in .sv files.

After analysing the .uhdm file it is apparent that it contains all modules in the MDIR directory, whether or not they are instantiated through the top module. 

* Is it possible to call surelog in such a way that modules which are not used, directly or indirectly, by the top module are **removed from the generated .uhdm file**? If not, it would be a good idea to **include** such an option.
* Is it possible to call surelog in such a way that it generates a **dependency file** listing all .sv files used, directly or indirectly by the top module, to create the design? This would be very useful in Makefile flows.

Thomasb81, 2023-02-27:
Hello

As far as I understand, Surelog is a tool that will allow you to fill a uhdm database. What you do with the uhdm database is your custom application.
Surelog is able to perform some elaboration. One elaboration step consist to complete the uhdm database with what you want.

using `-d uhdm`, the tool will dump the complete uhdm tree.
You should be able to observe a difference by adding `-noelab`,

So to answer your question, you probably  have to write your own application to iterate on only the part of the uhdm data base that interest you.
See:
https://github.com/chipsalliance/Surelog/blob/5fc1993668337fa646adbf1e5e0bd139f2304293/src/hellouhdm.cpp#L103
vs
https://github.com/chipsalliance/Surelog/blob/5fc1993668337fa646adbf1e5e0bd139f2304293/src/hellouhdm.cpp#L150

alaindargelas, 2023-03-08:
@StefanoBettelli, can we close this issue? Are you OK with the file suggested above?

StefanoBettelli, 2023-03-09:
Unfortunately I haven't had much time recently to look at it again. However, it seems that this project can hardly be used beyond linting, since there isn't enough documentation, and it is not clear the level of elaboration that is / can be done. There are projects to offload the SystemVerilog fronted of verilator and yosys to Surelog, but so far my efforts at using them consistently in a Makefile flow have failed, due to several shortcomings (not all in Surelog).

For the time being the issue can be closed. I will come back to it if there is any substantial progress.

closed 2023-03-09 by StefanoBettelli

alaindargelas, 2023-03-09:
@StefanoBettelli you really mean a "Makefile" flow or a "CMake" flow?
If you really mean "Makefile" like you have a larger project that uses Makefile and you want to shoehorn Surelog CMake into it,
then it is best to invoke the cmake from the Makefile like in:
https://github.com/chipsalliance/Surelog/blob/efd45b9291cb4aa03c54e9d3d72d90b885571971/Makefile#L23 and not try to convert any of the CMake steps to Makefile

As for the API, the API is the Standard VPI interface (very well documented in the SystemVerilog standard):
Accessed using: https://github.com/chipsalliance/UHDM
Extract of the Standard API for reference: https://github.com/chipsalliance/Surelog/blob/master/third_party/Verilog_Object_Model.pdf