chipsalliance/Surelog #3520: Git tag 0.0 does not match project version 1.48

alaindargelas, 2023-03-22:
After an update I get:

~/Surelog$ make
cmake -DCMAKE_BUILD_TYPE=Release -DCMAKE_INSTALL_PREFIX=/usr/local -DBUILD_SHARED_LIBS=OFF  -S . -B build
CMake Error at CMakeLists.txt:81 (message):
  Git tag 0.0 does not match project version 1.48.  Did you forget to update
  project version in CMakeLists.txt?

-- Configuring incomplete, errors occurred!
make: *** [Makefile:46: run-cmake-release] Error 1

timkpaine, 2023-03-22:
We can remove soon (@hzeller has done the required CI changes) but have you run a `git fetch --all` recently? You have out of date git tags.

With the recently merged changes, the cmakelists is the single source of version truth and we can get rid of the git check (here and UHDM).

hzeller, 2023-03-22:
The [tag v1.48](https://github.com/chipsalliance/Surelog/releases/tag/v1.48)
should be there on hash 71eb4028f18d6b12833578fbcfa4db7b06869f78 
So if it is not there, probably the suggested `git fetch --all` should help.

PR #3529 (2023-03-22): Always extract version number from cmake project.
If git is available, also extract latest commit hash. (TODO: patch-level ? i.e. number of versions since last tag)

Fixes #3520
commit 92d30a85 (2023-03-22): Always extract version number from cmake project.

If git is available, also extract latest commit hash.
(TODO: patch-level ? i.e. number of versions since last tag)

Fixes #3520
closed 2023-03-22 by hzeller