chipsalliance/Surelog #3537: With external dependencies: Surelog headers, but no libs are installed

hzeller, 2023-03-24:
Looking closely at the install output of Surelog when the external dependencies are used (vendored_dependencies=false), no surelog library is installed, only the includes:

https://github.com/chipsalliance/Surelog/actions/runs/4504130636/jobs/7930318611#step:7:232

hzeller, 2023-03-24:
ah, LOL, ok :)
then the pkg-config didn't work for other reasons, let me see what is going on there.

hzeller, 2023-03-24:
Ok, it is the location that trips up the pkg-config build: It ends up in `${PREFIX}/lib/libsurelog.so` not like the static library in `${PREFIX}/lib/surelog/libsurelog.a`, this is why it isn't found.

timkpaine, 2023-03-24:
but...the install command is the same for both library types....cmon cmake..

timkpaine, 2023-03-24:
ah its `ARCHIVE DESTINATION`, that applies to static libraries
https://github.com/chipsalliance/Surelog/blob/4bcb3f079707308a31126526393a9500d07d7b1d/CMakeLists.txt#L752

https://cmake.org/cmake/help/book/mastering-cmake/chapter/Install.html#installing-targets

closed 2023-03-24 by hzeller