chipsalliance/Surelog #3541: strange install location of `uvm_pkg.sv.slpa`

hzeller, 2023-03-24:
On `make install`, the pre-compiled system verilog libs end up in `${PREFIX}/bin/pkg/work/`
```
ls -l bin/pkg/work 
total 23980
6169944 Mar 24 09:22 ovm_pkg.sv.slpa
1209576 Mar 24 09:22 ovm_pkg.sv.slpp
14381088 Mar 24 09:23 uvm_pkg.sv.slpa
2783720 Mar 24 09:22 uvm_pkg.sv.slpp
```

Which is a somewhat unusual location (or rather: "one should not do that" location :) ).

Typically, one would expect that in `${PREFIX}/share/surelog` (up to you if that should have a `pkg/` or `pkg/work/` subdirectory from there).

CC: @timkpaine who also has packaging experience.

timkpaine, 2023-03-24:
Agreed, I also noticed `pkg` ending up in `bin`, I would expect something like `share/surelog/` or `share/surelog/sv/` or similar

alaindargelas, 2023-03-27:
@hs-apotell, CMakeLists.txt and your new FileSystem class needs to be updated, can you take on that task?
I prefer the "share/surelog/pkg/" path as we are not saving .sv files but precompiled packages in binary format.

hzeller, 2023-03-27:
Currently, the install location is in  `${PREFIX}/bin/pkg/work`. So instead of the bin/ path, `${PREFIX}/share/surelog/pkg` is preferred.

alaindargelas, 2023-03-28:
Yes, ${PREFIX}/share/surelog/pkg/work  will work.

alaindargelas, 2023-03-31:
Fixed by #3556

closed 2023-03-31 by alaindargelas