chipsalliance/Surelog #3544: Resolution function not properly record in uhdm db

Thomasb81, 2023-03-24:
On following systemVerilog code
```
nettype real my_real with my_function2;

function automatic real my_function2(input real driver []);
endfunction
```

https://github.com/chipsalliance/Surelog/blob/0b89e9b867b6c262ca43b2cf65c1c05b5c81dbb7/src/DesignCompile/CompileHelper.cpp#L572

`compileHelper::getTaskFunc` return nothing leading to miss the association of the resolution function to the datatype, because the tool search if function is already in the udhm db. But function definition is not yet there.

so later `if (resolution_func) ` will fail and we will not successfully record the association of resolution function to the data type. 

Accoding to chapter 23.8.1 and the need to _look in the complete compilation unit_ let me think this reference of the resolution function should be deferred to a later stage in the tool to allow resolution to be done.

Is there something planned in the tool for this ?

alaindargelas, 2023-03-27:
This is similar to other "late binding" happening for data types and function binding. Check for late binding in the code base, an equivalent needs to be added for the resolution functions.

alaindargelas, 2023-03-31:
Let me work on this.

Thomasb81, 2023-03-31:
There is no urgency at all.

On this resolution function there is also a need for function prototype checker.

alaindargelas, 2023-04-01:
Fixed by #3560.
Checker/Linter is another issue.

closed 2023-04-01 by alaindargelas