chipsalliance/Surelog #3545: 'make test' fails to run tests

yurivict, 2023-03-27:
When it is configured with ```-DSURELOG_BUILD_TESTS=ON``` ```make test``` fails:
```
[  0% 1/1] cd /usr/ports/cad/surelog/work/.build && /usr/local/bin/ctest --force-new-ctest-process
Test project /usr/ports/cad/surelog/work/.build
      Start  1: PPCache_test_NOT_BUILT
Could not find executable PPCache_test_NOT_BUILT
Looked in the following places:
PPCache_test_NOT_BUILT
PPCache_test_NOT_BUILT
Release/PPCache_test_NOT_BUILT
Release/PPCache_test_NOT_BUILT
Debug/PPCache_test_NOT_BUILT
Debug/PPCache_test_NOT_BUILT
MinSizeRel/PPCache_test_NOT_BUILT
MinSizeRel/PPCache_test_NOT_BUILT
RelWithDebInfo/PPCache_test_NOT_BUILT
RelWithDebInfo/PPCache_test_NOT_BUILT
Deployment/PPCache_test_NOT_BUILT
Deployment/PPCache_test_NOT_BUILT
Development/PPCache_test_NOT_BUILT
Development/PPCache_test_NOT_BUILT
Unable to find executable: PPCache_test_NOT_BUILT
 1/14 Test  #1: PPCache_test_NOT_BUILT ..............***Not Run   0.00 sec
      Start  2: CommandLineParser_test_NOT_BUILT
Could not find executable CommandLineParser_test_NOT_BUILT
Looked in the following places:
CommandLineParser_test_NOT_BUILT
CommandLineParser_test_NOT_BUILT
Release/CommandLineParser_test_NOT_BUILT
Release/CommandLineParser_test_NOT_BUILT
Debug/CommandLineParser_test_NOT_BUILT
Debug/CommandLineParser_test_NOT_BUILT
MinSizeRel/CommandLineParser_test_NOT_BUILT
MinSizeRel/CommandLineParser_test_NOT_BUILT
RelWithDebInfo/CommandLineParser_test_NOT_BUILT
RelWithDebInfo/CommandLineParser_test_NOT_BUILT
Deployment/CommandLineParser_test_NOT_BUILT
Deployment/CommandLineParser_test_NOT_BUILT
Development/CommandLineParser_test_NOT_BUILT
Development/CommandLineParser_test_NOT_BUILT
Unable to find executable: CommandLineParser_test_NOT_BUILT
 2/14 Test  #2: CommandLineParser_test_NOT_BUILT ....***Not Run   0.00 sec
      Start  3: PathId_test_NOT_BUILT
Could not find executable PathId_test_NOT_BUILT
Looked in the following places:
PathId_test_NOT_BUILT
PathId_test_NOT_BUILT
Release/PathId_test_NOT_BUILT
Release/PathId_test_NOT_BUILT
Debug/PathId_test_NOT_BUILT
Debug/PathId_test_NOT_BUILT
MinSizeRel/PathId_test_NOT_BUILT
MinSizeRel/PathId_test_NOT_BUILT
RelWithDebInfo/PathId_test_NOT_BUILT
RelWithDebInfo/PathId_test_NOT_BUILT
Deployment/PathId_test_NOT_BUILT
Deployment/PathId_test_NOT_BUILT
Development/PathId_test_NOT_BUILT
Development/PathId_test_NOT_BUILT
Unable to find executable: PathId_test_NOT_BUILT
 3/14 Test  #3: PathId_test_NOT_BUILT ...............***Not Run   0.00 sec
      Start  4: PlatformFileSystem_test_NOT_BUILT
Could not find executable PlatformFileSystem_test_NOT_BUILT
Looked in the following places:
PlatformFileSystem_test_NOT_BUILT
PlatformFileSystem_test_NOT_BUILT
Release/PlatformFileSystem_test_NOT_BUILT
Release/PlatformFileSystem_test_NOT_BUILT
Debug/PlatformFileSystem_test_NOT_BUILT
Debug/PlatformFileSystem_test_NOT_BUILT
[... 205 lines cut ...]
The following tests FAILED:
	  1 - PPCache_test_NOT_BUILT (Not Run)
	  2 - CommandLineParser_test_NOT_BUILT (Not Run)
	  3 - PathId_test_NOT_BUILT (Not Run)
	  4 - PlatformFileSystem_test_NOT_BUILT (Not Run)
	  5 - CompileExpression_test_NOT_BUILT (Not Run)
	  6 - CompileHelper_test_NOT_BUILT (Not Run)
	  7 - Elaboration_test_NOT_BUILT (Not Run)
	  8 - Uhdm_test_NOT_BUILT (Not Run)
	  9 - ExprBuilder_test_NOT_BUILT (Not Run)
	 10 - ParseFile_test_NOT_BUILT (Not Run)
	 11 - PreprocessFile_test_NOT_BUILT (Not Run)
	 12 - SymbolTable_test_NOT_BUILT (Not Run)
	 13 - StringUtils_test_NOT_BUILT (Not Run)
	 14 - NumUtils_test_NOT_BUILT (Not Run)
Errors while running CTest
Output from these tests are in: /usr/ports/cad/surelog/work/.build/Testing/Temporary/LastTest.log
Use "--rerun-failed --output-on-failure" to re-run the failed cases verbosely.
FAILED: CMakeFiles/test.util 
cd /usr/ports/cad/surelog/work/.build && /usr/local/bin/ctest --force-new-ctest-process
ninja: build stopped: subcommand failed.
```

Version: 1.50
FreeBSD 13.1

alaindargelas, 2023-03-27:
@timkpaine, @hzeller something to do with the recent packaging work?

timkpaine, 2023-03-27:
I doubt it, but there's a lot of output omitted here so difficult to tell.

hzeller, 2023-03-28:
it looks like you're building Surelog in a slightly different build environment. It is not quite clear to me if you're using the third_party/ libs in the submodules, or are using the libraries supplied in your build environment.
In that case, make sure to give `-DSURELOG_USE_HOST_GTEST=On` to the cmake configuration

Then, in order to run the tests, you need to build the target `UnitTests`, they are not build by default. So t)he following minimal example should get you going (here: using `my-build-dir` as the build directory

```
cmake -DSURELOG_USE_HOST_GTEST=On -S . -B my-build-dir
cmake` --build my-build-dir --target UnitTests
cd my-build-dir ; ctest
```

yurivict, 2023-03-29:
I use all git submodules.

```-DSURELOG_USE_HOST_GTEST=On``` failed, see a separate bug report about this.

---------

There are 2 failures:
```
14/99 Test #14: src/Common/PlatformFileSystemTest.WorkingDirs_NonIdeal ..................................***Failed    0.03 sec
Running main() from /usr/ports/cad/surelog/work/Surelog-1.50/third_party/googletest/googletest/src/gtest_main.cc
Note: Google Test filter = PlatformFileSystemTest.WorkingDirs_NonIdeal
[==========] Running 1 test from 1 test suite.
[----------] Global test environment set-up.
[----------] 1 test from PlatformFileSystemTest
[ RUN      ] PlatformFileSystemTest.WorkingDirs_NonIdeal
/usr/ports/cad/surelog/work/Surelog-1.50/src/Common/PlatformFileSystem_test.cpp:400: Failure
Expected equality of these values:
  expectedFsWorkingDirs
    Which is: { "", "/tmp/ws_nonideal" }
  actualFsWorkingDirs
    Which is: { "/home/yuri/bin", "/tmp/ws_nonideal" }
[  FAILED  ] PlatformFileSystemTest.WorkingDirs_NonIdeal (1 ms)
[----------] 1 test from PlatformFileSystemTest (1 ms total)

[----------] Global test environment tear-down
[==========] 1 test from 1 test suite ran. (1 ms total)
[  PASSED  ] 0 tests.
[  FAILED  ] 1 test, listed below:
[  FAILED  ] PlatformFileSystemTest.WorkingDirs_NonIdeal

 1 FAILED TEST

      Start 15: src/Common/PlatformFileSystemTest.WorkingDirs_Ideal
15/99 Test #15: src/Common/PlatformFileSystemTest.WorkingDirs_Ideal .....................................***Failed    0.03 sec
Running main() from /usr/ports/cad/surelog/work/Surelog-1.50/third_party/googletest/googletest/src/gtest_main.cc
Note: Google Test filter = PlatformFileSystemTest.WorkingDirs_Ideal
[==========] Running 1 test from 1 test suite.
[----------] Global test environment set-up.
[----------] 1 test from PlatformFileSystemTest
[ RUN      ] PlatformFileSystemTest.WorkingDirs_Ideal
/usr/ports/cad/surelog/work/Surelog-1.50/src/Common/PlatformFileSystem_test.cpp:515: Failure
Expected equality of these values:
  expectedFsWorkingDirs
    Which is: { "", "/tmp/ws_ideal/tests/testname", "/tmp/ws_ideal/third_party" }
  actualFsWorkingDirs
    Which is: { "/home/yuri/bin", "/tmp/ws_ideal/tests/testname", "/tmp/ws_ideal/third_party" }
[  FAILED  ] PlatformFileSystemTest.WorkingDirs_Ideal (1 ms)
[----------] 1 test from PlatformFileSystemTest (1 ms total)

[----------] Global test environment tear-down
[==========] 1 test from 1 test suite ran. (1 ms total)
[  PASSED  ] 0 tests.
[  FAILED  ] 1 test, listed below:
[  FAILED  ] PlatformFileSystemTest.WorkingDirs_Ideal

 1 FAILED TEST

      Start 16: src/Common/PlatformFileSystemTest.InMemoryTest
16/99 Test #16: src/Common/PlatformFileSystemTest.InMemoryTest ..........................................   Passed    0.60 sec
```

closed 2023-05-27 by alaindargelas

yurivict, 2023-06-11:
@alaindargelas 

These tests still fail in 1.64:
```
The following tests FAILED:
	 14 - src/Common/PlatformFileSystemTest.WorkingDirs_NonIdeal (Failed)
	 15 - src/Common/PlatformFileSystemTest.WorkingDirs_Ideal (Failed)
```

flatbuffers, antlr4 and googletest are git submodules, the rest are external dependencies.

What is wrong with these tests?

alaindargelas, 2023-06-12:
Those are @hs-apotell  tests

yurivict, 2023-06-12:
```/home/yuri/bin``` is picked from the PATH variable. Why does it do this?
And why is the other value an empty string?

hzeller, 2023-06-12:
Why do we need the program path in the first place ? Is it because all configuration is relative to it ? e.g. /foo/bin/surelog looks at /foo/bin/surelog/../share/something ?

Maybe the thing that is different is that @yurivict is working on BSD, while all our assumptions are fairly Linux-centric.

yurivict, 2023-06-12:
Is it looking for the program path? It can be resolved through the PATH variable. Other tests find it fine, I guess.