chipsalliance/Surelog #3546: How do I get access to package parameters?

maxslug, 2023-03-27:
I would like to make a small tool that would elaborate a SystemVerilog package and then create an elaborated version.

example input
```
package my_pkg;

parameter int unsigned  MY_WIDTH = 31;
parameter int unsigned SOMETHING = MY_WIDTH;

endpackage
```

Example output:
```
MY_WIDTH=31
SOMETHING=31
```

Maybe I can get more fancy had use this as a way to document package files and visualize the data structures (struct, union, MDA, etc) too.

Question:  Is this information available to us in Surelog?  Where would be the best spot to extract this?

Thanks!

alaindargelas, 2023-03-28:
@maxslug, Surelog does all of that already, you can run your example.
As you can see the UHDM debug output shows the elaborated parameter under the uhdmTopPackages section.

surelog -parse my_pkg.sv  -d uhdm -nobuiltin

```
design: (unnamed)
|vpiName:unnamed
|uhdmallPackages:
\_package: my_pkg (my_pkg::), file:/home/alain/Surelog/my_pkg.sv, line:1:1, endln:6:11
  |vpiParent:
  \_design: (unnamed)
  |vpiName:my_pkg
  |vpiFullName:my_pkg::
  |vpiParameter:
  \_parameter: (my_pkg::MY_WIDTH), line:3:25, endln:3:33
    |vpiParent:
    \_package: my_pkg (my_pkg::), file:/home/alain/Surelog/my_pkg.sv, line:1:1, endln:6:11
    |UINT:31
    |vpiTypespec:
    \_int_typespec: , line:3:11, endln:3:23
      |vpiParent:
      \_parameter: (my_pkg::MY_WIDTH), line:3:25, endln:3:33
      |vpiInstance:
      \_package: (my_pkg)
    |vpiName:MY_WIDTH
    |vpiFullName:my_pkg::MY_WIDTH
  |vpiParameter:
  \_parameter: (my_pkg::SOMETHING), line:4:24, endln:4:33
    |vpiParent:
    \_package: my_pkg (my_pkg::), file:/home/alain/Surelog/my_pkg.sv, line:1:1, endln:6:11
    |vpiTypespec:
    \_int_typespec: , line:4:11, endln:4:23
      |vpiParent:
      \_parameter: (my_pkg::SOMETHING), line:4:24, endln:4:33
      |vpiInstance:
      \_package: (my_pkg)
    |vpiName:SOMETHING
    |vpiFullName:my_pkg::SOMETHING
  |vpiParamAssign:
  \_param_assign: , line:3:25, endln:3:38
    |vpiParent:
    \_package: my_pkg (my_pkg::), file:/home/alain/Surelog/my_pkg.sv, line:1:1, endln:6:11
    |vpiRhs:
    \_constant: , line:3:36, endln:3:38
      |vpiDecompile:31
      |vpiSize:32
      |UINT:31
      |vpiTypespec:
      \_int_typespec: , line:3:11, endln:3:23
      |vpiConstType:9
    |vpiLhs:
    \_parameter: (my_pkg::MY_WIDTH), line:3:25, endln:3:33
  |vpiParamAssign:
  \_param_assign: , line:4:24, endln:4:44
    |vpiParent:
    \_package: my_pkg (my_pkg::), file:/home/alain/Surelog/my_pkg.sv, line:1:1, endln:6:11
    |vpiRhs:
    \_ref_obj: (MY_WIDTH), line:4:36, endln:4:44
      |vpiName:MY_WIDTH
    |vpiLhs:
    \_parameter: (my_pkg::SOMETHING), line:4:24, endln:4:33
  |vpiDefName:my_pkg
|uhdmtopPackages:
\_package: my_pkg (my_pkg::), file:/home/alain/Surelog/my_pkg.sv, line:1:1, endln:6:11
  |vpiParent:
  \_design: (unnamed)
  |vpiName:my_pkg
  |vpiFullName:my_pkg::
  |vpiParameter:
  \_parameter: (my_pkg::MY_WIDTH), line:3:25, endln:3:33
    |vpiParent:
    \_package: my_pkg (my_pkg::), file:/home/alain/Surelog/my_pkg.sv, line:1:1, endln:6:11
    |UINT:31
    |vpiTypespec:
    \_int_typespec: , line:3:11, endln:3:23
      |vpiParent:
      \_parameter: (my_pkg::MY_WIDTH), line:3:25, endln:3:33
      |vpiInstance:
      \_package: my_pkg (my_pkg::), file:/home/alain/Surelog/my_pkg.sv, line:1:1, endln:6:11
    |vpiName:MY_WIDTH
    |vpiFullName:my_pkg::MY_WIDTH
  |vpiParameter:
  \_parameter: (my_pkg::SOMETHING), line:4:24, endln:4:33
    |vpiParent:
    \_package: my_pkg (my_pkg::), file:/home/alain/Surelog/my_pkg.sv, line:1:1, endln:6:11
    |UINT:31
    |vpiTypespec:
    \_int_typespec: , line:4:11, endln:4:23
      |vpiParent:
      \_parameter: (my_pkg::SOMETHING), line:4:24, endln:4:33
      |vpiInstance:
      \_package: my_pkg (my_pkg::), file:/home/alain/Surelog/my_pkg.sv, line:1:1, endln:6:11
    |vpiName:SOMETHING
    |vpiFullName:my_pkg::SOMETHING
  |vpiParamAssign:
  \_param_assign: , line:3:25, endln:3:38
    |vpiParent:
    \_package: my_pkg (my_pkg::), file:/home/alain/Surelog/my_pkg.sv, line:1:1, endln:6:11
    |vpiRhs:
    \_constant: , line:3:36, endln:3:38
      |vpiDecompile:31
      |vpiSize:32
      |UINT:31
      |vpiTypespec:
      \_int_typespec: , line:3:11, endln:3:23
      |vpiConstType:9
    |vpiLhs:
    \_parameter: (my_pkg::MY_WIDTH), line:3:25, endln:3:33
  |vpiParamAssign:
  \_param_assign: , line:4:24, endln:4:44
    |vpiParent:
    \_package: my_pkg (my_pkg::), file:/home/alain/Surelog/my_pkg.sv, line:1:1, endln:6:11
    |vpiRhs:
    \_constant: , line:4:36, endln:4:44
      |vpiDecompile:31
      |vpiSize:32
      |UINT:31
      |vpiTypespec:
      \_int_typespec: , line:4:11, endln:4:23
      |vpiConstType:9
    |vpiLhs:
    \_parameter: (my_pkg::SOMETHING), line:4:24, endln:4:33
  |vpiDefName:my_pkg
  |vpiTop:1
===================
```

You can create your own application that reads the produced UHDM database or embed the whole Surelog/UHDM into your application.

There are plenty of examples on how to do that in the README.

maxslug, 2023-03-28:
This is fantastic, thanks for running the example through and providing the output @alaindargelas ! 

I've browsed around but I haven't found an easy way of traversing the UHDM through a scripting language.  I suppose I could just directly use the YAML or use https://github.com/mikefarah/yq.  

However I thought I would ask if there are any existing efforts to make a friendly front-end to UHDM for Python that I could join?   My 2 second version is something like:

```
u = UHDM.read_design(...)
for module in u.modules:
  for port in module.ports:
    print(f"{port.name} is an {port.type}")
```

alaindargelas, 2023-03-28:
Looks like you are in luck, there is a PR for UHDM-Python API in flight:
https://github.com/chipsalliance/UHDM/pull/895

The existing APIs are in C (Standard Verilog VPI) or C++ (UHDM):

Surelog + UHDM in your "main":
https://github.com/chipsalliance/Surelog/blob/master/src/hellouhdm.cpp

Load UHDM and do something about it:
https://github.com/chipsalliance/UHDM/blob/master/util/uhdm-hier.cpp

maxslug, 2023-03-28:
Great timing indeed!   It looks like that will provide access to the SWIG functions to Python libraries.   What is the general feel for having an abstraction from the syntax tree in Python?    It seems you had the same idea here before : https://github.com/chipsalliance/Surelog/issues/2635#issuecomment-1046123993  .  I feel like this will go a long way to bridge the gap for having a good SV compiler accessible for scripted chip design work.

alaindargelas, 2023-03-28:
You need to work it out with @Thomasb81, maybe a separate PR is needed to do https://github.com/chipsalliance/Surelog/issues/2635#issuecomment-1046123993

maxslug, 2023-03-28:
Sounds good, I'll wait for the dust to settle on PR#985 and then try to use the API directly and then propose some layer of abstraction.   If there is a semi-standard API out there I would try to get close to that.

alaindargelas, 2023-04-01:
@maxslug looks like https://github.com/chipsalliance/UHDM/pull/895 (merged) is ready for a beta tester!

alaindargelas, 2023-04-24:
#3612 addresses the documentation of the Python API to UHDM.

closed 2023-04-24 by alaindargelas

maxslug, 2023-04-24:
Thanks @alaindargelas !  You might want to leave a breadcrumb at https://github.com/chipsalliance/Surelog/blob/master/README.md#python-api about the new Python VPI API for elaborated code in UHDM.