chipsalliance/Surelog #3551: FindANTLR.cmake: should also look in `$CLASSPATH`

hzeller, 2023-03-28:
The `cmake/modules/FindANTLR.cmake` looks in a few hard-coded well-known locations for a potential antlr*complete.jar.

https://github.com/chipsalliance/Surelog/blob/edb409a1f730d59665c312551aadc0f76af1c82e/cmake/modules/FindANTLR.cmake#L13-L21

However, that is not necessarily where the file is.
It would be good if it was also consulting the `CLASSPATH` (arguably, it even should look there _first_ before going into guess-mode rummaging through directories).

While at it, looking for exactly version 4.12.0 is very specific, can we change that to 4.12.* to be future compatible in case there are new patchlevels coming out ?

timkpaine, 2023-03-29:
I took a look at where debian and homebrew put the asset, I didn't see anything about setting `classpath` or configuring the users's profile to have that var. Is it the standard for folks to maintain a classpath var like they might a PATH? Maybe antlr already has something to find where the jar is? But yeah easy to implement, just unfamiliar with Java ecosystem.

hzeller, 2023-03-29:
In NixOS, the jar file ends up in the CLASSPATH (and to show that it is not predictable: in my case, the setting is `CLASSPATH=/nix/store/6cnfr2cy5ninj4xnhnqk8x4vkrz8bydn-antlr-4.12.0/share/java/antlr-4.12.0-complete.jar`)

In the Java-world, `CLASSPATH` common - it is what `PATH` is for binaries.

timkpaine, 2023-03-29:
Ok makes sense, will add