chipsalliance/Surelog #3552: Configure fails with external flatbuffers

yurivict, 2023-03-29:
When flatbuffers-23.3.3,1 is installed and ```-DSURELOG_USE_HOST_FLATBUFFERS=ON``` is set the configure fails:
```
CMake Error at CMakeLists.txt:552 (target_link_libraries):
  Target "surelog" links to:

    flatbuffers::flatbuffers

  but the target was not found.  Possible reasons include:

    * There is a typo in the target name.
    * A find_package call is missing for an IMPORTED target.
    * An ALIAS target is missing.
```

cmake-3.25.1
FreeBSD 13.1

yurivict, 2023-03-29:
All cmake files are in place.

This is a flatbuffers bug: https://github.com/google/flatbuffers/issues/7884

But find_package for flatbuffers should be ```find_package(Flatbuffers CONFIG REQUIRED)``` (with the capital letter and CONFIG.)

hzeller, 2023-03-30:
In the CI, this is tested with flatbuffers, built straight from the source, and it works there.
What is different in your set-up ?

yurivict, 2023-03-30:
What cmake and flatbuffers versions are used?

For me the failure occurs with flatbuffers-23.3.3 and cmake-3.25.1

timkpaine, 2023-03-30:
We've tested mac, nixos, 2 versions of Ubuntu, centos 7, and redhat 8, will need more info and build logs to debug further.

closed 2023-03-30 by timkpaine

johannesneyer, 2023-04-05:
@timkpaine  I get the same error on arch linux:

```
> make ADDITIONAL_CMAKE_OPTIONS="-DSURELOG_USE_HOST_FLATBUFFERS=ON -DSURELOG_USE_HOST_ANTLR=ON -DSURELOG_USE_HOST_UHDM=ON -DSURELOG_USE_HOST_GTEST=ON -DANTLR_JAR_LOCATION=/usr/share/java/antlr-complete.jar" release-shared
cmake -DCMAKE_BUILD_TYPE=Release -DCMAKE_INSTALL_PREFIX=/usr/local -DBUILD_SHARED_LIBS=ON -DSURELOG_USE_HOST_FLATBUFFERS=ON -DSURELOG_USE_HOST_ANTLR=ON -DSURELOG_USE_HOST_UHDM=ON -DSURELOG_USE_HOST_GTEST=ON -DANTLR_JAR_LOCATION=/usr/share/java/antlr-complete.jar -S . -B build
-- The C compiler identification is GNU 12.2.1
-- The CXX compiler identification is GNU 12.2.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
Building version v1.51 [0460207f9332de32c08b942829ed8f1af233d28a]
-- Found ANTLR: /usr/share/java/antlr-complete.jar (found version "4.11.1")
-- Found UHDM: /usr/include
-- Found OpenSSL: /usr/lib/libcrypto.so (found version "3.0.8") found components: Crypto SSL
-- Found GTest: /usr/lib/cmake/GTest/GTestConfig.cmake (found version "1.13.0")
-- Found Python3: /usr/bin/python3.10 (found suitable version "3.10.10", minimum required is "3.3") found components: Interpreter
-- Python3_EXECUTABLE = /usr/bin/python3.10
CMake Warning at CMakeLists.txt:151 (message):
  Using tcmalloc; if this creates runtime issues, configure with
  -DSURELOG_WITH_TCMALLOC=Off

-- Found tcmalloc: /usr/lib/libtcmalloc.so
-- Found Java: /usr/bin/java (found suitable version "19.0.2", minimum required is "11") found components: Runtime
-- Java_JAVA_EXECUTABLE = /usr/bin/java
-- Configuring done (0.9s)
CMake Error at CMakeLists.txt:552 (target_link_libraries):
  Target "surelog" links to:

    flatbuffers::flatbuffers

  but the target was not found.  Possible reasons include:

    * There is a typo in the target name.
    * A find_package call is missing for an IMPORTED target.
    * An ALIAS target is missing.

-- Generating done (0.0s)
CMake Generate step failed.  Build files cannot be regenerated correctly.
make: *** [Makefile:49: run-cmake-release-shared] Error 1
```

johannesneyer, 2023-04-05:
this patch makes it work:
```diff
diff --git a/CMakeLists.txt b/CMakeLists.txt
index 2f8c88532..49248f746 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -84,7 +84,7 @@ endif()
 set(FLATBUFFERS_FLATC_EXECUTABLE "flatc")
 if(SURELOG_USE_HOST_FLATBUFFERS)
   find_package(Flatbuffers REQUIRED)
-  set(FLATBUFFERS_LIBRARY flatbuffers::flatbuffers)
+  set(FLATBUFFERS_LIBRARY flatbuffers)
 else()
   set(FLATBUFFERS_BUILD_TESTS OFF CACHE BOOL "Skip flatbuffers' tests")
   add_subdirectory(third_party/flatbuffers EXCLUDE_FROM_ALL)
```

johannesneyer, 2023-04-05:
Hmm, arch [does not seem to be patching flatbuffers](https://github.com/archlinux/svntogit-community/blob/packages/flatbuffers/trunk/PKGBUILD). I am not proficient enough in cmake to tell what is going on here. I'll just add a patch for this...